flyte/mqtt-io #92: local loop

maciaswojtas, 2020-03-31:
I would like to add enhancement with local loop. 
My usage scenario is that I want to control set of relays and light switches. 
Relays defined as output GPIO and Switches as input GPIO.
I have logic implemented in NodeRed which is deployed on other machine.
Locally I want to be able to have basic setup so pair input GPIO with output GPIO to be able to working with basic scenario locally. 

I would like to add configuration for input with information to which output this message should be also propagated locally. 

Tell me if such logic fits your idea of mqtt-gpio

BenjiU, 2020-04-14:
Hi, it is also in #19 mentioned, but the "last" decision was, that this program is a "stupid" application, only listening to commands. This also avoids, that you have "logic configuration" on several machines.

BenjiU, 2020-04-15:
When I thought about your issue yesterday, I came to a new scenario: I have a raspberry connected the switches and lights of a room. The switching logic is on a different machine. So if the network connection is broken, i can't switch the light on :-(
Maybe we could add such a logic and maybe one can define, if it is always activated or only, if the mqtt connection is broken.
What basic connections between input and output is necessary, do you think?

flyte, 2020-04-15:
Yeah, this has come up before, as @BenjiU says, in #19. It does make some sense, especially if the connection to the device running this software is unreliable or slow. I'd like to keep all discussion of the pros and cons to #19 though, so I'm closing this issue in favour of that one. Feel free to add comments there.

closed 2020-04-15 by flyte