flyte/mqtt-io #95: Proposed addition of serial port (and other streaming hardware)

ChrisEAlfred, 2020-04-20:
Brilliant work - I like your coding style.

I was thinking of contributing support for streaming hardware - initially serial ports (via PySerial).

Does this schema look ok to you:

```
stream_modules:
  - name: comms
    module: streamserial
    device: /dev/ttyS0
    baud: 115200
    cleanup: yes

stream_reads:
  - name: serialrx
    module: comms
    interval: 0.25

stream_writes:
  - name: serialtx
    module: comms
```

BenjiU, 2020-04-20:
Hey, I like that: I have a smart meter, that gives me a serial output and is currently parsed in perl, but i want to move it to this, for example. So i would send the raw data via mqtt and parse it somewhere else.
What does your interval 0.25 mean? 
My smartmeter gives an output every second: a few hundred bytes.
So we would have a background task collecting the data and send it every 0.25 sec or if there is a pause? I would like to collect the whole data first and not send a split data in two frames.

ChrisEAlfred, 2020-04-20:
The 0.25 is the stream receive polling interval in seconds.

I already have a working version - testing happily here.

As for your serial case, in past systems we have used an 'idle time' concept (also called idle line).
You buffer the data until there is a pause for more than the idle time, then send the buffer. Any character triggers the start of buffer reading.

So we could add an optional parameter like:

```
stream_reads:
  - name: serialrx
    module: comms
    interval: 0.25
    idlebuffer: yes   # optional, default to "no"
```

idlebuffer: yes will buffer received data until there is one interval of no more received data and the data will be sent in one go.

BenjiU, 2020-04-21:
Maybe I don't need the idlebuffer thing. I just set the interval to twice the frequence of my smart meter, so I get a whole buffer each transmittion, but just have to find the start :-)
Your implementation looks good. Shall we add the other parameters (as defaults with the common parameters) to the Serial() function?

- bytesize – Number of data bits. Possible values: FIVEBITS, SIXBITS, SEVENBITS, EIGHTBITS
- parity – Enable parity checking. Possible values: PARITY_NONE, PARITY_EVEN, PARITY_ODD PARITY_MARK, PARITY_SPACE
- stopbits – Number of stop bits. Possible values: STOPBITS_ONE, STOPBITS_ONE_POINT_FIVE, STOPBITS_TWO

Or do we wait, until someone needs them?

Notes:
Usage of CONFIG_SCHEMA in https://github.com/flyte/pi-mqtt-gpio/compare/develop...ChrisEAlfred:feature/streams#diff-7e5341524882fd38e265ba466aa4ac74R7-R10 is a good idea, but then we should remove it from the config.schema.yml https://github.com/flyte/pi-mqtt-gpio/compare/develop...ChrisEAlfred:feature/streams#diff-0ab99ddee45bf345e48d79ae48ea7ecdR164-R171

flyte, 2020-04-21:
This is an interesting proposition indeed. Would you mind creating a PR and moving the discussion over there? It'll make it easier for us to establish what code changes are involved. Doesn't matter if it's not ready to merge yet, we can mark it as a work in progress.

Thanks!

ChrisEAlfred, 2020-04-22:
PR created

flyte, 2021-01-24:
https://github.com/flyte/pi-mqtt-gpio/pull/97

closed 2021-01-24 by flyte