flyte/mqtt-io #96: Example of pin address for BeagleBone Black

jfiger, 2020-04-20:
I did not find an example of pin address for BeagleBone Black in the documentation.
 pin: P8_7 is accepted  but not working for digital output. No error message.

config.yml
mqtt:
  host: 192.168.10.123
  port: 1883
  user: "mqtt"
  password: "mqtt"
  topic_prefix: home/bbb

gpio_modules:
  - name: beaglebone
    module: beaglebone

digital_outputs:
  - name: lights
    module: beaglebone
    pin: P8_7
    on_payload: "ON"
    off_payload: "OFF"
    initial: low  # This optional value controls the initial state of the pin 
    retain: yes # This option value controls if the message is retained.

BenjiU, 2020-04-21:
Hmm, naming should be the same, as for the used "adafruit-beaglebone-io-python" lib. So P8_7 looks good.
I could think of any "access rights". Are you running the application as root or other user, that has privileges to set pins as output? But I don't have any BeagleBone here to test :-(

jfiger, 2020-04-21:
OK. Thank you. It's working now with P8_7 both output and input. It was a topic mistake.

closed 2020-04-21 by jfiger