flyte/mqtt-io #101: Awning control - Need help and guidance..

itayraviv, 2020-05-08:
Hi,
First of all, this looks awesome and might be the solution i've been looking for but i need some help with implementing it.
I currently run HA on a laptop running dockers with mosquito, HA and others.
I also have an Rpi3 at my yard for other uses. here is what i want to do:
I want to take a the awning remote control, open it up and solder wires from the buttons to the Rpi GPIO so i could simulate a button press through the remote HA.
i know its not possible to track its location or open/close percentage through the GPIO, controlling it will be a good start.
Did someone maybe implemented such a solution? and can share some information or assist?

Thank you.

ChrisEAlfred, 2020-05-11:
If there is only a couple of buttons you can use a relay.

Otherwise, you can use 74HC4016 quad analog switches across the buttons. Or if you buzz out the key matrix, you can use 74HC4051s - one for 'row' and one for 'column' and connect their common I/O (pin 3) together.

I have used the analog switches method many times to control old synthesizer keyboards.

flyte, 2020-05-30:
Hi @itayraviv 
I'm going to close this issue as it's not really related to a project issue. I hope @ChrisEAlfred helped answer your question though. If not, perhaps the Home Assistant forums might be a more suitable place for the discussion.

closed 2020-05-30 by flyte