flyte/mqtt-io #103: Why it subscribes to topic ‘/output... in a cycle infinitly?

a-x-, 2020-05-10:
I suppose subscription is a one time action right after starting a daemon

BenjiU, 2020-05-10:
Hi, do you have a log output for us please? And maybe your config.yml would be helpful. Thanks.

a-x-, 2020-05-10:
raspberry pi 1,

```
2020-05-10 19:40:37,902 mqtt_gpio (INFO): Connected to the MQTT broker with protocol v3.1.1.
2020-05-10 19:40:37,910 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set'
2020-05-10 19:40:37,917 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set_on_ms'
2020-05-10 19:40:37,924 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set_off_ms'
2020-05-10 19:40:37,931 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set'
2020-05-10 19:40:37,941 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set_on_ms'
2020-05-10 19:40:37,948 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set_off_ms'
2020-05-10 19:40:39,924 mqtt_gpio (INFO): Connected to the MQTT broker with protocol v3.1.1.
2020-05-10 19:40:39,932 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set'
2020-05-10 19:40:39,939 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set_on_ms'
2020-05-10 19:40:39,945 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set_off_ms'
2020-05-10 19:40:39,952 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set'
2020-05-10 19:40:39,958 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set_on_ms'
2020-05-10 19:40:39,965 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set_off_ms'
2020-05-10 19:40:41,952 mqtt_gpio (INFO): Connected to the MQTT broker with protocol v3.1.1.
2020-05-10 19:40:41,964 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set'
2020-05-10 19:40:41,974 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set_on_ms'
2020-05-10 19:40:41,984 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set_off_ms'
2020-05-10 19:40:41,994 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set'
2020-05-10 19:40:42,004 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set_on_ms'
2020-05-10 19:40:42,013 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set_off_ms'
```

a-x-, 2020-05-10:
```yaml
mqtt:
  host: rpi4.local
  port: 1883
  user: ""
  password: ""
  discovery: yes # HomeAssistant

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_inputs:
  - name: hall_movs
    module: raspberrypi
    pin: 14
    interrupt_payload: 'ON'
    interrupt: rising
    bouncetime: 233
    on_payload: "OFF" # inverted due to bug in pi-mqtt-gpio
    off_payload: "ON"
    pullup: no
    pulldown: yes
  - name: hall_wall_switch
    module: raspberrypi
    pin: 23
    on_payload: "OFF" # inverted due to bug in lib
    off_payload: "ON"
    retain: yes
    pullup: yes
    pulldown: no
```

a-x-, 2020-05-10:
also home assistant becomes flickering and freezing after this daemon on rpi1 started 

I also have second pi-mqtt-gpio runned on another raspberry (rpi zero) in the same lan. The second one works good

BenjiU, 2020-05-10:
Yeah, something is wrong with this rpi1. because it always reconnects:
2020-05-10 19:40:37,902 mqtt_gpio (INFO): Connected to the MQTT broker with protocol v3.1.1.
2020-05-10 19:40:41,952 mqtt_gpio (INFO): Connected to the MQTT broker with protocol v3.1.1.
is it running with supervisor? Maybe something crashed and supervisor starts it again?
Have you looked into the logfiles of your mqtt client? Maybe there is some information about the disconnect?

a-x-, 2020-05-10:
> is it running with supervisor? 
no, I just run it with python3 -m pi_mqtt_gpio.server ~/.config/pi-mqtt-gpio.yml

a-x-, 2020-05-10:
mqtt client? hmm,

I use mosquitto as a mqtt server, but I has not intersting logs:
```
journalctl -f -u mosquitto
-- Logs begin at Thu 2019-02-14 13:11:59 MSK. --
May 09 00:09:54 rpi4 systemd[1]: Starting Mosquitto MQTT v3.1/v3.1.1 Broker...
May 09 00:09:55 rpi4 systemd[1]: Started Mosquitto MQTT v3.1/v3.1.1 Broker.
```

a-x-, 2020-05-10:
also, /var/log/mosquitto/mosquitto.log:
```
1589131208: New connection from 192.168.88.180 on port 1883.
1589131208: Client pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709 already connected, closing old connection.
1589131208: Socket error on client pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709, disconnecting.
1589131208: New client connected from 192.168.88.180 as pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709 (c0, k60).
1589131209: New connection from 192.168.88.211 on port 1883.
1589131209: Client pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709 already connected, closing old connection.
1589131209: Socket error on client pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709, disconnecting.
1589131209: New client connected from 192.168.88.211 as pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709 (c0, k60).
1589131210: New connection from 192.168.88.180 on port 1883.
1589131210: Client pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709 already connected, closing old connection.
1589131210: Socket error on client pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709, disconnecting.
1589131210: New client connected from 192.168.88.180 as pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709 (c0, k60).
1589131211: New connection from 192.168.88.211 on port 1883.
1589131211: Client pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709 already connected, closing old connection.
1589131211: Socket error on client pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709, disconnecting.
1589131211: New client connected from 192.168.88.211 as pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709 (c0, k60).
1589131212: New connection from 192.168.88.180 on port 1883.
```

a-x-, 2020-05-10:
Have any ideas?

a-x-, 2020-05-10:
while it constanly reconnecting, it partially works

BenjiU, 2020-05-10:
Is something connected to the gpios? Or you change the log level according to readme? Maybe that give some hints? Since it works on a different pi, I don't need to try to reproduce it.

a-x-, 2020-05-10:
Ok, I make pi-mqtt-gpio log more verbose (DEBUG level):
```
2020-05-10 22:52:36,050 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m26) [(b'/output/hall_wall_light/set_off_ms', 1)]

2020-05-10 22:52:36,062 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set_off_ms'
2020-05-10 22:52:36,073 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m27) [(b'/output/hall_led/set', 1)]
2020-05-10 22:52:36,081 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set'
2020-05-10 22:52:36,087 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m28) [(b'/output/hall_led/set_on_ms', 1)]
2020-05-10 22:52:36,092 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set_on_ms'
2020-05-10 22:52:36,099 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m29) [(b'/output/hall_led/set_off_ms', 1)]
2020-05-10 22:52:36,104 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set_off_ms'
2020-05-10 22:52:36,111 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q1, r1, m30), 'b'/status'', ... (7 bytes)
2020-05-10 22:52:36,118 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q0, r1, m31), 'b'homeassistant/binary_sensor/pi-mqtt-gpio-da39a3ee/hall_movs/config'', ... (379 bytes)
2020-05-10 22:52:36,126 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q0, r1, m32), 'b'homeassistant/binary_sensor/pi-mqtt-gpio-da39a3ee/hall_wall_switch/config'', ... (400 bytes)
2020-05-10 22:52:36,134 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q0, r1, m33), 'b'homeassistant/switch/pi-mqtt-gpio-da39a3ee/hall_wall_light/config'', ... (447 bytes)
2020-05-10 22:52:36,142 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q0, r1, m34), 'b'homeassistant/switch/pi-mqtt-gpio-da39a3ee/hall_led/config'', ... (419 bytes)
2020-05-10 22:52:36,154 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-10 22:52:36,160 mqtt_gpio (DEBUG): MQTT client: Received SUBACK

2020-05-10 22:52:36,167 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-10 22:52:36,175 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-10 22:52:36,181 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-10 22:52:36,187 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-10 22:52:36,194 mqtt_gpio (DEBUG): MQTT client: Received PUBACK (Mid: 30)
^C
2020-05-10 22:52:36,980 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q1, r1, m35), 'b'/status'', ... (7 bytes)
2020-05-10 22:52:36,994 mqtt_gpio (DEBUG): MQTT client: Received PUBACK (Mid: 35)
2020-05-10 22:52:37,004 mqtt_gpio (DEBUG): MQTT client: Sending DISCONNECT
```

a-x-, 2020-05-10:
and yes, configured gpio-pins are connected and working ok. I can read/write them via wiringpi cli

I cannot find any failover-cycle in your code, also.

a-x-, 2020-05-11:
I removed `while True:`
and then I see that mqtt_pgio 
1. do Startup,
2. then says: `Last will set on '/status' as 'dead'.`
3. then successfully performs:
    `MQTT client: Sending CONNECT (u0, p0, wr1, wq1, wf1, c0, k60) client_id=b'pi-mqtt-gpio-****'`
4. subscribes to topics and publishes something
5. gets acks
6. and Sends DISCONNECT

<details><summary>full log</summary>

```
2020-05-11 09:50:45,664 mqtt_gpio (INFO): Startup
2020-05-11 09:50:50,041 mqtt_gpio (DEBUG): Last will set on '/status' as 'dead'.
2020-05-11 09:50:50,563 mqtt_gpio (DEBUG): MQTT client: Sending CONNECT (u0, p0, wr1, wq1, wf1, c0, k60) client_id=b'pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709'
2020-05-11 09:50:50,571 mqtt_gpio (DEBUG): MQTT client: Received CONNACK (1, 0)
2020-05-11 09:50:50,574 mqtt_gpio (INFO): Connected to the MQTT broker with protocol v3.1.1.
2020-05-11 09:50:50,577 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m1) [(b'/output/hall_wall_light/set', 1)]
2020-05-11 09:50:50,580 mqtt_gpio (INFO): Polling: Input 'hall_wall_switch' state changed to True
2020-05-11 09:50:50,584 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q0, r1, m2), 'b'/input/hall_wall_switch'', ... (3 bytes)
scheduler
2020-05-11 09:50:50,586 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set'
2020-05-11 09:50:50,593 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m4) [(b'/output/hall_wall_light/set_on_ms', 1)]
2020-05-11 09:50:50,596 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set_on_ms'
2020-05-11 09:50:50,590 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q1, r1, m3), 'b'/status'', ... (7 bytes)
2020-05-11 09:50:50,599 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m5) [(b'/output/hall_wall_light/set_off_ms', 1)]
2020-05-11 09:50:50,604 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set_off_ms'
2020-05-11 09:50:50,607 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m6) [(b'/output/hall_led/set', 1)]
2020-05-11 09:50:50,610 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set'
2020-05-11 09:50:50,613 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m7) [(b'/output/hall_led/set_on_ms', 1)]
2020-05-11 09:50:50,616 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set_on_ms'
2020-05-11 09:50:50,619 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m8) [(b'/output/hall_led/set_off_ms', 1)]
2020-05-11 09:50:50,621 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set_off_ms'
2020-05-11 09:50:50,624 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q1, r1, m9), 'b'/status'', ... (7 bytes)
2020-05-11 09:50:50,631 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q0, r1, m10), 'b'homeassistant/binary_sensor/pi-mqtt-gpio-da39a3ee/hall_movs/config'', ... (379 bytes)
2020-05-11 09:50:50,635 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q0, r1, m11), 'b'homeassistant/binary_sensor/pi-mqtt-gpio-da39a3ee/hall_wall_switch/config'', ... (400 bytes)
2020-05-11 09:50:50,639 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q0, r1, m12), 'b'homeassistant/switch/pi-mqtt-gpio-da39a3ee/hall_wall_light/config'', ... (447 bytes)
2020-05-11 09:50:50,643 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q0, r1, m13), 'b'homeassistant/switch/pi-mqtt-gpio-da39a3ee/hall_led/config'', ... (419 bytes)
2020-05-11 09:50:50,654 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 09:50:50,657 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 09:50:50,661 mqtt_gpio (DEBUG): MQTT client: Received PUBACK (Mid: 3)
2020-05-11 09:50:50,664 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 09:50:50,668 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 09:50:50,671 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 09:50:50,675 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 09:50:50,679 mqtt_gpio (DEBUG): MQTT client: Received PUBACK (Mid: 9)
2020-05-11 09:50:50,682 mqtt_gpio (DEBUG): MQTT client: Sending DISCONNECT
```

</details>

BenjiU, 2020-05-11:
> have an idea why it becones ‘dead’?
What do you mean? At 2020-05-11 09:50:50,041 mqtt_gpio? This is only a trace for reading the config file.

> I removed while True:
Why have you removed this? If you remove this, than mqtt_client terminates instantly.

a-x-, 2020-05-11:
> terminates instantly.

I understand. It’s temorary for more convenient log grabbing w/o repetitions

BenjiU, 2020-05-11:
then you may change the sleep(0.01) to sleep(1)

BenjiU, 2020-05-11:
You may change the print("") in the try block to:
```
    except KeyboardInterrupt:
        print("Exiting Loop keyboard")
```

a-x-, 2020-05-11:
Ok, I did it,

<details><summary>a grabbed long repetitive log and finished it by ^C finally</summary>

```
2020-05-11 10:40:00,275 mqtt_gpio (INFO): Startup
2020-05-11 10:40:04,649 mqtt_gpio (DEBUG): Last will set on '/status' as 'dead'.
/home/pi/.local/lib/python3.5/site-packages/pi_mqtt_gpio/modules/raspberrypi.py:47: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self.io.setup(pin, direction, pull_up_down=pullup, initial=initial)
2020-05-11 10:40:05,165 mqtt_gpio (DEBUG): MQTT client: Sending CONNECT (u0, p0, wr1, wq1, wf1, c0, k3600) client_id=b'pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709'
2020-05-11 10:40:05,173 mqtt_gpio (DEBUG): MQTT client: Received CONNACK (1, 0)
2020-05-11 10:40:05,176 mqtt_gpio (INFO): Connected to the MQTT broker with protocol v3.1.1.
2020-05-11 10:40:05,180 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m1) [(b'/output/hall_wall_light/set', 1)]
2020-05-11 10:40:05,182 mqtt_gpio (INFO): Polling: Input 'hall_wall_switch' state changed to True
2020-05-11 10:40:05,187 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q0, r1, m2), 'b'/input/hall_wall_switch'', ... (3 bytes)
2020-05-11 10:40:05,189 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set'
2020-05-11 10:40:05,195 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m3) [(b'/output/hall_wall_light/set_on_ms', 1)]
2020-05-11 10:40:05,199 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set_on_ms'
2020-05-11 10:40:05,203 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m4) [(b'/output/hall_wall_light/set_off_ms', 1)]
2020-05-11 10:40:05,206 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set_off_ms'
2020-05-11 10:40:05,210 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m5) [(b'/output/hall_led/set', 1)]
2020-05-11 10:40:05,214 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set'
2020-05-11 10:40:05,218 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m6) [(b'/output/hall_led/set_on_ms', 1)]
2020-05-11 10:40:05,221 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set_on_ms'
2020-05-11 10:40:05,225 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m7) [(b'/output/hall_led/set_off_ms', 1)]
2020-05-11 10:40:05,229 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set_off_ms'
2020-05-11 10:40:05,233 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q1, r1, m8), 'b'/status'', ... (7 bytes)
2020-05-11 10:40:05,238 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q0, r1, m9), 'b'homeassistant/binary_sensor/pi-mqtt-gpio-da39a3ee/hall_movs/config'', ... (379 bytes)
2020-05-11 10:40:05,243 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q0, r1, m10), 'b'homeassistant/binary_sensor/pi-mqtt-gpio-da39a3ee/hall_wall_switch/config'', ... (400 bytes)2020-05-11 10:40:05,248 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q0, r1, m11), 'b'homeassistant/switch/pi-mqtt-gpio-da39a3ee/hall_wall_light/config'', ... (447 bytes)
2020-05-11 10:40:05,252 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q0, r1, m12), 'b'homeassistant/switch/pi-mqtt-gpio-da39a3ee/hall_led/config'', ... (419 bytes)
2020-05-11 10:40:05,262 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 10:40:05,267 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 10:40:05,309 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 10:40:05,313 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 10:40:05,317 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 10:40:05,321 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 10:40:05,326 mqtt_gpio (DEBUG): MQTT client: Received PUBACK (Mid: 8)
2020-05-11 10:40:06,193 mqtt_gpio (INFO): next loop
2020-05-11 10:40:07,195 mqtt_gpio (DEBUG): MQTT client: Sending CONNECT (u0, p0, wr1, wq1, wf1, c0, k3600) client_id=b'pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709'
2020-05-11 10:40:07,204 mqtt_gpio (DEBUG): MQTT client: Received CONNACK (1, 0)
2020-05-11 10:40:07,208 mqtt_gpio (INFO): next loop
2020-05-11 10:40:07,210 mqtt_gpio (INFO): Connected to the MQTT broker with protocol v3.1.1.
2020-05-11 10:40:07,219 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m13) [(b'/output/hall_wall_light/set', 1)]
2020-05-11 10:40:07,224 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set'
2020-05-11 10:40:07,230 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m14) [(b'/output/hall_wall_light/set_on_ms', 1)]
2020-05-11 10:40:07,236 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set_on_ms'
2020-05-11 10:40:07,242 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m15) [(b'/output/hall_wall_light/set_off_ms', 1)]
2020-05-11 10:40:07,246 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_wall_light/set_off_ms'
2020-05-11 10:40:07,251 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m16) [(b'/output/hall_led/set', 1)]
2020-05-11 10:40:07,255 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set'
2020-05-11 10:40:07,261 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m17) [(b'/output/hall_led/set_on_ms', 1)]
2020-05-11 10:40:07,265 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set_on_ms'
2020-05-11 10:40:07,270 mqtt_gpio (DEBUG): MQTT client: Sending SUBSCRIBE (d0, m18) [(b'/output/hall_led/set_off_ms', 1)]
2020-05-11 10:40:07,275 mqtt_gpio (INFO): Subscribed to topic: '/output/hall_led/set_off_ms'
[... cut ...]
2020-05-11 10:40:27,639 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 10:40:27,644 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 10:40:27,648 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 10:40:27,652 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 10:40:27,657 mqtt_gpio (DEBUG): MQTT client: Received SUBACK
2020-05-11 10:40:27,661 mqtt_gpio (DEBUG): MQTT client: Received PUBACK (Mid: 129)
2020-05-11 10:40:28,595 mqtt_gpio (INFO): next loop
^CExiting due to KeyboardInterrupt
```

</details>

a-x-, 2020-05-11:
I added a `connected` flag-variable, and I skip now `on_connect` operations if connection was established once previously. 

But it did not help with flickering home assistant due to cyclic pi_mqtt reconnections

a-x-, 2020-05-11:
Also, I added a stack trace (w/o exception) into on_conn if it’s repetitive connection:

```
2020-05-11 11:06:49,626 mqtt_gpio (INFO): next loop
2020-05-11 11:06:50,623 mqtt_gpio (DEBUG): MQTT client: Sending CONNECT (u0, p0, wr1, wq1, wf1, c0, k3600) client_id=b'pi-mqtt-gpio-da39a3ee5e6b4b0d3255bfef95601890afd80709'
2020-05-11 11:06:50,632 mqtt_gpio (DEBUG): MQTT client: Received CONNACK (1, 0)
  File "/usr/lib/python3.5/threading.py", line 882, in _bootstrap
    self._bootstrap_inner()
2020-05-11 11:06:50,643 mqtt_gpio (INFO): next loop
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/paho/mqtt/client.py", line 3428, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/home/pi/.local/lib/python3.5/site-packages/paho/mqtt/client.py", line 1782, in loop_forever
    rc = self.loop(timeout, max_packets)
  File "/home/pi/.local/lib/python3.5/site-packages/paho/mqtt/client.py", line 1177, in loop
    rc = self.loop_read(max_packets)
  File "/home/pi/.local/lib/python3.5/site-packages/paho/mqtt/client.py", line 1568, in loop_read
    rc = self._packet_read()
  File "/home/pi/.local/lib/python3.5/site-packages/paho/mqtt/client.py", line 2315, in _packet_read
    rc = self._packet_handle()
  File "/home/pi/.local/lib/python3.5/site-packages/paho/mqtt/client.py", line 2958, in _packet_handle
    return self._handle_connack()
  File "/home/pi/.local/lib/python3.5/site-packages/paho/mqtt/client.py", line 3046, in _handle_connack
    self, self._userdata, flags_dict, result)
  File "/home/pi/.local/lib/python3.5/site-packages/pi_mqtt_gpio/server.py", line 356, in on_conn
    traceback.print_stack()
```

a-x-, 2020-05-11:
I even upgraded raspbian to latest buster release...

flyte, 2020-05-11:
Looking at the logs and behaviour, it seems like the MQTT connection is dropping for some reason, and the paho-mqtt library is reconnecting by itself and re-running the `on_conn()` function each time.

I'd try using `mosquitto_sub` from the command line to check the MQTT connection independently of pi-mqtt-gpio, then if that works, probably test the `paho-mqtt` library by using some example code from https://pypi.org/project/paho-mqtt/#usage-and-api and see if that suffers from the same issue.

a-x-, 2020-05-11:
mosquitto_sub works good...

and I even tried on another freshly re-burned raspberry pi 3

a-x-, 2020-05-11:
I copy-pasted paho-mqtt test code as is. It repeat something weird:

```

$SYS/broker/uptime b'240594 seconds'
$SYS/broker/load/messages/received/1min b'5.88'
$SYS/broker/load/messages/sent/1min b'105.41'
$SYS/broker/load/publish/received/1min b'0.06'
$SYS/broker/load/publish/sent/1min b'99.70'
$SYS/broker/load/bytes/received/1min b'94.27'
$SYS/broker/load/bytes/sent/1min b'4178.42'
$SYS/broker/load/sockets/1min b'0.63'
$SYS/broker/load/connections/1min b'0.63'
$SYS/broker/load/messages/received/5min b'15.88'
$SYS/broker/load/messages/sent/5min b'46.36'
$SYS/broker/load/publish/sent/5min b'34.12'
$SYS/broker/load/bytes/received/5min b'1985.65'
$SYS/broker/load/bytes/sent/5min b'2849.30'
$SYS/broker/load/sockets/5min b'2.00'
$SYS/broker/load/connections/5min b'2.00'
$SYS/broker/load/messages/received/15min b'16.46'
$SYS/broker/load/messages/sent/15min b'29.36'
$SYS/broker/load/publish/sent/15min b'16.90'
$SYS/broker/load/bytes/received/15min b'2173.72'
$SYS/broker/load/bytes/sent/15min b'2298.21'
$SYS/broker/load/sockets/15min b'1.82'
$SYS/broker/load/connections/15min b'1.82'
$SYS/broker/messages/received b'82275'
$SYS/broker/messages/sent b'97513'
$SYS/broker/publish/messages/sent b'38060'
$SYS/broker/bytes/received b'11461569'
$SYS/broker/bytes/sent b'10080040'
$SYS/broker/publish/bytes/sent b'8189598'
```

a-x-, 2020-05-11:
hm,
$SYS/broker/clients/total b'7'
$SYS/broker/clients/maximum b'8'

looks like client count limit

flyte, 2020-05-11:
It's also worth testing with https://test.mosquitto.org/ to rule out any issues with your Mosquitto server.

BenjiU, 2020-05-11:
No, its a max count, not a limit: SYS/broker/clients/maximum: The maximum number of active clients that have been connected to the broker. This is only calculated when the $SYS topic tree is updated, so short lived client connections may not be counted.

a-x-, 2020-05-11:
WOW, I stopped pi-mqtt-gpio on another raspberry in my lan
And than pi-mqtt-gpio successfully ran on this rpi!

So, finnaly, bug sounds like that: “Multiple instances of pi-mqtt-gpio in the same lAN cannot be runned”

flyte, 2020-05-11:
Ahh, right. This will be an issue with the client ID then. It's based on the `topic_prefix` config value. It wouldn't make sense to run two instances on the same `topic_prefix` (or have you found a valid use case for that?), so they should be different between your two Pis.

flyte, 2020-05-11:
I see that you're not actually using a `topic_prefix` setting. We'll need to make this potential gotcha clear in the readme.

flyte, 2020-05-11:
See the first two examples of configuration on the readme, and how it affects the MQTT topics used. Hopefully that'll explain what `topic_prefix` does.

a-x-, 2020-05-11:
Can I safely use topic_prefix with discovery option?

I used uniq names of inputs/outputs and I did think it enough

a-x-, 2020-05-11:
Looks like topic_prefix breaks `discovery`

a-x-, 2020-05-11:
> Can I safely use topic_prefix with discovery option?

Yes, but, names must be unique too (include prefix again)!

a-x-, 2020-05-11:
Thank you, guys!

closed 2020-05-11 by a-x-

flyte, 2020-05-11:
Glad you got it working! Thank for updating the ticket with your findings.
We'll have a look and see what we can do to avoid people hitting the same
problem in the future.