flyte/mqtt-io #105: Seems pi-mqtt-gpio exclusively acquires pins

a-x-, 2020-05-13:
I have other software on raspberry, that uses pins too.
For instance, I have input-pin with pir motion sensor.

When I installed pi-mqtt-gpio, my soft does not receive interruptions from the pir pin more.

Now, I cannot find: is it true, that linux does not allow multiple interruption listeners

BenjiU, 2020-05-18:
Do you use the same pin in different applications? This might not work. But different Pins in different applications should work, i guess.

flyte, 2020-05-18:
I'd have to double check, but I seem to recall only one instance of the RPi.GPIO library being able to use the GPIO pins at a given moment. If the underlying library doesn't support concurrent access to the GPIO device, then there's not a lot we can do really. Happy to be proven wrong though!

flyte, 2020-05-30:
[This article](https://www.wbga.ca/14/) suggests that multiple processes should be able to access the RPi GPIO simultaneously, so I'll add this to the list of things to review before the next release.

flyte, 2020-05-30:
Hmm, I've tested this out using two instances of pi-mqtt-gpio on the same pi, with different MQTT client IDs, configured to look for the same pin interrupt and it works fine. I get both interrupts reported:

```
# I sent this message to turn on an output
pimqttgpio/test/output/jumper_output/set ON

# Both instances responded to say that the output had turned on
pimqttgpio/test/output/jumper_output ON
pimqttgpio/test/output/jumper_output ON

# Both instances received and published the interrupt on the input pin that's
# electrically connected to the output pin with a jumper
pimqttgpio/test/input/jumper_input trigger
pimqttgpio/test/input/jumper_input trigger
```

I can't speak for your other application, @a-x- but it seems like this one is able to share interrupts at the same time. I'll close this for now, but I'd be happy to see evidence to the contrary , if you can show us.

closed 2020-05-30 by flyte