flyte/mqtt-io #106: crash on OrangePi zero

llevet, 2020-05-13:
I try to start on Orange Pi zero but it crash at pin configuration.

OS : Armbian_20.02.1_Orangepizero_buster_current_5.4.20

pip install pi-mqtt-gpio  -> OK
apt-get install python-dev  -> OK
pip install OrangePi.GPIO  -> OK
pip install OPi.GPIO -> OK

Content of pi-mqtt-gpio.yml :
mqtt:
  host: 172.31.31.44
  port: 1883
  user: ""
  password: ""
  topic_prefix: pimqttopz1/kitchen

gpio_modules:
  - name: orangepi
    module: orangepi
    board: zero
    mode: board

digital_outputs:
  - name: lights
    module: orangepi
    pin: 7
    on_payload: "ON"
    off_payload: "OFF"
    initial: low
    retain: yes

python -m pi_mqtt_gpio.server pi-mqtt-gpio.yml

2020-05-13 22:00:20,382 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 887, in <module>
    main(args)
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 759, in main
    GPIO_MODULES[gpio_config["name"]] = configure_gpio_module(gpio_config)
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 427, in configure_gpio_module
    gpio_module = import_module("pi_mqtt_gpio.modules.%s" % gpio_config["module"])
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/modules/orangepi.py", line 66
    raise IOError("channel %d setup failed" % pin) from e
                                                      ^

llevet, 2020-05-14:
Ok find the problem.

First remove package OPi.GPIO because is in interaction with OrangePi.GPIO.

Second : in pi_mqtt_gpio/modules/orangepi.py

replace line 66 : 
raise IOError("channel %d setup failed" % pin) from e
by line : 
raise IOError("channel %d setup failed" % pin)

And it is working.

flyte, 2020-05-14:
Hmm, that `raise from` should be valid code. Could you let me know what version of python you're running please?

llevet, 2020-05-14:
Python 2.7.16

llevet, 2020-05-14:
Error was : 

  File "/home/llevet/ve/lib/python2.7/site-packages/pi_mqtt_gpio/modules/orangepi.py", line 66
    raise IOError("channel %d setup failed" % pin) from e
                                                      ^
SyntaxError: invalid syntax

llevet, 2020-05-14:
the ^ is on "from"

flyte, 2020-05-14:
Ah yes, it's python 3 syntax. Best to remove that for now then while we're still supporting it. I recommend changing to python 3 though if you can. 2.x is no longer supported in general now.

llevet, 2020-05-14:
Yes i agree with you for all.
Good choice to remove it at this time because the default Raspbian Buster for Pi and for OrangePi at this yime is deliver by default with python 2.7.16

Maybe a note in Readme file to inform that is beter to use python 3 and adapte all syntax with it.
ex: pip install pi-mqtt-gpio -> pip3 install pi-mqtt-gpio ...

commit ebb6e93f (2020-05-30): Use exception chaining in a Python 2 compatible way. Closes #106
closed 2020-05-30 by flyte