flyte/mqtt-io #107: raspberrypi: RuntimeError: Error accessing GPIO.

gitmirko, 2020-05-17:
Hi,
I'm struggling with getting the docker container to access the GPIO. I'm running a raspberry pi 3B+ with raspbian. I've attached a AM2302 on pin 4 and configured pi-mqtt-gpio as follow:
```
mqtt:
  host: 10.0.0.10
  port: 8883
  tls:
    enabled: yes
    cert_reqs: CERT_NONE
    insecure: yes
  user: "rpi"
  password: "FXHucuGYBy7kHR3MtN"
  topic_prefix: home/wohnzimmer

gpio_modules:
  - name: raspberrypi
    module: raspberrypi
    cleanup: no

sensor_modules:
  - name: dht22
    module: dht22
    type: AM2302 # can be  DHT11, DHT22 or AM2302
    pin: 4

sensor_inputs:
  - name: dht22_temperature
    module: dht22
    interval: 4 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 2 # number of digits to be round
    type: temperature # Can be temperature or humidity

  - name: dht22_humidity
    module: dht22
    interval: 4 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 2 # number of digits to be round
    type: humidity # Can be temperature or humidity
logging:
  version: 1
  formatters:
    simple:
      format: "%(asctime)s %(name)s (%(levelname)s): %(message)s"
  handlers:
    console:
      class: logging.StreamHandler
      level: DEBUG
      formatter: simple
      stream: ext://sys.stdout
  loggers:
    mqtt_gpio:
      level: DEBUG
      handlers: [console]
      propagate: yes
```
This is the output of the container and the log (I've killed the container with CTRL-C):
```
pi@raspberrypi:~ $ docker run --name="mqtt-gpio" -e GROUP_ID=997 --privileged -v /sys:/sys -v /home/pi/docker/mqtt-gpio/config.yml:/config.yml -v /etc/localtime:/etc/localtime:ro flyte/mqtt-gpio:armv7l
2020-05-17 23:05:06,062 mqtt_gpio (INFO): Startup
2020-05-17 23:05:06,738 mqtt_gpio (DEBUG): Last will set on 'home/wohnzimmer/status' as 'dead'.
Collecting RPi.GPIO
  Downloading RPi.GPIO-0.7.0.tar.gz (30 kB)
Building wheels for collected packages: RPi.GPIO
  Building wheel for RPi.GPIO (setup.py): started
  Building wheel for RPi.GPIO (setup.py): finished with status 'done'
  Created wheel for RPi.GPIO: filename=RPi.GPIO-0.7.0-cp36-cp36m-linux_armv7l.whl size=61774 sha256=f80cdc287689b7badb02dd13fc9207db797bec8e42c739dedf000cb47f2c4a66
  Stored in directory: /home/mqttgpio/.cache/pip/wheels/84/c0/d6/73a6e9f98b9d29edd7c6e981dce47852c3321f34518c80dce1
Successfully built RPi.GPIO
Installing collected packages: RPi.GPIO
Successfully installed RPi.GPIO-0.7.0
Collecting Adafruit_DHT
  Downloading Adafruit_DHT-1.4.0.tar.gz (15 kB)
Building wheels for collected packages: Adafruit-DHT
  Building wheel for Adafruit-DHT (setup.py): started
  Building wheel for Adafruit-DHT (setup.py): finished with status 'done'
  Created wheel for Adafruit-DHT: filename=Adafruit_DHT-1.4.0-cp36-cp36m-linux_armv7l.whl size=30318 sha256=f2113af4b4610a647e995f6293a05f882b74f0e3d6cd87b810dcdb747dfe8350
  Stored in directory: /home/mqttgpio/.cache/pip/wheels/af/d8/00/6f0c819ab5f38ea5ececf1b698a9e53888e271300d5f202260
Successfully built Adafruit-DHT
Installing collected packages: Adafruit-DHT
Successfully installed Adafruit-DHT-1.4.0
2020-05-17 23:05:38,608 mqtt_gpio (DEBUG): MQTT client: Sending CONNECT (u1, p1, wr1, wq1, wf1, c0, k60) client_id=b'pi-mqtt-gpio-cd95fea6685fdba5f9bbd3d2d4b76d9db771cfc2'
2020-05-17 23:05:38,611 mqtt_gpio (DEBUG): sensor_timer_thread: calculated cycle_time will be 4 seconds
2020-05-17 23:05:38,612 mqtt_gpio (DEBUG): MQTT client: Received CONNACK (1, 0)
2020-05-17 23:05:38,613 mqtt_gpio (INFO): Connected to the MQTT broker with protocol v3.1.1.
2020-05-17 23:05:38,613 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q1, r1, m1), 'b'home/wohnzimmer/status'', ... (7 bytes)
2020-05-17 23:05:38,619 mqtt_gpio (DEBUG): MQTT client: Received PUBACK (Mid: 1)
2020-05-17 23:06:38,700 mqtt_gpio (DEBUG): MQTT client: Sending PINGREQ
2020-05-17 23:06:38,704 mqtt_gpio (DEBUG): MQTT client: Received PINGRESP
2020-05-17 23:07:38,788 mqtt_gpio (DEBUG): MQTT client: Sending PINGREQ
2020-05-17 23:07:38,791 mqtt_gpio (DEBUG): MQTT client: Received PINGRESP
^C
2020-05-17 23:08:13,603 mqtt_gpio (DEBUG): MQTT client: Sending PUBLISH (d0, q1, r1, m2), 'b'home/wohnzimmer/status'', ... (7 bytes)
2020-05-17 23:08:13,610 mqtt_gpio (DEBUG): MQTT client: Received PUBACK (Mid: 2)
2020-05-17 23:08:13,613 mqtt_gpio (DEBUG): MQTT client: Sending DISCONNECT
2020-05-17 23:08:13,616 mqtt_gpio (INFO): Cleanup disabled for module 'raspberrypi'.
Exception in thread pi-mqtt-gpio_SensorReader:
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 612, in sensor_timer_thread
    value = sensor.get_value(sens_conf)
  File "/home/mqttgpio/pi_mqtt_gpio/modules/dht22.py", line 49, in get_value
    humidity, temperature = self.sensor.read_retry(self.sensor_type, self.pin)
  File "/home/mqttgpio/.local/share/virtualenvs/mqttgpio-815CbdL_/lib/python3.6/site-packages/Adafruit_DHT/common.py", line 94, in read_retry
    humidity, temperature = read(sensor, pin, platform)
  File "/home/mqttgpio/.local/share/virtualenvs/mqttgpio-815CbdL_/lib/python3.6/site-packages/Adafruit_DHT/common.py", line 81, in read
    return platform.read(sensor, pin)
  File "/home/mqttgpio/.local/share/virtualenvs/mqttgpio-815CbdL_/lib/python3.6/site-packages/Adafruit_DHT/Raspberry_Pi_2.py", line 34, in read
    raise RuntimeError('Error accessing GPIO.')
RuntimeError: Error accessing GPIO.
```
I'm out of ideas. Could someone help me out?

Best regards
Mirko

BenjiU, 2020-05-18:
Hi, you do have the problem with all gpios or only via dht22/Adafruit lib?

gitmirko, 2020-05-18:
I've just tried adding an normal pin as output, but even with passing an additional device I get another kind of error:
```
[...]
digital_outputs:
  - name: led
    module: raspberrypi
    pin: 18
    on_payload: "ON"
    off_payload: "OFF"
    initial: low
[...]
```

```
pi@raspberrypi:~ $ docker run --name="mqtt-gpio" --privileged --device /dev/mem --device /dev/gpiomem --device /dev/i2c-1 -v /home/pi/docker/mqtt-gpio/config.yml:/config.yml -v /etc/localtime:/etc/localtime:ro flyte/mqtt-gpio:armv7l
2020-05-18 22:16:37,346 mqtt_gpio (INFO): Startup
2020-05-18 22:16:38,021 mqtt_gpio (DEBUG): Last will set on 'home/wohnzimmer/status' as 'dead'.
Collecting RPi.GPIO
  Downloading RPi.GPIO-0.7.0.tar.gz (30 kB)
Building wheels for collected packages: RPi.GPIO
  Building wheel for RPi.GPIO (setup.py): started
  Building wheel for RPi.GPIO (setup.py): finished with status 'done'
  Created wheel for RPi.GPIO: filename=RPi.GPIO-0.7.0-cp36-cp36m-linux_armv7l.whl size=61793 sha256=66134c6ff5dc2712bbc79535676f79730c3aa32feb47caf5ef6d5e9176f65b42
  Stored in directory: /home/mqttgpio/.cache/pip/wheels/84/c0/d6/73a6e9f98b9d29edd7c6e981dce47852c3321f34518c80dce1
Successfully built RPi.GPIO
Installing collected packages: RPi.GPIO
Successfully installed RPi.GPIO-0.7.0
Collecting Adafruit_DHT
  Downloading Adafruit_DHT-1.4.0.tar.gz (15 kB)
Building wheels for collected packages: Adafruit-DHT
  Building wheel for Adafruit-DHT (setup.py): started
  Building wheel for Adafruit-DHT (setup.py): finished with status 'done'
  Created wheel for Adafruit-DHT: filename=Adafruit_DHT-1.4.0-cp36-cp36m-linux_armv7l.whl size=30317 sha256=5d34264809782db299aa4fe389928e961591baea99f64ccc20c976e68ef4090d
  Stored in directory: /home/mqttgpio/.cache/pip/wheels/af/d8/00/6f0c819ab5f38ea5ececf1b698a9e53888e271300d5f202260
Successfully built Adafruit-DHT
Installing collected packages: Adafruit-DHT
Successfully installed Adafruit-DHT-1.4.0
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 901, in <module>
    main(args)
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 802, in main
    initialise_digital_output(out_conf, GPIO_MODULES[out_conf["module"]])
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 540, in initialise_digital_output
    gpio.setup_pin(out_conf["pin"], PinDirection.OUTPUT, None, out_conf)
  File "/home/mqttgpio/pi_mqtt_gpio/modules/raspberrypi.py", line 47, in setup_pin
    self.io.setup(pin, direction, pull_up_down=pullup, initial=initial)
RuntimeError: No access to /dev/mem.  Try running as root!
```

BenjiU, 2020-05-19:
> docker run --name="mqtt-gpio" --privileged --device /dev/mem --device /dev/gpiomem 

https://docs.docker.com/engine/reference/run/#runtime-privilege-and-linux-capabilities says, that you can use --device, if you don't want to use --privileged. May you check one without the other: just --privileged or --device=?
syntax on help page, looks different:
`docker run --device=/dev/sda:/dev/xvdc`

gitmirko, 2020-05-19:
I've just tried only with _--privileged_ and another try specifying the devices with the syntax from the linked documentation. No luck at all:
```
root@raspberrypi:~# docker run --name="mqtt-gpio" --privileged -v /home/pi/docker/mqtt-gpio/config.yml:/config.yml -v /etc/localtime:/etc/localtime:ro flyte/mqtt-gpio:armv7l
2020-05-19 22:21:23,774 mqtt_gpio (INFO): Startup
2020-05-19 22:21:25,507 mqtt_gpio (DEBUG): Last will set on 'home/wohnzimmer/status' as 'dead'.
Collecting RPi.GPIO
  Downloading RPi.GPIO-0.7.0.tar.gz (30 kB)
Building wheels for collected packages: RPi.GPIO
  Building wheel for RPi.GPIO (setup.py): started
  Building wheel for RPi.GPIO (setup.py): finished with status 'done'
  Created wheel for RPi.GPIO: filename=RPi.GPIO-0.7.0-cp36-cp36m-linux_armv7l.whl size=61764 sha256=250e4ed1ce647e6958e2f2851eec2bc97ba6cf26832b60139fe9e9c84fb05461
  Stored in directory: /home/mqttgpio/.cache/pip/wheels/84/c0/d6/73a6e9f98b9d29edd7c6e981dce47852c3321f34518c80dce1
Successfully built RPi.GPIO
Installing collected packages: RPi.GPIO
Successfully installed RPi.GPIO-0.7.0
Collecting Adafruit_DHT
  Downloading Adafruit_DHT-1.4.0.tar.gz (15 kB)
Building wheels for collected packages: Adafruit-DHT
  Building wheel for Adafruit-DHT (setup.py): started
  Building wheel for Adafruit-DHT (setup.py): finished with status 'done'
  Created wheel for Adafruit-DHT: filename=Adafruit_DHT-1.4.0-cp36-cp36m-linux_armv7l.whl size=30333 sha256=3f7313bdb7fae4f2503c69df3377a3088bfb5275940ebb379ea4be24896c2888
  Stored in directory: /home/mqttgpio/.cache/pip/wheels/af/d8/00/6f0c819ab5f38ea5ececf1b698a9e53888e271300d5f202260
Successfully built Adafruit-DHT
Installing collected packages: Adafruit-DHT
Successfully installed Adafruit-DHT-1.4.0
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 901, in <module>
    main(args)
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 802, in main
    initialise_digital_output(out_conf, GPIO_MODULES[out_conf["module"]])
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 540, in initialise_digital_output
    gpio.setup_pin(out_conf["pin"], PinDirection.OUTPUT, None, out_conf)
  File "/home/mqttgpio/pi_mqtt_gpio/modules/raspberrypi.py", line 47, in setup_pin
    self.io.setup(pin, direction, pull_up_down=pullup, initial=initial)
RuntimeError: No access to /dev/mem.  Try running as root!
```

```
root@raspberrypi:~# docker run --name="mqtt-gpio" --device /dev/mem:/dev/mem --device /dev/gpiomem:/dev/gpiomem --device /dev/i2c-1:/dev/i2c-1 -v /home/pi/docker/mqtt-gpio/config.yml:/config.yml -v /etc/localtime:/etc/localtime:ro flyte/mqtt-gpio:armv7l
2020-05-19 22:32:26,060 mqtt_gpio (INFO): Startup
2020-05-19 22:32:27,430 mqtt_gpio (DEBUG): Last will set on 'home/wohnzimmer/status' as 'dead'.
Collecting RPi.GPIO
  Downloading RPi.GPIO-0.7.0.tar.gz (30 kB)
Building wheels for collected packages: RPi.GPIO
  Building wheel for RPi.GPIO (setup.py): started
  Building wheel for RPi.GPIO (setup.py): finished with status 'done'
  Created wheel for RPi.GPIO: filename=RPi.GPIO-0.7.0-cp36-cp36m-linux_armv7l.whl size=61783 sha256=2b09aa35b6c22e49825b250cf1a309918dc1139e3bfc5efd42c977aa252f01de
  Stored in directory: /home/mqttgpio/.cache/pip/wheels/84/c0/d6/73a6e9f98b9d29edd7c6e981dce47852c3321f34518c80dce1
Successfully built RPi.GPIO
Installing collected packages: RPi.GPIO
Successfully installed RPi.GPIO-0.7.0
Collecting Adafruit_DHT
  Downloading Adafruit_DHT-1.4.0.tar.gz (15 kB)
Building wheels for collected packages: Adafruit-DHT
  Building wheel for Adafruit-DHT (setup.py): started
  Building wheel for Adafruit-DHT (setup.py): finished with status 'done'
  Created wheel for Adafruit-DHT: filename=Adafruit_DHT-1.4.0-cp36-cp36m-linux_armv7l.whl size=30320 sha256=35500c891345e234538200a35ed94e0b40411c7a3fc94efad754f3310c56f02a
  Stored in directory: /home/mqttgpio/.cache/pip/wheels/af/d8/00/6f0c819ab5f38ea5ececf1b698a9e53888e271300d5f202260
Successfully built Adafruit-DHT
Installing collected packages: Adafruit-DHT
Successfully installed Adafruit-DHT-1.4.0
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 901, in <module>
    main(args)
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 802, in main
    initialise_digital_output(out_conf, GPIO_MODULES[out_conf["module"]])
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 540, in initialise_digital_output
    gpio.setup_pin(out_conf["pin"], PinDirection.OUTPUT, None, out_conf)
  File "/home/mqttgpio/pi_mqtt_gpio/modules/raspberrypi.py", line 47, in setup_pin
    self.io.setup(pin, direction, pull_up_down=pullup, initial=initial)
RuntimeError: No access to /dev/mem.  Try running as root!
```

IlmLV, 2020-06-17:
> I've just tried only with _--privileged_ and another try specifying the devices with the syntax from the linked documentation. No luck at all:
>
> ```
> root@raspberrypi:~# docker run --name="mqtt-gpio" [...]

same issue for me

SaturnusDJ, 2020-07-09:
https://hub.docker.com/r/ebekebe/mqtt-gpio/ runs.

overlordpy, 2020-07-29:
I am having a similar issue but I am trying to run a python script with a php file. Whenever I try to run it i get RuntimeError: No access to /dev/mem. Try running as root!. I am using apache2 to run a webserver and i have setup then index.php file to run a simple python. Anyone know how to solve this error.

mchmielarz, 2022-11-02:
Hi, I have the same error, however, it's for the newest version of mqtt_io add-on:
```
Add-on: MQTT IO
 Expose GPIO modules and digital sensors via MQTT for remote control and monitoring.
-----------------------------------------------------------
 Add-on version: 0.1.3
 You are running the latest version of this add-on.
 System: Home Assistant OS 9.3  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2022.10.5
 Home Assistant Supervisor: 2022.10.0```

I'm using this config:
```
mqtt:
  topic_prefix: home/depot/climate
  ha_discovery:
    enabled: yes

sensor_modules:
  - name: dht_depot
    module: dht22
    type: DHT22
    pin: 4

sensor_inputs:
  - name: depot_temp
    module: dht_depot
    interval: 10
    digits: 4
    type: temperature
    ha_discovery:
      name: Depot temperature
      device_class: temperature
  - name: depot_humidity
    module: dht_depot
    interval: 10
    digits: 4
    type: humidity
    ha_discovery:
       name: Depot humidity
       device_class: humidity
```

And here is the log:
```2022-11-02 20:37:11 mqtt_io.server [INFO] Connecting to MQTT...
2022-11-02 20:37:12 mqtt_io.server [INFO] Connected to MQTT
2022-11-02 20:37:21 mqtt_io.server [ERROR] Exception when retrieving value from sensor 'depot_temp':
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/mqtt_io/server.py", line 561, in poll_sensor
    value = await get_sensor_value()
  File "/usr/lib/python3.10/site-packages/backoff/_async.py", line 133, in retry
    ret = await target(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/backoff/_async.py", line 66, in retry
    ret = await target(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/mqtt_io/server.py", line 556, in get_sensor_value
    return await sensor_module.async_get_value(sens_conf)
  File "/usr/lib/python3.10/site-packages/mqtt_io/modules/sensor/__init__.py", line 57, in async_get_value
    return await loop.run_in_executor(ThreadPoolExecutor(), self.get_value, sens_conf)
  File "/usr/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/lib/python3.10/site-packages/mqtt_io/modules/sensor/dht22.py", line 62, in get_value
    humidity, temperature = self.sensor.read_retry(self.sensor_type, self.pin)
  File "/usr/lib/python3.10/site-packages/Adafruit_DHT/common.py", line 94, in read_retry
    humidity, temperature = read(sensor, pin, platform)
  File "/usr/lib/python3.10/site-packages/Adafruit_DHT/common.py", line 81, in read
    return platform.read(sensor, pin)
  File "/usr/lib/python3.10/site-packages/Adafruit_DHT/Raspberry_Pi_2.py", line 34, in read
    raise RuntimeError('Error accessing GPIO.')
RuntimeError: Error accessing GPIO.```

Is it something wrong with my config? I have tried two different DHT22 devices connected in the same way

BenjiU, 2022-11-03:
How do you execute it? Looks like a privilege problem!

mchmielarz, 2022-11-03:
I simply added the add-on to the HA and clicked the run button. I'm using HASSIO so I assumed it runs with full access to everything.

mchmielarz, 2022-11-03:
So I would have to install it with other distribution and then install HA and mqtt.io separately. Thanks for help! I'll back with my findings here

BenjiU, 2022-11-04:
@mchmielarz i will close this issue, because it was originally related to docker (2020). Please create a new issue or use #293 
Thanks!

closed 2022-11-04 by BenjiU