flyte/mqtt-io #108: raspberrypi no function on GPIO output

neocit, 2020-05-18:
I have installed like described on a RaspberryPi 3 B+ with a extension relay on the pi.

Installation
pip install pi-mqtt-gpio
Execution
python -m pi_mqtt_gpio.server config.yml

My Config yml:
mqtt:
  host: test.mosquitto.org
  port: 1883
  user: ""
  password: ""
  topic_prefix: HG-Pool/MQTT
  status_topic: status

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_outputs:
  - name: PoolBeleuchtung
    module: raspberrypi
    pin: 11
    on_payload: "ON"
    off_payload: "OFF"

  - name: test
    module: raspberrypi
    pin: 37
    on_payload: "ON"
    off_payload: "OFF"
    initial: low
    retain: yes

  - name: test2
    module: raspberrypi
    pin: 38
    on_payload: "ON"
    off_payload: "OFF"
    initial: high
    retain: yes

  - name: PoolPumpe
    module: raspberrypi
    pin: 16
    on_payload: "ON"
    off_payload: "OFF"

  - name: SolarVentil
    module: raspberrypi
    pin: 22
    on_payload: "ON"
    off_payload: "OFF"

  - name: Gartenbewaesserung
    module: raspberrypi
    pin: 15
    on_payload: "ON"
    off_payload: "OFF"

  - name: GPIO21
    module: raspberrypi
    pin: 40
    on_payload: "ON"
    off_payload: "OFF"
 
digital_inputs:
  - name: test3
    module: raspberrypi
    pin: 37  # This device is connected to pin 21 of the Raspberry PI GPIO
    on_payload: "ON"
    off_payload: "OFF"
    pullup: no
    pulldown: yes

logging:
  version: 1
  formatters:
    simple:
      format: "%(asctime)s %(name)s (%(levelname)s): %(message)s"
  handlers:
    console:
      class: logging.StreamHandler
      level: DEBUG
      formatter: simple
      stream: ext://sys.stdout
  loggers:
    mqtt_gpio:
      level: INFO
      handlers: [console]
      propagate: yes

Terminal Output:
Installation

pi@RaspberryXXXX:~ $ python -m pi_mqtt_gpio.server config.yml
2020-05-18 09:22:08,343 mqtt_gpio (INFO): Startup
/home/pi/.local/lib/python2.7/site-packages/pi_mqtt_gpio/modules/raspberrypi.py:47: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self.io.setup(pin, direction, pull_up_down=pullup, initial=initial)
2020-05-18 09:22:09,112 mqtt_gpio (INFO): Polling: Input 'test3' state changed to False
2020-05-18 09:22:09,123 mqtt_gpio (INFO): Connected to the MQTT broker with protocol v3.1.1.
2020-05-18 09:22:09,124 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/PoolBeleuchtung/set'
2020-05-18 09:22:09,125 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/PoolBeleuchtung/set_on_ms'
2020-05-18 09:22:09,126 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/PoolBeleuchtung/set_off_ms'
2020-05-18 09:22:09,127 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/test/set'
2020-05-18 09:22:09,128 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/test/set_on_ms'
2020-05-18 09:22:09,129 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/test/set_off_ms'
2020-05-18 09:22:09,130 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/test2/set'
2020-05-18 09:22:09,130 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/test2/set_on_ms'
2020-05-18 09:22:09,131 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/test2/set_off_ms'
2020-05-18 09:22:09,131 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/PoolPumpe/set'
2020-05-18 09:22:09,132 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/PoolPumpe/set_on_ms'
2020-05-18 09:22:09,133 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/PoolPumpe/set_off_ms'
2020-05-18 09:22:09,133 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/SolarVentil/set'
2020-05-18 09:22:09,134 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/SolarVentil/set_on_ms'
2020-05-18 09:22:09,135 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/SolarVentil/set_off_ms'
2020-05-18 09:22:09,135 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/Gartenbewaesserung/set'
2020-05-18 09:22:09,136 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/Gartenbewaesserung/set_on_ms'
2020-05-18 09:22:09,136 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/Gartenbewaesserung/set_off_ms'
2020-05-18 09:22:09,137 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/GPIO21/set'
2020-05-18 09:22:09,137 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/GPIO21/set_on_ms'
2020-05-18 09:22:09,137 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/GPIO21/set_off_ms'
2020-05-18 09:22:34,377 mqtt_gpio (INFO): Received message on topic u'HG-Pool/MQTT/output/test/set': 'on'
2020-05-18 09:22:34,378 mqtt_gpio (WARNING): Payload u'on' does not relate to configured on/off values 'ON' and 'OFF'
2020-05-18 09:22:42,449 mqtt_gpio (INFO): Received message on topic u'HG-Pool/MQTT/output/test/set': 'OFF'
2020-05-18 09:22:42,450 mqtt_gpio (INFO): Set 'raspberrypi' output 'test' to False
2020-05-18 09:22:50,220 mqtt_gpio (INFO): Received message on topic u'HG-Pool/MQTT/output/test/set': 'ON'
2020-05-18 09:22:50,221 mqtt_gpio (INFO): Set 'raspberrypi' output 'test' to True
2020-05-18 09:22:50,247 mqtt_gpio (INFO): Polling: Input 'test3' state changed to True

Seems that the MQTT was working but on the GPIO nothing is changing, like they are not active.

Or have i made something wrong?

Thanks neocity

BenjiU, 2020-05-18:
How do you check the status of the output pin? How does your hardware extension looks like?

neocit, 2020-05-18:
I have a Rpi Relay Board with 3 relais with LED state. // testet with Thingsboard they will be active.
The Relay ist this one : https://www.pi-shop.ch/rpi-relay-board
and its connected with the 40pin's on top of the raspiboard. 
Raspi is connected with LAN and hdmi. Raspian Buster latest Version.

BenjiU, 2020-05-18:
I think you used the wrong pin numbering. We use BCM numbering: gpio.setmode(gpio.BCM)

Channel No. | RPi Pin No. | wiringPi | BCM | Descriptions
-- | -- | -- | -- | --
CH1 | 37 | P25 | 26 | Channel 1
CH2 | 38 | P28 | 20 | Channel 2
CH3 | 40 | P29 | 21 | Channel 3

So you should use 20, 21 or 26 instead of 37,38 and 40.

neocit, 2020-05-18:
Many Thanks. That work!
Not seen in Github description.

closed 2020-05-18 by neocit