flyte/mqtt-io #109: Add more than one DS18B20 not possible?!

neocit, 2020-05-19:
First of all great Projekt and worked great. Thanks

Now i need support or help to install my DS18B20 temp sensor's.
I have tried with the following config in yml.

`sensor_modules:
  - name: Luft
    module: ds18b
    type: DS18B20
    address: 0516c00797ff

#  - name: Wasser
#    module: ds18b
#    type: DS18B20
#    address: 0416c25554ff

  - name: Solar
    module: ds18b
    type: DS18B20
    address: 0416c1d8e7ff

  - name: Einlauf
    module: ds18b
    type: DS18B20
    address: 0416c24081ff

  - name: Ruecklauf
    module: ds18b
    type: DS18B20
    address: 0416c23c08ff

sensor_inputs:
  - name: Luft
    module: ds18b
    interval: 60
    digits: 2

  - name: Wasser
    module: ds18b
    interval: 60
    digits: 2

  - name: Solar
    module: ds18b
    interval: 60
    digits: 2

  - name: Einlauf
    module: ds18b
    interval: 60
    digits: 2

  - name: Ruecklauf
    module: ds18b
    interval: 60
    digits: 2`

ERROR: pi@RaspberryXXXX:~ $ python -m pi_mqtt_gpio.server config.yml
2020-05-19 12:57:52,008 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/pi/.local/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 887, in <module>
    main(args)
  File "/home/pi/.local/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 793, in main
    sens_conf
  File "/home/pi/.local/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 538, in validate_sensor_input_config
    "pi_mqtt_gpio.modules.%s" % SENSOR_CONFIGS[sens_conf["module"]]["module"]
KeyError: 'ds18b'

Tried with the following yml:
`sensor_modules:
  - name: ds18b
    module: ds18b
    type: DS18B20
    address: 0516c00797ff

#  - name: ds18b
#    module: ds18b
#    type: DS18B20
#    address: 0416c25554ff

  - name: ds18b
    module: ds18b
    type: DS18B20
    address: 0416c1d8e7ff

  - name: ds18b
    module: ds18b
    type: DS18B20
    address: 0416c24081ff

  - name: ds18b
    module: ds18b
    type: DS18B20
    address: 0416c23c08ff

sensor_inputs:
  - name: Luft
    module: ds18b
    interval: 60
    digits: 2

  - name: Wasser
    module: ds18b
    interval: 60
    digits: 2

  - name: Solar
    module: ds18b
    interval: 60
    digits: 2

  - name: Einlauf
    module: ds18b
    interval: 60
    digits: 2

  - name: Ruecklauf
    module: ds18b
    interval: 60
    digits: 2`

OUTPUT: pi@RaspberryXXXXl:~ $ python -m pi_mqtt_gpio.server config.yml
2020-05-19 13:16:49,498 mqtt_gpio (INFO): Startup
/home/pi/.local/lib/python2.7/site-packages/pi_mqtt_gpio/modules/raspberrypi.py:47: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self.io.setup(pin, direction, pull_up_down=pullup, initial=initial)
2020-05-19 13:16:52,871 mqtt_gpio (INFO): Polling: Input 'PoolBeleuchtung' state changed to False
2020-05-19 13:16:52,884 mqtt_gpio (INFO): Connected to the MQTT broker with protocol v3.1.1.
2020-05-19 13:16:52,886 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/PoolBeleuchtung/set'
2020-05-19 13:16:52,888 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/PoolBeleuchtung/set_on_ms'
2020-05-19 13:16:52,890 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/PoolBeleuchtung/set_off_ms'
2020-05-19 13:16:52,892 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/PoolPumpe/set'
2020-05-19 13:16:52,894 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/PoolPumpe/set_on_ms'
2020-05-19 13:16:52,896 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/PoolPumpe/set_off_ms'
2020-05-19 13:16:52,898 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/SolarVentil/set'
2020-05-19 13:16:52,899 mqtt_gpio (INFO): Polling: Input 'PoolPumpe' state changed to False
2020-05-19 13:16:52,902 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/SolarVentil/set_on_ms'
2020-05-19 13:16:52,905 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/SolarVentil/set_off_ms'
2020-05-19 13:16:52,908 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/Gartenbewaesserung/set'
2020-05-19 13:16:52,910 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/Gartenbewaesserung/set_on_ms'
2020-05-19 13:16:52,913 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/Gartenbewaesserung/set_off_ms'
2020-05-19 13:16:52,915 mqtt_gpio (INFO): Polling: Input 'SolarVentil' state changed to False
2020-05-19 13:16:52,916 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/Manuell/set'
2020-05-19 13:16:52,920 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/Manuell/set_on_ms'
2020-05-19 13:16:52,922 mqtt_gpio (INFO): Subscribed to topic: 'HG-Pool/MQTT/output/Manuell/set_off_ms'
2020-05-19 13:16:52,928 mqtt_gpio (INFO): Polling: Input 'Gartenbewaesserung' state changed to False
2020-05-19 13:16:52,955 mqtt_gpio (INFO): Polling: Input 'Manuell' state changed to False
2020-05-19 13:16:53,830 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'Wasser' value 18.63
2020-05-19 13:16:54,710 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'Einlauf' value 18.63
2020-05-19 13:16:55,590 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'Ruecklauf' value 18.63
2020-05-19 13:16:56,470 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'Solar' value 18.63
2020-05-19 13:16:57,351 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'Luft' value 18.63
2020-05-19 13:17:53,831 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'Wasser' value 18.63
2020-05-19 13:17:54,714 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'Einlauf' value 18.63
2020-05-19 13:17:55,682 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'Ruecklauf' value 18.63
2020-05-19 13:17:57,520 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'Solar' value 18.63
2020-05-19 13:17:58,470 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'Luft' value 18.63

Worked, but only 1 Sensor will bea read for all sensor_inputs.

How can i match the sensor id with the sensor name?

Thanks for your help.

Neocity

BenjiU, 2020-05-19:
Hi, something is mixed up in your config.yml. This works for me:
```
sensor_modules:
  - name: ds18b22_1
    module: ds18b
    type: DS18S20
    address: 000803702352

  - name: ds18b22_2
    module: ds18b
    type: DS18S20
    address: 000803742764

  - name: ds18b22_3
    module: ds18b
    type: DS18S20
    address: 000803747ef7

  - name: ds18b22_4
    module: ds18b
    type: DS18S20
    address: 00080374db56

  - name: ds18b22_5
    module: ds18b
    type: DS18S20
    address: 000803702e49

sensor_inputs:
  - name: temp_1
    module: ds18b22_1
    interval: 60
    digits: 2

  - name: temp_2
    module: ds18b22_2
    interval: 60
    digits: 2

  - name: temp_3
    module: ds18b22_3
    interval: 60
    digits: 2

  - name: temp_4
    module: ds18b22_4
    interval: 60
    digits: 2

  - name: temp_5
    module: ds18b22_5
    interval: 60
    digits: 2
```

neocit, 2020-05-19:
Many, many Thanks. Now it works! Great.

Changed it like you have it.
sensor_modules:
  - name: ds18b20_1
    module: ds18b
    type: DS18B20
    address: 0516c00797ff

sensor_modules:
  - name: ds18b20_2
    module: ds18b
    type: DS18B20
    address: 0416c0097fff
...

sensor_inputs:
  - name: Luft
    module: ds18b20_1
    interval: 60
    digits: 2

  - name: Solar
    module: ds18b20_2
    interval: 60
    digits: 2

2020-05-19 15:03:30,310 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'Solar' value 23.56
2020-05-19 15:03:31,190 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'Luft' value 21.5

closed 2020-05-19 by neocit