flyte/mqtt-io #113: Two Digital Outputs error

niraviry, 2020-06-20:
I have tried to use two digital outputs on 2 different pins. The only difference between the two is the name.
Once I add the 2nd output, I get an error in the pi_mqtt_gpio.log on the 2nd output :
-1 
  - off_payload
      - unknown field

The on_payload and the off_payload were set to "ON" and "OFF", "on" & "off" as well as "OPEN" and "CLOSED". The error is still there.
The error causes the yaml file not to load so nothing works.

The 1st one without the second one works fine.

flyte, 2020-06-20:
Hi @niraviry could you please provide your configuration file, with any sensitive information such as usernames and passwords removed?

niraviry, 2020-06-20:
Not in a position to do that now but it is about the same as in the readme.

niraviry, 2020-06-22:
Here is the code:
`gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_inputs:
  - name: button_1
    module: raspberrypi
    pin: 17
    on_payload: "ON"
    off_payload: "OFF"
#    inverted: yes
    pullup: yes
    pulldown: no
#    interrupt: falling
    bouncetime: 200
    retain: yes

  - name: button_2
    module: raspberrypi
    pin: 22
    on_payload: "ON"
    off_payload: "OFF"
#    inverted: yes
    pullup: yes
    pulldown: no
#    interrupt: falling
    bouncetime: 200
    retain: yes

digital_outputs:
  - name: doorlock
    module: raspberrypi
    pin: 23
    on_payload: "ON"
    off_payload: "OFF"
    initial: "high"
    retain: yes
#  - name: audio_ps_inhibit
#    module: raspberrypi
#    pin: 24
#    on_payload: "ON"
#    off_playload: "OFF"
#    initial: "high"
#    retain: yes`

I have omitted the start of the yml file as it contains private info.
Once I remote the remarks of the 2nd output I/O, I get the above error.
If I remove the off_payload and on_payload, I get error that these definitions are missing.
The indentation is OK after removing the #.

niraviry, 2020-06-23:
I have closed the issue by mistake.

BenjiU, 2020-06-24:
there is a typo in audio_ps_inhibit: off_playload :-)

niraviry, 2020-06-24:
Anyone has any clue to why I get the above error?

niraviry, 2020-06-24:
That was quick. Thanks.

BenjiU, 2020-06-24:
when you fix your typo i mentioned, I don't get your error when starting with your config file.
With the type, i also get your error:
```
digital_outputs:
- 1:
  - off_playload:
    - unknown field
```

closed 2020-06-24 by BenjiU