flyte/mqtt-io #114: Support for Capacitive Soil Moisture Sensor

barista123, 2020-06-26:
I would like to add Capacitive Soil Moisture Sensors to my setup such as these: [Capacitive Soil Moisture Sensor](https://www.amazon.com/KeeYees-Capacitive-Moisture-Sensor-Module/dp/B07R174TM1/ref=asc_df_B07R174TM1/?tag=hyprod-20&linkCode=df0&hvadid=343269677504&hvpos=&hvnetw=g&hvrand=13432535740877640298&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9026229&hvtargid=pla-818669536757&psc=1&tag=&ref=&adgrpid=77503953508&hvpone=&hvptwo=&hvadid=343269677504&hvpos=&hvnetw=g&hvrand=13432535740877640298&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9026229&hvtargid=pla-818669536757) to my setup. I know that they need a ADC. Is it possible with the current state of this project?  If so, please recommend hardware. I am new to working with the GPIOs and I am not a programmer.

Thanks for the project

BenjiU, 2020-06-27:
Hi, would the MCP3008 fit for you? Are you able to buy it? It has a 10bit ADC, what should be enough for your coils, right? There is also a lib for it, so wie could implement it: https://pypi.org/project/mcp3008/

barista123, 2020-06-27:
I do not currently have the MCP3008, but I am able to buy them.  I will be ordering the moisture sensors next week. If you can implement the lib or tell me what I need to do, I will order MCP3008 at the same time and test it.

BenjiU, 2020-06-27:
Okay, I will prepare a feature branch the next days. So you may test it :-)

barista123, 2020-06-27:
Thanks. It will probably be next weekend before I get the parts. I will also need some basic instructions on how to pull the branch for testing.  I installed the master via pip3.

barista123, 2020-07-01:
@BenjiU I see that you created a pull request. I should be receiving the MCP3008 today or tomorrow and will be ready to test. Please let me know how to install the branch and I will give it a try.

BenjiU, 2020-07-01:
Hi, I already wrote something, but closed the browser, bevor saving :-D
1. (optional) create virtual env `virtualenv ve` enter it `cd ve` and activate it `. bin/activate`
2. clone the project `git clone https://github.com/flyte/pi-mqtt-gpio.git`
3. checkout the feature branch `git checkout feature/mcp3008`
4. create your config file e.g. `config.mcp3008.yml`
```
mqtt:
  host: localhost
  port: 1883
  user: "xxx"
  password: "xxx"
  topic_prefix: mcptest

sensor_modules:
  - name: mcp3008_sensor
    module: mcp3008

sensor_inputs:
  - name: mcp3008_value
    module: mcp3008_sensor
    interval: 15
    digits: 2
```
5. start the application `python pi_mqtt_gpio/server.py config.mcp3008.yml`
6. send me the erros, that will occur :-D

barista123, 2020-07-03:
Trying to checkout the feature and keep getting this error

fatal: not a git repository (or any of the parent directories): .git

barista123, 2020-07-03:
Had to rebuild my pi. Just getting a chance to test. Got this error on startup

2020-07-03 14:22:24,341 mqtt_gpio (INFO): Startup
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting RPi.GPIO
  Downloading https://www.piwheels.org/simple/rpi-gpio/RPi.GPIO-0.7.0-cp37-cp37m-linux_armv7l.whl (69 kB)
     |████████████████████████████████| 69 kB 225 kB/s
Installing collected packages: RPi.GPIO
Successfully installed RPi.GPIO-0.7.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting spidev
  Downloading https://www.piwheels.org/simple/spidev/spidev-3.5-cp37-cp37m-linux_armv7l.whl (40 kB)
     |████████████████████████████████| 40 kB 239 kB/s
Installing collected packages: spidev
Successfully installed spidev-3.5
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/srv/ve/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 1216, in <module>
    main(args)
  File "/srv/ve/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 1043, in main
    SENSOR_MODULES[sensor_config["name"]] = configure_sensor_module(sensor_config)
  File "/srv/ve/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 586, in configure_sensor_module
    return sensor_module.Sensor(sensor_config)
  File "/srv/ve/pi-mqtt-gpio/pi_mqtt_gpio/modules/mcp3008.py", line 27, in __init__
    self.spi.open(0,0)
FileNotFoundError: [Errno 2] No such file or directory

BenjiU, 2020-07-04:
Hi barista123,

I got the same error, but I did not do the Step 0. Can you see device /dev/spi0.0? `ls /dev/spi0.0`?

barista123, 2020-07-04:
ls /dev/spi0.0 and /dev/spi0.1 report "no such file or directory"

It appears that the device is "/dev/spidev0.0" and "/dev/spidev0.1"

I discovered this by running "ls /dev/spi*"

Let me know what I can do next.

barista123, 2020-07-05:
Round two: I get this now (discovered that my spi were not enabled on first try)

2020-07-05 06:43:50,139 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/srv/ve/lib/python3.7/site-packages/pi_mqtt_gpio/server.py", line 887, in <module>
    main(args)
  File "/srv/ve/lib/python3.7/site-packages/pi_mqtt_gpio/server.py", line 773, in main
    SENSOR_MODULES[sensor_config["name"]] = configure_sensor_module(sensor_config)
  File "/srv/ve/lib/python3.7/site-packages/pi_mqtt_gpio/server.py", line 448, in configure_sensor_module
    sensor_module = import_module("pi_mqtt_gpio.modules.%s" % sensor_config["module"])
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'pi_mqtt_gpio.modules.mcp3008'

BenjiU, 2020-07-05:
But you are on the the "feature/mcp3008" branch`git status`? Does the file mcp3008.py exists?

barista123, 2020-07-05:
On branch feature/mcp3008
Your branch is up to date with 'origin/feature/mcp3008'.

nothing to commit, working tree clean

Where would I find the mcp3008.py file? I'm not seeing it in the directory

barista123, 2020-07-05:
Looks like it was an error in the way I was calling the program. It now started up, without error. Let me plug my sensor back in and see if I get any readings

barista123, 2020-07-05:
It appears to have gotten past the error, but I am not seeing any output from the sensor. Is it because we have not told the system which of the 8 channels to monitor?

I am setting up another program real quick to see if I get any output to confirm that my hardware is working

BenjiU, 2020-07-05:
By default it should read CH0. Otherwise you may configure it with the channel parameter:
```
  - name: mcp3008_voltage
    module: mcp3008_sensor
    interval: 300  # measurement every 5min
    channel: CH4 # measure on CH4 of MCP3008
```

barista123, 2020-07-05:
using these instructions, I have confirmed that the sensor works: https://learn.adafruit.com/raspberry-pi-analog-to-digital-converters/mcp3008

I will try assigning the channel and see if that helps.

barista123, 2020-07-05:
Now getting this error:

/srv/ve/pi-mqtt-gpio/pi_mqtt_gpio/server.py:728: DeprecationWarning: fractions.gcd() is deprecated. Use math.gcd() instead.
  cycle_time = reduce(lambda x, y: gcd(x, y), arr)
Exception in thread pi-mqtt-gpio_SensorReader:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/srv/ve/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 747, in sensor_timer_thread
    value = sensor.get_value(sens_conf)
  File "/srv/ve/pi-mqtt-gpio/pi_mqtt_gpio/modules/mcp3008.py", line 50, in get_value
    channel = self.channels.get(config.channel, "invalid")
AttributeError: 'dict' object has no attribute 'channel'

BenjiU, 2020-07-05:
Hi, I fixed something: please do a `git pull` and try it again :-)

barista123, 2020-07-05:
Getting closer. . .
That got it past the error, but now I get a sensor value of 0 regardless of the input

BenjiU, 2020-07-05:
I added some traces for debugging `git pull`. Please attach the "MCP3008: xxx" traces.

barista123, 2020-07-05:
2020-07-05 08:35:44,788 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/srv/ve/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 1216, in <module>
    main(args)
  File "/srv/ve/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 1043, in main
    SENSOR_MODULES[sensor_config["name"]] = configure_sensor_module(sensor_config)
  File "/srv/ve/pi-mqtt-gpio/pi_mqtt_gpio/server.py", line 576, in configure_sensor_module
    sensor_module = import_module("pi_mqtt_gpio.modules.%s" % sensor_config["module"])
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 724, in exec_module
  File "<frozen importlib._bootstrap_external>", line 860, in get_code
  File "<frozen importlib._bootstrap_external>", line 791, in source_to_code
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/srv/ve/pi-mqtt-gpio/pi_mqtt_gpio/modules/mcp3008.py", line 47
    _LOG.warning("MCP3008: adc %s", bytes(adc).hex())
    ^
IndentationError: unexpected indent

BenjiU, 2020-07-05:
sorry, my fault. :-( Please pull again!

barista123, 2020-07-05:
Got this:

> 2020-07-05 08:41:46,269 mqtt_gpio (INFO): Startup
> 2020-07-05 08:41:47,505 mqtt_gpio (WARNING): MCP3008: Reading from channel 0
> 2020-07-05 08:41:47,507 mqtt_gpio (WARNING): MCP3008: adc 000000
> 2020-07-05 08:41:47,508 mqtt_gpio (WARNING): MCP3008: data 0
> 2020-07-05 08:41:47,509 mqtt_gpio (WARNING): MCP3008: value 0
> 2020-07-05 08:41:47,510 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'mcp3008_value' value 0

When I ended the program with "ctrl+ c" I got this:

> /srv/ve/pi-mqtt-gpio/pi_mqtt_gpio/modules/raspberrypi.py:73: RuntimeWarning: No channels have been set up yet - nothing to clean up!  Try cleaning up at the end of your program instead!
  self.io.cleanup()

In contrast, the AdaFruit example that I linked above returns this:

root@DietPi:/srv/ve1# sudo python3 simpletest.py

> Reading MCP3008 values, press Ctrl-C to quit...
> |    0 |    1 |    2 |    3 |    4 |    5 |    6 |    7 |
> ---------------------------------------------------------
> |  870 |    3 |    4 |    4 |    5 |    4 |    5 |    7 |
> |  870 |    6 |    5 |    4 |    3 |    1 |    0 |    0 |
> |  870 |    3 |    4 |    5 |    6 |    5 |    5 |    6 |
> |  870 |    3 |    0 |    2 |    1 |    1 |    2 |    3 |
> |  870 |    6 |    6 |    6 |    6 |    5 |    5 |    5 |
> |  870 |    3 |    2 |    2 |    2 |    1 |    2 |    3 |
> |  870 |    6 |    6 |    6 |    6 |    5 |    5 |    5 |
> |  870 |    2 |    3 |    2 |    3 |    4 |    5 |    7 |
> |  870 |    6 |    5 |    4 |    2 |    1 |    1 |    1 |
> |  870 |    5 |    6 |    6 |    5 |    4 |    2 |    1 |
> |  870 |    2 |    3 |    5 |    5 |    5 |    6 |    7 |
> |  870 |    4 |    3 |    2 |    2 |    2 |    3 |    5 |
> |  870 |    6 |    6 |    5 |    4 |    2 |    0 |    0 |
> ^CTraceback (most recent call last):
>   File "simpletest.py", line 39, in <module>
>     time.sleep(0.5)

barista123, 2020-07-05:
I should clarify, the second readings are from a dry sensor. A wet sensor provides the following readings:

root@DietPi:/srv/ve1# sudo python3 simpletest.py
Reading MCP3008 values, press Ctrl-C to quit...
|    0 |    1 |    2 |    3 |    4 |    5 |    6 |    7 |
---------------------------------------------------------
|  445 |    2 |    2 |    2 |    3 |    4 |    4 |    6 |
|  445 |    7 |    7 |    6 |    6 |    5 |    4 |    4 |
|  445 |    2 |    2 |    3 |    4 |    5 |    6 |    9 |
|  445 |    5 |    4 |    2 |    1 |    1 |    2 |    5 |
|  445 |    7 |    7 |    6 |    5 |    3 |    1 |    0 |
|  445 |    3 |    4 |    5 |    5 |    5 |    6 |    9 |
|  445 |    6 |    5 |    5 |    4 |    1 |    0 |    0 |
|  445 |    4 |    5 |    5 |    5 |    5 |    6 |    8 |
^CTraceback (most recent call last):
  File "simpletest.py", line 39, in <module>
    time.sleep(0.5)

BenjiU, 2020-07-05:
Okay, I'll try to implement the Adafruit lib :-) but I can't start before this evening.

barista123, 2020-07-05:
Thanks. I appreciate the help.

BenjiU, 2020-07-05:
Hi, I implemented the Adafruit lib. Please pull and run python3 with sudo. When i use python2, it can't load the adafruit lib.
`sudo python3 -m pi_mqtt_gpio.server config.mcp3008.yml`

barista123, 2020-07-05:
It appears to be working. Without a sensor installed I get a value of 3 which is comparable to the Adafruit simpletest script. I am in the process of moving the pi to a new enclosure. So I will reinstall the sensor this evening and try again.

Also, since I am using pip3, I had to manually install the Adafruit library. The script tried to use pip instead of pip3 and failed.  I got an error ( unfortunately, I am testing on my cell phone and could not copy the error) I will uninstall the library later and try to get a copy of the error for you.

BenjiU, 2020-07-06:
Hi, sounds good. I had a look at my python environment and noticed, that `pip installs` to python3.6, but `python` runs python2.7 :-) when I manually do `pip2 install Adafruit-xxx`, I can run it with python (not python3). So maybe I works for you with `python xxx`.

barista123, 2020-07-06:
Just tested this morning. I’m getting a reading from the sensor comparable to the test runs I ran yesterday. I will change the parameters this evening after work to confirm that it updates with changes. But it appears to be working.

barista123, 2020-07-07:
I apologize. I did not get a chance to do further testing last night. I will try tonight and post my results.

barista123, 2020-07-07:
On a different topic, I can open a different issue if appropriate. My temperature sensor, DHT22, does not like the environment that I am using it in (greenhouse). If I cannot find a suitable location, I may move to a SHT-30 (https://www.adafruit.com/product/4099). Would it be difficult to incorporate this sensor?

The system appears to be working great BTW, this is an extremely useful project.

barista123, 2020-07-07:
Information and python library for the sensor can be found here

barista123, 2020-07-08:
Great news, the sensor reacts as expected. It looks like this feature is working as expected. Thanks for all your help!

closed 2020-10-17 by flyte