flyte/mqtt-io #116: Digital Input - Polling

barista123, 2020-07-09:
Is it possible to poll the status of a digital input like sensors?  I am using a couple float switches to check the status of a water tank. My front end is home assistant. When the home assistant server is restarted for updates, config changes, etc. It does not retain the last known state of the float switches and defaults to off.  If I manually trigger the switch or restart pi-mqtt-gpio the state updates correctly.

Thanks

flyte, 2020-07-09:
You'll probably want to use `retain` for that. For example:

```yaml
digital_inputs:
  - name: water_tank
    module: whatever
    pin: 22
    on_payload: "ON"
    off_payload: "OFF"
    retain: yes
```

barista123, 2020-07-09:
Thank you. I thought about the retain, but did not see it discussed with
respect to inputs.

chrisnik007, 2020-07-11:
Can you please explain how retain works in respect of digital inputs? I have similar situation, where I am monitoring a specific gpio (linked to alarm) and I would like to have a polling interval, i.e. to re-publish the mqtt message at specific intervals - similar to what you have for other sensors. Is this possible?

BenjiU, 2020-08-06:
Retain is a MQTT mechanism to store the latest value of a topic in the MQTT broker and publish it to new connected clients, so they know the latest (equal to the current value), even if the new client "missed" the value change. This makes the cyclic transmission of the same value obsolete. It works for outputs and for inputs of pi_mqtt_gpio. So you should not need to have a polling for GPIOs. For sensors the polling is only used to trigger the sensor read (I currently don't know if we would send the same value twice).

closed 2020-10-18 by flyte