flyte/mqtt-io #117: piface digital not loading

claudegel, 2020-07-09:
I want to use piface digital. There is a module piface2 for the piface digital 2 Cna I use this module with the old piface digital ?
When trying this module I get 
2020-07-09 09:24:14,803 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 887, in <module>
    main(args)
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 788, in main
    initialise_digital_output(out_conf, GPIO_MODULES[out_conf["module"]])
KeyError: 'piface2'

flyte, 2020-07-09:
Hi @claudegel I'm not actually sure whether it works with the original PiFace. Could you provide your config through, so we can see what's causing this particular error? Just remove any sensitive things like IPs and usernames/passwords.

claudegel, 2020-07-10:
I finally got it to work. It was not clear in the doc about where to write the module name.

```
mqtt:
  host: 192.168.xx.xx
  port: 1883
  user: "username"
  password: "passwd"
  topic_prefix: arrosage

gpio_modules:
  - name: piface2       <-----
    module: piface2

digital_outputs:
  - name: valve_avant
    module: piface2
    pin: 0
    on_payload: "ON"
    off_payload: "OFF"
    initial: low

  - name: valve_arriere
    module: piface2
    pin: 1
    on_payload: "ON"
    off_payload: "OFF"
    initial: low
```
my error was in the name: parameter

it work perfectly whit the piface digital

flyte, 2020-07-10:
Glad to hear it! Thanks for reporting back.

closed 2020-07-10 by flyte