flyte/mqtt-io #118: Multiple DHT22

barista123, 2020-07-10:
Is it possible to included multiple DHT22 sensors? If so, how do you identify which sensor_input belongs to which dht22? I have distinguished the DHT22 in the sensor_module portion by pin number?  

I apologize if this is another basic question, still trying to figure this out.

BenjiU, 2020-08-06:
Hi, sorry for the long delay. This should be possible, by different pins. But it I haven't tested it:
```
mqtt:
  host: test.mosquitto.org
  port: 1883
  user: ""
  password: ""
  topic_prefix: home

sensor_modules:
  - name: dht22_home
    module: dht22
    type: AM2302 # can be  DHT11, DHT22 or AM2302
    pin: 4

  - name: dht22_basement
    module: dht22
    type: AM2302 # can be  DHT11, DHT22 or AM2302
    pin: 15

sensor_inputs:
  - name: dht22_home_temperature 
    module: dht22_home
    interval: 10 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 4 # number of digits to be round
    type: temperature # Can be temperature or humidity

  - name: dht22_home_humidity 
    module: dht22_home
    interval: 10 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 4 # number of digits to be round
    type: humidity # Can be temperature or humidity   

  - name: dht22_basement_temperature 
    module: dht22_basement
    interval: 100 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 2 # number of digits to be round
    type: temperature # Can be temperature or humidity

  - name: dht22_basement_humidity 
    module: dht22_basement
    interval: 100 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 2 # number of digits to be round
    type: humidity # Can be temperature or humidity   
```

You name the dht22 modules different and give the name in den sensor inputs.

barista123, 2020-08-06:
I can confirm that this does work. The duplicative use of the “module” designation is where I got confused.

closed 2020-10-17 by flyte