flyte/mqtt-io #120: device address in ds18b module

claudegel, 2020-07-14:
How do you get the address id for sensor in that module. I have a DS18B20 sensor connected to input 2 of my piface digital.
I get 
```
w1thermsensor.errors.NoSensorFoundError: 
Could not find sensor of type DS18B20 with id 000803702e49 <-- this is from the doc
Please check cabling and check your /boot/config.txt for
dtoverlay=w1-gpio
```
Should I have dtoverlay=w1-gpio in my /boot/config.txt ?

BenjiU, 2020-07-14:
1) yes, you should have `dtoverlay=w1-gpio in your` config.txt and reboot.
2) you can get the id of YOUR sensor with `ls -l /sys/bus/w1/devices`: the last numbers are the ids:
`lrwxrwxrwx 1 root root 0 Mai 13 07:30 10-000802bf634d` -> 000802bf634d

claudegel, 2020-07-15:
Ok I've `dtoverlay=w1-gpio` activ and I've rebooted but still no go. I've found that I can do 
`w1thermsensor ls` but I get `Found 0 sensors:`
My sensor is connected red on 5V, yellow on input #2 of piface digital and black on ground.
Input #2 is it ok or should I use another. My sensor is DS18B20 water proof with 3 feet cable. Once it is working I'll buy a small one.
When I launch pi_mqtt_gpio I get the following error
```
w1thermsensor.errors.NoSensorFoundError: 
Could not find sensor of type DS18B20 with id 34c000000000
Please check cabling and check your /boot/config.txt for
dtoverlay=w1-gpio
```

claudegel, 2020-08-05:
Ok I got it . I've plugged the data cable of the sensor directly on the GPIO 4 under the piface. Now I have this
```
pi@phytopi ~ $ w1thermsensor ls
Found 1 sensors:
  1. HWID: 0319a27966d7 Type: DS18B20
```
and this
```
pi@phytopi ~ $ w1thermsensor all
Got temperatures of 1 sensors:
  Sensor 1 (0319a27966d7) measured temperature: 26.69 celsius

now I need to export it via mqtt to my HA
```

claudegel, 2020-08-05:
As this sensor is plugged directly into the GPIO4 of the raspberrypi and not on the piface, should I have more then one gpio_modules listed like this:
```
gpio_modules:
  - name: piface2
    module: piface2
 - name: raspberrypi
    module: raspberrypi
```
Or should I have two seperate  gpio_modules: entry
```
gpio_modules:
  - name: piface2
    module: piface2

digital_outputs:
  - name: valve_avant
    module: piface2
    pin: 1
    on_payload: "ON"
    off_payload: "OFF"
    initial: low # low or high

...
gpio_modules:
 - name: raspberrypi
    module: raspberrypi

sensor_modules:
  - name: ds18b20
...
```

claudegel, 2020-08-06:
Great it's working. Here is my config for piface relay control and ds18b20 sensor on the GPIO 4. I had to plug the data cable of the sensor directly on GPIO4 under the piface.
```
mqtt:
  host: 192.168.20.178
  port: 1883
  user: "user"
  password: "passwd"
  topic_prefix: arrosage

gpio_modules:
  - name: piface2  
    module: piface2

  - name: raspberrypi   #needed for sensor
    module: raspberrypi

digital_outputs:
  - name: valve_avant
    module: piface2
    pin: 1
    on_payload: "ON"  
    off_payload: "OFF"
    initial: low # low or high

  - name: valve_arriere
    module: piface2
    pin: 0
    on_payload: "ON"  
    off_payload: "OFF"
    initial: low

  - name: fan
    module: piface2
    pin: 2
   initial: high
    on_payload: "ON"
    off_payload: "OFF"

sensor_modules:
  - name: ds18b20
    module: ds18b
    type: DS18B20
    address: 0319a27966d7

sensor_inputs:
  - name: ds18b20
    module: ds18b20
    interval: 60   
    digits: 2     
```
Thanks great software for my domotic

closed 2020-08-06 by claudegel