flyte/mqtt-io #124: Orange pi module

aleshka2201, 2020-08-01:
Hi!
I want use pi-mqtt-gpio with my orange pi and external mqtt server.
But I reciverd an error:

alexey@orangepione:~$ python -m pi_mqtt_gpio.server config.yml
2020-08-01 11:50:37,558 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/alexey/.local/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 887, in <module>
    main(args)
  File "/home/alexey/.local/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 759, in main
    GPIO_MODULES[gpio_config["name"]] = configure_gpio_module(gpio_config)
  File "/home/alexey/.local/lib/python2.7/site-packages/pi_mqtt_gpio/server.py", line 427, in configure_gpio_module
    gpio_module = import_module("pi_mqtt_gpio.modules.%s" % gpio_config["module"])
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/home/alexey/.local/lib/python2.7/site-packages/pi_mqtt_gpio/modules/orangepi.py", line 66
    raise IOError("channel %d setup failed" % pin) from e
                                                      ^
SyntaxError: invalid syntax

My config.yml:

mqtt:
  host: 192.168.1.90
  port: 1883
  user: ""
  password: ""
  topic_prefix: home/orange

gpio_modules:
  - name: orangepi
    module: orangepi
    board: one 
    mode: board

#digital_outputs:
  - name: lights
    module: orangepi
    pin: 7
    on_payload: "ON"
    off_payload: "OFF"
    initial: low  # This optional value controls the initial state of the pin b$
    retain: yes # This option value controls if the message is retained. Defaul$

  - name: fan
    module: orangepi
    pin: 8
    inverted: yes  # This pin may control an open-collector output which is "on$
    on_payload: "ON"
    off_payload: "OFF"

Do you have any solutions?
thank you

aleshka2201, 2020-08-01:
With my raspberry pi all works fine..

BenjiU, 2020-08-06:
Hi, I think the problem is already fixed in #106 
May you test the current develop branch, since it is not officially released yet, i think.

closed 2020-10-17 by flyte