flyte/mqtt-io #125: Initial output pin states

mr2c12, 2020-08-16:
Dear all,
this is quite similar to #76 but not really the same.

I think that the state of the output pins should be published at boot. Since we can configure the inital state for these pins, and the initial state could be different from the last one (e.g. pin is HIGH, restart, default LOW, pin is LOW but noone is notified), situations can occur where the known state of a pin is different from the actual one.

Any thought?

flyte, 2020-10-17:
I've made an option to publish the value of the output on startup and added instructions to the readme (see commits above). I'm not in a position to test it out at the moment, but please let me know if it works for you if you do. Otherwise I'll test it as soon as I can, and make a new release.

mr2c12, 2020-10-18:
Any way to get the required changes in my virtualenv+pip installation?

flyte, 2020-10-18:
Nah it's probably mine! I'll give it a test myself in a while. Cheers

flyte, 2020-10-18:
OK, I fixed the issue - I was trying to publish before connecting to MQTT :)

@mr2c12 Please give develop another pull and try again!

mr2c12, 2020-10-20:
Sorry for the late reply.

Actually it is working (i.e., publishing on start) but there is something wrong in the data published.

Consider my config.yaml:

`digital_outputs:
  - name: pin_buzzer
    module: raspberrypi
    pin: 22
    on_payload: "ON"
    off_payload: "OFF"
    initial: low
    retain: no
    publish_initial: yes
  - name: pin24_led
    module: raspberrypi
    pin: 24
    on_payload: "ON"
    off_payload: "OFF"
    initial: low
    retain: no
    publish_initial: yes
  - name: pin26_r1
    module: raspberrypi
    pin: 26
    on_payload: "OFF"
    off_payload: "ON"
    initial: high
    retain: no
    publish_initial: yes
  - name: pin19_r2
    module: raspberrypi
    pin: 19
    on_payload: "OFF"
    off_payload: "ON"
    initial: high
    retain: no
    publish_initial: yes
  - name: pin13_r3
    module: raspberrypi
    pin: 13
    on_payload: "OFF"
    off_payload: "ON"
    initial: high
    retain: no
    publish_initial: yes
  - name: pin6_r4
    module: raspberrypi
    pin: 6
    on_payload: "OFF"
    off_payload: "ON"
    initial: high
    retain: no
    publish_initial: yes
  - name: pin23_gate
    module: raspberrypi
    pin: 23
    on_payload: "ON"
    off_payload: "OFF"
    initial: low
    retain: no
    publish_initial: yes`

This is the result in the mqtt topic:
[image]

The published value seems to be the opposite.
E.g. pin23_gate, while actually being OFF, is published as ON.

Please let me know if there is something else you need!

mr2c12, 2020-11-13:
There is maybe an error here:

```
initial_setting = out_conf.get("initial")
if initial_setting is not None and out_conf.get("publish_initial", False):
      payload = out_conf["on_payload" if initial_setting else "off_payload"]
```

Payload value is always the "on_payload" (as per my configuration).
Should it check that the "initial_setting" is "high" in the last line of code above?
Something like:

```
payload = out_conf["on_payload" if initial_setting == "high" else "off_payload"]
```

?

I also took a look at raspberrypi module, and in setup_pin you'll find:
`
initial = {None: -1, "low": 0, "high": 1}[pin_config.get("initial")]
`
So probably make sense.

mr2c12, 2020-11-13:
I can confirm that the solution above works!

flyte, 2021-01-24:
Well spotted! Thanks. I've fixed that now and it'll be in the next release.

closed 2021-01-24 by flyte