flyte/mqtt-io #127: cannot install on raspbian rpi4 latest

claudegel, 2020-08-27:
When I run pip3 install pi-mqtt-gpio I get the following error
Could not install packages due to an EnvironmentError: 404 Client Error: Not Found for url: https://www.iwheels.org/simple/cerberus.
Do I need to install other dependency ?

claudegel, 2020-08-27:
Cerberus is missing ?

zellneralex, 2020-08-28:
It seams like all versions of cerberus get deleted from www.piwheels.org

claudegel, 2020-08-31:
Ok I've copied over from another pi de cerberus  and Cerberus-1.3.2.egg-info dir and it is working. I still have an issue with auto startup at boot, see #129 above.

g8ryjessop, 2020-09-04:
add this line to your /etc/pip.conf

`--extra-index-url=https://pypi.org/project`

then execute

`pip3 install Cerberus`

claudegel, 2020-09-05:
That work. I was able to install Cerberus.

closed 2020-09-13 by claudegel