flyte/mqtt-io #128: raspberrybi 3B: docker container reports: No module named 'yaml'

RaimondB, 2020-08-29:
When starting the image I get the logs below. It states it cannot import the yaml module.
This seems to be something wrong in the build of the docker image?

```
gpio-to-mqtt_1  | Creating a virtualenv for this project…
gpio-to-mqtt_1  | Pipfile: /home/mqttgpio/Pipfile
gpio-to-mqtt_1  | Using /usr/local/bin/python (3.6.10) to create virtualenv…
⠋ Creating virtual environment...created virtual environment CPython3.6.10.final.0-32 in 3360ms
gpio-to-mqtt_1  |   creator CPython3Posix(dest=/root/.local/share/virtualenvs/mqttgpio-815CbdL_, clear=False, global=False)
gpio-to-mqtt_1  |   seeder FromAppData(download=False, pip=latest, setuptools=latest, wheel=latest, via=copy, app_data_dir=/root/.local/share/virtualenv/seed-app-data/v1)
gpio-to-mqtt_1  |   activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
gpio-to-mqtt_1  |
gpio-to-mqtt_1  |✔ Successfully created virtual environment!
gpio-to-mqtt_1  | Virtualenv location: /root/.local/share/virtualenvs/mqttgpio-815CbdL_
gpio-to-mqtt_1  | Traceback (most recent call last):
gpio-to-mqtt_1  |   File "/usr/local/lib/python3.6/runpy.py", line 183, in _run_module_as_main
gpio-to-mqtt_1  |     mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
gpio-to-mqtt_1  |   File "/usr/local/lib/python3.6/runpy.py", line 109, in _get_module_details
gpio-to-mqtt_1  |     __import__(pkg_name)
gpio-to-mqtt_1  |   File "/home/mqttgpio/pi_mqtt_gpio/__init__.py", line 1, in <module>
gpio-to-mqtt_1  |     import yaml
gpio-to-mqtt_1  | ModuleNotFoundError: No module named 'yaml'
```

vasylkuzenko, 2020-08-30:
try "pip install pyyaml"

if will say: ModuleNotFoundError: No module named 'paho', need "pip install paho-mqtt" too

RaimondB, 2020-08-30:
I would expect that the docker image already contains all needed dependencies. So the dockerfile should be fixed then?

flyte, 2021-01-24:
Sorry for the delay.

Please can you give information about the docker image you're using, the command you're running and your redacted config file?

RaimondB, 2021-01-31:
I am now using the following contents for the Dockerfile, in fact doing the "manual" setup, but as a Dockerfile.
And not using the pipenv, since the docker environment already provides total isolation.

```
FROM python:3.8-buster

ENV LANG C.UTF-8
ENV LC_ALL C.UTF-8

WORKDIR /home/root

RUN pip install pi-mqtt-gpio

VOLUME /config

CMD [ "python", "-m", "pi_mqtt_gpio.server", "/config/config.yml" ]
```

closed 2022-11-04 by BenjiU