flyte/mqtt-io #129: Auto start not working

claudegel, 2020-08-30:
I've added to /etc/rc.local just before the last line exit 0
`python -m pi_mqtt_gpio.server /home/pi/config.yml &`
to have the program started automatically. But it fail somehow.
If I start it manually with 
`python -m pi_mqtt_gpio.server /home/pi/config.yml`
It work ok
Where can I check to find out what is going wrong ?
My config
```
mqtt:
  host: 192.168.20.178
  port: 1883
  user: "user"
  password: "passwd"
  topic_prefix: solarium

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_outputs:
  - name: toit_spa
    module: raspberrypi
    pin: 18
    on_payload: "ON"
    off_payload: "OFF"
    initial: low # low or high
    retain: yes

  - name: toit_solarium
    module: raspberrypi
    pin: 17
    on_payload: "ON"
    off_payload: "OFF"
    initial: low
    retain: yes

  - name: toit_toggle
    module: raspberrypi
    pin: 23
    on_payload: "ON"
    off_payload: "OFF"
    initial: low
    retain: yes

digital_inputs:
  - name: rain
    module: raspberrypi
    pin: 22
    on_payload: "on"
    off_payload: "off"
    pullup: yes
    pulldown: no
```

g8ryjessop, 2020-09-04:
I've put in the whole path and without the '-m' option and this works for me.

`/usr/bin/python3 /usr/local/lib/python3.7/dist-packages/pi_mqtt_gpio/server.py /home/pi/config.yml &`

claudegel, 2020-09-06:
I was able to install on my raspi 4 but I can just start manually via 
`sudo /usr/bin/python3 /usr/local/lib/python3.7/dist-packages/pi_mqtt_gpio/server.py /home/pi/config.yml &
`
But if I put that lline in /etc/rc.local (without sudo) it do not work. My pi is connected to a card with 3 relay. When I start manually, relay 1 and 2 are down and relay 3 is up. But when I reboot, all three relay become up. Don't know if it is pi_mqtt_gpio that is triggering them up or they just become up.

claudegel, 2020-09-06:
Ok I've added the following line in rc.local, just above line saying # Print the IP address:
` exec > /tmp/rc-local.out 2>&1;set -x`

and I got the following error:
```
+ python -m pi_mqtt_gpio.server /home/pi/config.yml
2020-09-05 14:11:38,756 mqtt_gpio (INFO): Startup
2020-09-05 14:11:39,116 mqtt_gpio (CRITICAL): Unable to connect to MQTT server: [Errno 101] Network is unreachable

```
so pi_mqtt_gpio start before the network is established and crash because there is no mqtt server. That's why I can start it manually but not via rc.local

How can I delay pi_mqtt_gpio to wait until network is up

claudegel, 2020-09-06:
Got it... I've added a delay before starting pi_mqtt_gpio and it work
```
sleep 20
python -m pi_mqtt_gpio.server /home/pi/config.yml &
```

closed 2020-09-13 by claudegel