flyte/mqtt-io #135: SuperPi-case-J

claudegel, 2020-09-30:
Can you tell me if this board is supported
https://www.52pi.com/cases-heatsinks-fans/30-superpi-case-j.html

BenjiU, 2020-09-30:
Hi, from your link i googled this wiki page:
https://wiki.52pi.com/index.php/DockerPi_4_Channel_Relay_SKU:_EP-0099
according to the information there, it is currently not supported, because it is connected via i2c.
But there is a short example, that may be implemented easily.

claudegel, 2020-10-08:
I've produce a module for it dockerpi.py
How can I test it locally before submitting it. Do I need to just copy that file to /usr/local/lib/python3.7/dist-packages/pi_mqtt_gpio/modules ? dio I need to erase `__pycache__` dir and restart my pi

BenjiU, 2020-11-07:
The normal way would be to clone this repository and start it from there. Then you don't have to "destroy" your system installation.

claudegel, 2021-12-30:
Module dokerpi is working for Dokerpi 4 relay. I've sent a pull request. Thank you

closed 2022-11-04 by BenjiU