flyte/mqtt-io #136: Digital outputs multiple pin, same topic

terminet85, 2020-10-07:
There is any change to turn on/off multiple pin with the same mqtt topic?

RPi.GPIO module should be able to manage it
`chan_list = [11,12]    # add as many channels as you want!
                       # you can tuples instead i.e.:
                       #   chan_list = (11,12)
GPIO.setup(chan_list, GPIO.OUT)
`

terminet85, 2020-10-08:
I've made a PR in order to fix schema check. 
RPi.GPIO  are already capable to manage lists, so the problem was only in the config check.

`digital_outputs:
  - name: fireplace
    module: raspberrypi
    pin:
       - 17
       - 27
       - 22
       - 5
`

flyte, 2020-10-17:
Hi, thanks for your work on this. It's a good idea, but as far as I'm aware, your PR would break compatibility with all of the other output modules, such as `pcf8574` which has a `set_pin()` implementation of:

```python
    def set_pin(self, pin, value):
        self.io.port[pin] = value
```

As you can see, the `pin` value can not be a list in this example.

In order to implement this, we'd need to establish which modules' underlying libraries are able to take lists of pins and set them all at once, then enable this functionality on just these modules.

As it stands, this is a relatively low priority for me compared to some of the other things on my list, so I wouldn't work on it any time soon, but pull requests to the effect are welcome.

I'll close the existing PR as it stands, but keep this issue open to track any progress.