flyte/mqtt-io #139: Usage with Brewblox (pi-mqtt-gpio 0.5.1 has requirement PyYAML<=5.2,>=4.2b1, but you'll have pyyaml 5.3.1 which is incompatible.)

secuspec, 2020-10-16:
pi-mqtt-gpio would be an interesting addition to Brewblox (https://brewblox.netlify.app) a brewery control platform. 
Brewblox currently uses pyyaml 5.3.1, while pi-mqtt-gpio requires version 0.5.1.

Is there a chance to upgrade pi-mqtt-gpio to the later version?

Thanks,
Steffen

flyte, 2020-10-16:
Yep, I'll spend some time this weekend sorting out some of these issues including bringing this up to date.

I'd certainly be up for helping to integrate more readily with your software, as I'm into brewing beer too!

flyte, 2020-10-17:
The PyYAML requirement is now `>=5.3.1` on the `develop` branch. I'll make a new release shortly.

closed 2020-10-17 by flyte

secuspec, 2020-10-17:
Excellent 👍

Will upgrade and test.

Best regards,
Steffen