flyte/mqtt-io #141: Asyncio core rewrite testing

flyte, 2020-10-18:
I've built an asyncio version of the core, which I'd like people to test out before I inflict it on the general public. It exists within `feature/asyncio`, and for the most part should work with existing config files. The MQTT section may need changing slightly if you're using TLS, so have a look in the `config.schema.yml` file to see how that's changed.

The program should be launched with `python -m mqtt_io your.config.yml` as the module has changed.

Please use this Issue or [Discord](https://discord.gg/gWyV9W4) to discuss your experiences. I fully expect stuff to be quite broken at this point! At the moment there's no stream support either, but I'll get that implemented at some point.

Tagging the people who have contributed in the past, in case they're able to help out. Don't worry if you're not interested, and thank you again for your previous contributions!
@BenjiU 
@yozik04
@RoelGo 
@ChrisEAlfred 
@a-tom-s 
@tomas-petrauskas 
@shbatm 
@magyarl 
@jaburges 
@hpeyerl 
@bwduncan 
@SkySails 
@PeterWoolery 
@ComlOnline

BenjiU, 2020-11-05:
Hi @flyte,

I just gave it a try:
1) the requirements.txt need a fix, i think. `pip3 install hbmqtt` fixed it
```
pi@bennibustest:~/pi-mqtt-gpio $ python3 -m mqtt_io config.stdio.yml 
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/pi-mqtt-gpio/mqtt_io/__main__.py", line 7, in <module>
    from .server import MqttGpio
  File "/home/pi/pi-mqtt-gpio/mqtt_io/server.py", line 9, in <module>
    from hbmqtt.client import ClientException, MQTTClient
ModuleNotFoundError: No module named 'hbmqtt'
```
2) On cancel with STRG+C i got this error. But not every time.
```
^C2020-11-05 19:40:17 mqtt_io.server [WARNING ] Received exit signal SIGINT
2020-11-05 19:40:17 mqtt_io.server [INFO    ] Waiting for main task to complete...
2020-11-05 19:40:17 mqtt_io.server [INFO    ] Cancelling 2 remaining tasks
2020-11-05 19:40:17 mqtt_io.server [INFO    ] Waiting for 2 remaining tasks to complete...
^C2020-11-05 19:40:18 mqtt_io.server [WARNING ] Received exit signal SIGINT
2020-11-05 19:40:18 mqtt_io.server [INFO    ] Waiting for main task to complete...
2020-11-05 19:40:19 mqtt_io.server [INFO    ] Cancelling 2 remaining tasks
2020-11-05 19:40:19 mqtt_io.server [INFO    ] Waiting for 2 remaining tasks to complete...
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/pi-mqtt-gpio/mqtt_io/__main__.py", line 30, in <module>
    main()
  File "/home/pi/pi-mqtt-gpio/mqtt_io/__main__.py", line 26, in main
    mqtt_gpio.run()
  File "/home/pi/pi-mqtt-gpio/mqtt_io/server.py", line 423, in run
    self.loop.run_until_complete(self._init_mqtt())
  File "/usr/local/lib/python3.6/asyncio/base_events.py", line 467, in run_until_complete
    return future.result()
concurrent.futures._base.CancelledError
```

3) python2 (Python 2.7.13) does not work, but python3.6 is okay:
```
pi@bennibustest:~/pi-mqtt-gpio $ python -m mqtt_io config.stdio.yml 
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 163, in _run_module_as_main
    mod_name, _Error)
  File "/usr/lib/python2.7/runpy.py", line 114, in _get_module_details
    return _get_module_details(pkg_main_name)
  File "/usr/lib/python2.7/runpy.py", line 119, in _get_module_details
    code = loader.get_code(mod_name)
  File "/usr/lib/python2.7/pkgutil.py", line 283, in get_code
    self.code = compile(source, self.filename, 'exec')
  File "/home/pi/pi-mqtt-gpio/mqtt_io/__main__.py", line 19
    print(str(e), file=sys.stderr)
                      ^
SyntaxError: invalid syntax
pi@bennibustest:~/pi-mqtt-gpio $ python --version
Python 2.7.13
```
But my temp, humidity and distance module work and sending mqtt messages.

flyte, 2020-11-30:
Cool, thanks @BenjiU 

The async version will drop support for Python 2, so that's fine.

Quitting async programs is considerably more complicated than synchronous ones, unfortunately. I'll have to look into that again.

I think I added the hbmqtt requirement to the Pipfile and the requirements.txt is out of date.

flyte, 2021-01-21:
@BenjiU 

Could you let me know the config file you used for your tests please?

Also, it looks like that error popped up when `_init_mqtt()` was cancelled. This leads me to think that you pressed STRG+C quite early on in the program and it hadn't yet connected to the MQTT server. Does that sound right?

Have you done any further testing since reporting back?

Cheers

closed 2021-02-25 by flyte