flyte/mqtt-io #142: Docker reporting its not a pi

g8ry, 2020-10-20:
I'm running a Pi 4B and I've just used the following command to build the container... Am I missing something ?

docker run -d --name=mqtt-gpio --privileged -v /apps/docker/mqtt-gpio/config.yml:/config.yml flyte/mqtt-gpio:armv7l

The logs inside the container are showing the following:-

2020-10-20 07:22:05,889 mqtt_gpio (INFO): Startup
Collecting RPi.GPIO
  Downloading RPi.GPIO-0.7.0.tar.gz (30 kB)
Building wheels for collected packages: RPi.GPIO
  Building wheel for RPi.GPIO (setup.py): started
  Building wheel for RPi.GPIO (setup.py): finished with status 'done'
  Created wheel for RPi.GPIO: filename=RPi.GPIO-0.7.0-cp36-cp36m-linux_armv7l.whl size=61792 sha256=daeee1d56c873e7ca47587fdcaca469d1c3ecf9594fa036874addbe3c03313e1
Successfully built RPi.GPIO
Installing collected packages: RPi.GPIO
Successfully installed RPi.GPIO-0.7.0
Collecting Adafruit_DHT
  Downloading Adafruit_DHT-1.4.0.tar.gz (15 kB)
Building wheels for collected packages: Adafruit-DHT
  Building wheel for Adafruit-DHT (setup.py): started
  Building wheel for Adafruit-DHT (setup.py): finished with status 'done'
  Created wheel for Adafruit-DHT: filename=Adafruit_DHT-1.4.0-cp36-cp36m-linux_armv7l.whl size=29577 sha256=5e895c0f91c2f06d9828176159ce4be9fe1f0a487625f5ecdc358c454bca027e
  Stored in directory: /home/mqttgpio/.cache/pip/wheels/af/d8/00/6f0c819ab5f38ea5ececf1b698a9e53888e271300d5f202260
Successfully built Adafruit-DHT
Installing collected packages: Adafruit-DHT
Successfully installed Adafruit-DHT-1.4.0
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 901, in <module>
    main(args)
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 798, in main
    initialise_digital_input(in_conf, GPIO_MODULES[in_conf["module"]])
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 492, in initialise_digital_input
    gpio.setup_pin(pin, PinDirection.INPUT, pud, in_conf)
  File "/home/mqttgpio/pi_mqtt_gpio/modules/raspberrypi.py", line 47, in setup_pin
    self.io.setup(pin, direction, pull_up_down=pullup, initial=initial)
RuntimeError: Not running on a RPi!

g8ry, 2020-10-20:
After some investigation, I only get the error when I have the 'digital_inputs' in my config. If I remove it and leave my other inputs/modules in there, it works fine.

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_inputs:
  - name: button
    module: raspberrypi
    pin: 21
    on_payload: "ON"
    off_payload: "OFF"
    pullup: yes
    pulldown: no

TBobsin, 2020-11-06:
Hello, 
i get an similar error on my raspberry PI 4 8GB running with
uname -a 
Linux smarthome 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux.

>config.yml

 `mqtt:`
  `host: mqtt`
  `port: 1883`
  `user: usrmqtt`
  `password: secure`
  `topic_prefix: haus/mqtt-gpio`
`gpio_modules:`
  `- name: raspberrypi`
    `module: raspberrypi`
`digital_inputs:`
  `- name: bwm`
    `module: raspberrypi`
    `pin: 4`
    `on_payload: "ON"`
    `off_payload: "OFF"`
    `pullup: yes`
    `pulldown: no`

>docker-compose.yml

`mqtt-gpio:`
  `  image: flyte/mqtt-gpio:armv7l`
    `container_name: mqtt-gpio`
    `restart: always`
    `privileged: true <- changed nothing`
    `volumes:`
     ` - /opt/mqtt-gpio/config.yml:/config.yml`
      `- /dev/gpiomem:/dev/gpiomem <- changed nothing`

2020-11-06 19:57:42,877 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 901, in <module>
    main(args)
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 773, in main
    GPIO_MODULES[gpio_config["name"]] = configure_gpio_module(gpio_config)
  File "/home/mqttgpio/pi_mqtt_gpio/server.py", line 451, in configure_gpio_module
    return gpio_module.GPIO(gpio_config)
  File "/home/mqttgpio/pi_mqtt_gpio/modules/raspberrypi.py", line 19, in __init__
    import RPi.GPIO as gpio
  File "/home/mqttgpio/.local/share/virtualenvs/mqttgpio-815CbdL_/lib/python3.6/site-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!

BenjiU, 2020-11-07:
Here they figured out, it may happen with an old python version, that didn't know the new raspberry PIs (3b+ and newer). Python2 did not work, Python3.7 worked.
https://www.raspberrypi.org/forums/viewtopic.php?t=273369
May you upgrade python in your docker container to 3.7 and check, if it still occurs?

g8ry, 2020-11-07:
Ive just built a new docker image with python3.7 with the relevant changes to the Pipfile and its still not working. I also tried python3.8 too.
I'll do some more testing

g8ry, 2020-11-08:
I Finally fixed it... All I needed to do was the following, then everything worked fine with the existing docker image python3.6:-

chmod og+rwx /dev/gpio*

jjarokergc, 2020-11-12:
@g8ry : are you running raspbian OS?  The chmod fix you used suggests a permission issue.  GPIO access used to require root privileges on Raspbian but I think that is no longer the case.  

My environment is an RPi4 8GB with Ubuntu 20.04 and I encounter the "Not Running on Pi" error too.  I also encounter the runtime error, "Failed to add edge detection" when I attempt to use "interrupt_payload: "triggered".  The chmod change above will not fix that error.

A better way to handle this configuration problem is through permissions.  The answer on this link is one approach that works for me:

[Failed to add edge detection - Raspberry Pi 3 GPIO](https://stackoverflow.com/questions/41679547/failed-to-add-edge-detection-raspberry-pi-3-gpio).  It uses a system account and rules to selectively grant permission to a non-root user.

g8ry, 2020-11-12:
> @g8ry : are you running raspbian OS? The chmod fix you used suggests a permission issue. GPIO access used to require root privileges on Raspbian but I think that is no longer the case.
> 
> My environment is a RPi4 8GB with Ubuntu 20.04 and I encounter the "Not Running on Pi" error too. I also encounter the runtime error, "Failed to add edge detection" when I attempt to use "interrupt_payload: "triggered". The chmod change above will not fix that error.
> 
> A better way to handle this configuration problem is through permissions. The answer on this link is one approach that works for me:
> 
> [Failed to add edge detection - Raspberry Pi 3 GPIO](https://stackoverflow.com/questions/41679547/failed-to-add-edge-detection-raspberry-pi-3-gpio). It uses a system account and rules to selectively grant permission to a non-root user.

closed 2020-11-12 by g8ry

g8ry, 2020-11-12:
@jjarokergc 
I'm running Ubuntu 20.10 on a 2GB RPi4
I'll take a look at that later. Thanks 👍