flyte/mqtt-io #143: could you add BME280 I2C support ?

tkranz, 2020-10-28:
Hi, great project !

is it possible to add the above device ?

thanks
Thomas

adrian-vlad, 2020-10-28:
Hi. It's already there: https://github.com/flyte/pi-mqtt-gpio/blob/develop/pi_mqtt_gpio/modules/bme280.py.
Probably not released yet.

tkranz, 2020-10-29:
is there any example ? i have installed the new version, but get the below error

`ast will set on 'mqtt1/status' as 'dead'.
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 570, in <module>
    SENSOR_MODULES[sensor_config["name"]] = configure_sensor_module(sensor_config)
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 415, in configure_sensor_module
    sensor_module = import_module("pi_mqtt_gpio.modules.%s" % sensor_config["module"])
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named bme280

root@mqtt1:~# cat /home/pi/pi-mqtt-gpio.yml
mqtt:
  host: 192.168.1.11
  port: 1883
  user: "openhabian"
  password: "habianopen"
  topic_prefix: mqtt1

gpio_modules:
  - name: raspberrypi
    module: raspberrypi

digital_outputs:
  - name: gpio20
    module: raspberrypi
    pin: 20
    on_payload: "ON"
    off_payload: "OFF"
    initial: high  # This optional value controls the initial state of the pin before receipt of any messages from MQTT. Valid options are 'low' and 'high'.
    retain: yes # This option value controls if the message is retained. Default no.

sensor_modules:
  - name: bme280test
    module: bme280
    i2c_bus_num: 1
    chip_addr: 0x76

root@mqtt1:~# python /home/pi/bme280.py
Chip ID     : 96
Version     : 0
T, 24.45
P, 1006.52309742
H, 53.3023271719
root@mqtt1:~#`

tkranz, 2020-10-29:
have reinstalled the full pkg but still the follwing error

`
2020-10-29 16:39:49,845 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 1267, in <module>
    main(args)
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 1083, in main
    SENSOR_MODULES[sensor_config["name"]] = configure_sensor_module(sensor_config)
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 588, in configure_sensor_module
    return sensor_module.Sensor(sensor_config)
  File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/modules/bme280.py", line 31, in __init__
    self.calib = bme280.load_calibration_params(self.bus, self.address)
AttributeError: 'module' object has no attribute 'load_calibration_params'
`
`
sensor_modules:
  - name: bme280
    module: bme280
    i2c_bus_num: 1
    chip_addr: 0x76

sensor_inputs:
  - name: bme280_temperature
    module: bme280
    interval: 15 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 4 # number of digits to be round

  - name: bme280_pressure
    module: bme280
    interval: 10 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 4 # number of digits to be round

  - name: bme280_humidity
    module: bme280
    interval: 10 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 4 # number of digits to be round
`

adrian-vlad, 2020-10-29:
Did you try running it with python3?

I've only tried it with python3. If you cannot switch to python3, I will try to reproduce your setup with python2 and see if I can figure what the problem is.

tkranz, 2020-10-30:
ok, with python3 it works but receive wrong values, could you post your setup here ?

root@mqtt1:/home/pi# python3 -m pi_mqtt_gpio.server /home/pi/pi-mqtt-gpio.yml
2020-10-30 13:38:36,227 mqtt_gpio (INFO): Startup
/usr/local/lib/python3.7/dist-packages/pi_mqtt_gpio/server.py:730: DeprecationWarning: fractions.gcd() is deprecated. Use math.gcd() instead.
  cycle_time = reduce(lambda x, y: gcd(x, y), arr)
2020-10-30 13:38:39,888 mqtt_gpio (INFO): Connected to the MQTT broker with protocol v3.1.1.
2020-10-30 13:38:39,892 mqtt_gpio (INFO): Subscribed to topic: 'mqtt1/output/gpio20/set'
2020-10-30 13:38:39,895 mqtt_gpio (INFO): Subscribed to topic: 'mqtt1/output/gpio20/set_on_ms'
2020-10-30 13:38:39,898 mqtt_gpio (INFO): Subscribed to topic: 'mqtt1/output/gpio20/set_off_ms'
2020-10-30 13:38:39,912 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'bme280_temperature' value 24.62
2020-10-30 13:38:39,937 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'bme280_pressure' value 24.62
2020-10-30 13:38:39,961 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'bme280_humidity' value 24.63

adrian-vlad, 2020-10-30:
`sensor_modules:
- name: bme280
  module: bme280
  i2c_bus_num: 1
  chip_addr: 0x76

sensor_inputs:
- name: bme280_temperature
  module: bme280
  interval: 10
  digits: 4
  type: temperature
  unit_of_measurement: "°C"

- name: bme280_humidity
  module: bme280
  interval: 10
  digits: 5
  type: humidity
  unit_of_measurement: "%"

- name: bme280_pressure
  module: bme280
  interval: 10
  digits: 6
  type: pressure
  unit_of_measurement: "hPa"`
This is the relevant part

tkranz, 2020-10-30:
here the working example, no i will integrate it into openhab 👍 

sensor_modules:
  - name: bme280_sensor
    module: bme280
    i2c_bus_num: 1
    chip_addr: 0x76

sensor_inputs:
  - name: bme280_temperature
    module: bme280_sensor
    interval: 15 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 2 # number of digits to be round
    type: temperature

  - name: bme280_pressure
    module: bme280_sensor
    interval: 15 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 2 # number of digits to be round
    type: pressure

  - name: bme280_humidity
    module: bme280_sensor
    interval: 15 #interval in seconds, that a value is read from the sensor and a update is published
    digits: 2 # number of digits to be round
    type: humidity

closed 2020-10-30 by tkranz