flyte/mqtt-io #144: Feature request: could you add support for ADS1115 analog/digital converter ?

tkranz, 2020-10-30:
I see there is support for MPC3008, is it possible to add ADS115 as well ?

thanks
Thomas

BenjiU, 2020-11-05:
Hi,

may you check my implementation on branch `feature/ADS1115`? I don't have the chip here, so i can't test it.

```
mqtt:
  host: mqtt
  port: 1883
  user: ""
  password: ""
  topic_prefix: topic

sensor_modules:
  - name: ads1115_sensor
    module: ads1115

sensor_inputs:
  - name: ads1115_voltage
    module: ads1115_sensor
    interval: 20  # measurement every 20sec
    channel: CH0 # measure on CH0 of ADS1115
```

tkranz, 2020-11-09:
Great ! will test it today and let you now the results.

tkranz, 2020-11-10:
have testet the branch and it works fine, only thing i couldn't find is how to set the i2c address.

here my configuration:

`
sensor_modules:
  - name: ads1115_sensor
    module: ads1115

sensor_inputs:
  - name: ads1115_voltage0
    module: ads1115_sensor
    interval: 20  # measurement every 20sec
    channel: CH0 # measure on CH0 of ADS1115

  - name: ads1115_voltage1
    module: ads1115_sensor
    interval: 20  # measurement every 20sec
    channel: CH1 # measure on CH1 of ADS1115

  - name: ads1115_voltage2
    module: ads1115_sensor
    interval: 20  # measurement every 20sec
    channel: CH2 # measure on CH2 of ADS1115

  - name: ads1115_voltage3
    module: ads1115_sensor
    interval: 20  # measurement every 20sec
    channel: CH3 # measure on CH3 of ADS1115
`
and the log 

`root@mqtt2:~# tail -20 /var/log/pi-mqtt-gpio.log
2020-11-10 08:57:48,945 mqtt_gpio (WARNING): MCP3008: value 4419
2020-11-10 08:57:48,958 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'ads1115_voltage1' value 4591
2020-11-10 08:57:48,961 mqtt_gpio (WARNING): MCP3008: Reading from channel 2
2020-11-10 08:57:48,975 mqtt_gpio (WARNING): MCP3008: value 4538
2020-11-10 08:57:48,988 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'ads1115_voltage2' value 4294
2020-11-10 08:57:48,991 mqtt_gpio (WARNING): MCP3008: Reading from channel 3
2020-11-10 08:57:49,004 mqtt_gpio (WARNING): MCP3008: value 4496
2020-11-10 08:57:49,017 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'ads1115_voltage3' value 4713
2020-11-10 08:58:08,913 mqtt_gpio (WARNING): MCP3008: Reading from channel 0
2020-11-10 08:58:08,925 mqtt_gpio (WARNING): MCP3008: value 22706
2020-11-10 08:58:08,937 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'ads1115_voltage0' value 22687
2020-11-10 08:58:08,940 mqtt_gpio (WARNING): MCP3008: Reading from channel 1
2020-11-10 08:58:08,953 mqtt_gpio (WARNING): MCP3008: value 4481
2020-11-10 08:58:08,966 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'ads1115_voltage1' value 4358
2020-11-10 08:58:08,969 mqtt_gpio (WARNING): MCP3008: Reading from channel 2
2020-11-10 08:58:08,982 mqtt_gpio (WARNING): MCP3008: value 4602
2020-11-10 08:58:08,995 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'ads1115_voltage2' value 4574
2020-11-10 08:58:08,999 mqtt_gpio (WARNING): MCP3008: Reading from channel 3
2020-11-10 08:58:09,012 mqtt_gpio (WARNING): MCP3008: value 4347
2020-11-10 08:58:09,025 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'ads1115_voltage3' value 4327
`

closed 2020-11-11 by tkranz

BenjiU, 2020-11-11:
I reopen it for i2c address selection, until the pullrequest is done...
Are your measured values plausible?

tkranz, 2020-11-11:
[image]

ok, yes i have 4 capacitive humidity sensor connected ! looks fine so far.