flyte/mqtt-io #148: Reading multiple PCF8574 on one bus leads to slow reactions to changes

tomilein85, 2020-12-09:
If I am only using one PCF8574 on the bus, the pi-mqtt-gpio is working great. Looking at it with the logic analyzer, it is reading the data twice (with a delay of 0.1mSec) each 10mSec from the PCF8574 and reacts to any button change I make (using it with push buttons).
Strangly it is also in this configuration reading the bus sometimes only once (which i would expect or twice with no delay).

As soon as I add multiple PCF8574 (in the configuration, they are connected the whole time to the bus) the reaction to push button inputs is no more reliable enough. Looking at my logic analyzer, I see that he is still reading data all 10mSec from the bus, BUT, he is not doing this in the correct order for the connected devices. He is reading the first device multiple times (over 40mSec) and then switching to the second device and reading this.

And it gets even worse, if I am using multiple I2C buses on the RPI. Then I have pauses of 110mSec between some iterations.
Attached are some outputs from the logic analyzer and the different configurations I used for this.

Please advise if I am doing something wrong in the configuration.
[config_1bus_1dev.txt](https://github.com/flyte/pi-mqtt-gpio/files/5665241/config_1bus_1dev.txt)
[config_2bus.txt](https://github.com/flyte/pi-mqtt-gpio/files/5665242/config_2bus.txt)
[config_1bus_3dev.txt](https://github.com/flyte/pi-mqtt-gpio/files/5665243/config_1bus_3dev.txt)

[2busses_logic.txt](https://github.com/flyte/pi-mqtt-gpio/files/5665234/2busses_logic.txt)
[1bus_1dev_logic.txt](https://github.com/flyte/pi-mqtt-gpio/files/5665235/1bus_1dev_logic.txt)
[1bus_3dev_logic.txt](https://github.com/flyte/pi-mqtt-gpio/files/5665236/1bus_3dev_logic.txt)

BenjiU, 2020-12-09:
Yes, this looks really interesting :-) May you attache the logs from pi_mqtt_gpio? If it really does that confusing reading or maybe the smbus lib does something strange?

It noticed, there is a smbus2 lib for python. Maybe we test it with the new lib in pfc module?

BenjiU, 2020-12-09:
Or we use this one: https://pypi.org/project/pcf8574-io/
As you're able to record i2c bus signals, I think you may change the python modules for testing or do you need assistance?

tomilein85, 2020-12-09:
Will try both on friday/saturday and post back. I am not a pyhton genius, but I hope I can figure it out how to do what I want from it.

tomilein85, 2020-12-13:
So, I have tried it with the pcf8574-io pyhton thing (only one bus, need to extend the pcf8574 also to a second bus). But what I see is, that for each input, the tool is sending a own request towards the pcf8574, so if I configure 4 pins on a pcf8754, it will send out 8 requests (and will use 40msec to do so). Instead of using only one request to get out all bits of one request.
[1bus_3dev_pcf_io.txt](https://github.com/flyte/pi-mqtt-gpio/files/5684797/1bus_3dev_pcf_io.txt)

BenjiU, 2020-12-17:
Okay, thanks for your effort. The pcf8574 libs (all of them :-)) don't store the register intern and provide all request for pins inside. This would be an optimization, I also thought about: read the i2c register once, and use the information for all pins/inputs, that are "updated" within e.g. 20ms. But this is a different point, I think.

May you help me understand your captures? 0x79 (should be decimal 79, 0x4F hex, I think) is the i2c address pcf8574 with your 4 configured inputs? Is the first the read request and the second line the "response"? Allways 0xFF, so everything is "high"?

```
0.034705375000000,I2C,Setup Read to [0x79] + ACK   # Address and ReadFlag
0.034849375000000,I2C,0xFF + NAK                           # Response/data "byte"

0.045464750000000,I2C,Setup Read to [0x79] + ACK
0.045608750000000,I2C,0xFF + NAK

0.046107375000000,I2C,Setup Read to [0x79] + ACK
0.046251375000000,I2C,0xFF + NAK

0.056829125000000,I2C,Setup Read to [0x79] + ACK
0.056973125000000,I2C,0xFF + NAK

0.057466500000000,I2C,Setup Read to [0x79] + ACK
0.057610500000000,I2C,0xFF + NAK

0.068190375000000,I2C,Setup Read to [0x79] + ACK
0.068334375000000,I2C,0xFF + NAK

0.068827000000000,I2C,Setup Read to [0x79] + ACK
0.068971000000000,I2C,0xFF + NAK

0.079591500000000,I2C,Setup Read to [0x79] + ACK
0.079735500000000,I2C,0xFF + NAK
```

Have you tried to match your i2c capture to the py_mqtt_gpio log files? Can you see the reads in python logs with a "distance" of also 10ms e.g.?

tomilein85, 2020-12-19:
0x79 is hex which includes the read bit, a bit stupid the logic analyzer in this setup. So in the analyzer 0x79 ==> 0x3c as real address.
Could you please point me at the location of the detailed log files?
From what I understand, there would be the need to cache the read or at least change the way py_mqtt_gpio works. Currently it will poll each pin a first time and then wait 10msec, If it would poll all pins a first time and after 10ms all pins again, this could be a possible solution (the i2c read does not take that long).

tomilein85, 2020-12-27:
I have changed the code to no longer read the data in a serial way (it is still using pin mode instead of port mode) and a test at the house worked (the sleep between data reads is dependent on the last read time for a pin, in the default implementation it is always 10mSecs). But now I saw, flyte has actually a pcf library which is using the port mode instead of the pin mode (a read of 8 pins is currently taking 4msecs). So i will try to use his implementation to get down to 0.5msecs for a whole PCF8574. This will end up in a mixed version. Reads will use the flyte pcf implementation writes will use the generic pcf implementation.

tomilein85, 2020-12-28:
I have now tried to push the changes back, so i will close the issue here.
I decided not to go to the port reads instead of pin reads, since this would mess up a lot in the code and could also make troubles with 16bit expanders, which I do not have to verify them.

closed 2020-12-28 by tomilein85