flyte/mqtt-io #151: Broken in at least python 3.9

w4tsn, 2021-01-11:
`import fractions from gcd` was deprecated in python 3.6 and doesn't work in python 3.9 anymore.

The fix is to `import math` and exchange `gcd` by `math.gcd`. There are only two ocurences in server.py. I don't know how to incomporate this into the project without breaking it for older python installations though.

alive-corpse, 2021-01-18:
Python 3.7.3 from raspbian repository also shows warning like this:
```
/opt/pi-mqtt-gpio/lib/python3.7/site-packages/pi_mqtt_gpio-0.5.2-py3.7.egg/pi_mqtt_gpio/server.py:730: DeprecationWarning: fractions.gcd() is deprecated. Use math.gcd() instead.         
  cycle_time = reduce(lambda x, y: gcd(x, y), arr
```

The problem can be solved just by replacing string "from fractions import gcd" to string "from math import gcd" in the begginig of server.py file.

flyte, 2021-01-18:
I've added a backwards-compatible import to the `develop` branch, thanks. It'll be in the next release :+1:

closed 2021-01-18 by flyte