flyte/mqtt-io #153: Adding support for lcd1604/2004 via pcf8574

alive-corpse, 2021-01-19:
Hi! It's amazing project, I want to write some article about it in my blog.

I think supporting for text lcd displays with i2c pcf8574 modules will be usefull and not so hard to make.

Suggested example of config:
```yaml
gpio_modules:
  - name: LCD # some/mqtt/prefix/output/LCD
    module: pcf8574lcd
    i2c_bus_num: 1
    chip_addr: 0x20
    width: 16 # count of symbols in one line
    backlight: yes # yes/no/manual/timeout
    # timeout: shut off after getting new data from mqtt + %timeout% seconds)
    # manual: turning on/off when topic some/mqtt/prefix/output/LCD/backlight is changing to 1/0

....

digital_outputs:
  - name: lcdline1 # some/mqtt/prefix/output/LCD/line1
    module: pcf8574lcd
    line: 1 # number of line (row)
    scroll: true # scroll text in line if it's wider than screen
    scroll_speed: 0.8 # speed of scroll in seconds or some cycle counts (int instead of float in this case)

digital_outputs:
  - name: lcdline2 # some/mqtt/prefix/output/LCD/line2
    module: pcf8574lcd
    line: 2
    scroll: false
```

Usecases:
Streaming of ip address, temperature, cpu loads, disk utilisation or some other technical information.
Streaming of internet radio station or mp3 file name/timings from mpd daemon or moc player.
Streaming of sensors values/relays statuses connected to gpio or weather forecast data.
And any other stuff like this.

Library that can be possible used: https://github.com/sterlingbeason/LCD-1602-I2C

BenjiU, 2021-01-19:
Maybe, stream is better then digital_output:

```
stream_modules:
  - name: pcf8574lcd
...

stream_writes:
  - name: lcdline1
    module: pcf8574lcd
    line: 1 # number of line (row)
    scroll: true # scroll text in line if it's wider than screen
    scroll_speed: 0.8 # speed of scroll in seconds or some cycle counts (int instead of float in this case)
```

flyte, 2021-02-25:
Yeah, this is an interesting proposition. I'll have to have a think about how we layer functionality on top of other modules. For example, we could have an LCD stream module that specifies `digital_output` pins to be connected to the display. Not sure whether we'd be able to pump the pins fast enough though, if any kind of speedy timing was required.

flyte, 2021-02-25:
Oh, and also

> I want to write some article about it in my blog

That would be great! It'd be good for you to test out the new asyncio version #178 though, so that any information to write is up to date. Let me know on discord (link on readme) or in the "Discussions" bit on this repo if you need any help.