flyte/mqtt-io #155: 1Wire Sensor IndexError: list index out of range

JuergenKnill, 2021-01-24:
We are use an RPi Zero and get  often this index error

 Jan 23 14:26:55 serverraum python[18640]: Exception in thread pi-mqtt-gpio_SensorReader:
Jan 23 14:26:55 serverraum python[18640]: Traceback (most recent call last):
Jan 23 14:26:55 serverraum python[18640]:   File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
Jan 23 14:26:55 serverraum python[18640]:     self.run()
Jan 23 14:26:55 serverraum python[18640]:   File "/usr/lib/python2.7/threading.py", line 754, in run
Jan 23 14:26:55 serverraum python[18640]:     self.__target(*self.__args, **self.__kwargs)
Jan 23 14:26:55 serverraum python[18640]:   File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/server.py", line 598, in sensor_timer_thread
Jan 23 14:26:55 serverraum python[18640]:     value = sensor.get_value(sens_conf)
Jan 23 14:26:55 serverraum python[18640]:   File "/usr/local/lib/python2.7/dist-packages/pi_mqtt_gpio/modules/ds18b.py", line 39, in get_value
Jan 23 14:26:55 serverraum python[18640]:     temperature = self.sensor.get_temperature()
Jan 23 14:26:55 serverraum python[18640]:   File "/usr/local/lib/python2.7/dist-packages/w1thermsensor/core.py", line 328, in get_temperature
Jan 23 14:26:55 serverraum python[18640]:     value = self.raw_sensor_count
Jan 23 14:26:55 serverraum python[18640]:   File "/usr/local/lib/python2.7/dist-packages/w1thermsensor/core.py", line 264, in raw_sensor_count
Jan 23 14:26:55 serverraum python[18640]:     bytes = self.raw_sensor_strings[1].split()
Jan 23 14:26:55 serverraum python[18640]:   File "/usr/local/lib/python2.7/dist-packages/w1thermsensor/core.py", line 243, in raw_sensor_strings
Jan 23 14:26:55 serverraum python[18640]:     if data[0].strip()[-3:] != "YES":
Jan 23 14:26:55 serverraum python[18640]: IndexError: list index out of range

it works for a view hours an then this error we get! after the restart the same one!

sometimes it works for 1 hour and other for 6  hours 

we have set in the 
/usr/local/lib/python2.7/dist-packages/w1thermsensor/core.py file  the RETRY_DELAY_SECONDS from 1 sec to 2 sec and then to 3 sec !

the result was better but we get then the error later it do not save the problem

Kind regards

Jknill

BenjiU, 2021-01-24:
Hi, I think this error was already fixed in w1thermsensor. https://github.com/timofurrer/w1thermsensor/issues/96
Maybe you can update your Python bibs?

flyte, 2021-01-24:
Looks like you're using global python libraries, so try a `sudo pip install --upgrade w1thermsensor`.

As a general piece of unsolicited advice, I recommend using a virtualenv (and Python 3 if you can) to install your Python packages into. There's a nice article on it here: https://realpython.com/python-virtual-environments-a-primer/

JuergenKnill, 2021-01-24:
it looks i habe the actual w1thermsensor

root@serverraum:/home/pi# pip install --upgrade w1thermsensor
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: w1thermsensor in /usr/local/lib/python2.7/dist-packages (1.3.0)
Requirement already satisfied, skipping upgrade: click in /usr/local/lib/python2.7/dist-packages (from w1thermsensor) (7.1.2)

flyte, 2021-01-24:
Ah, apparently that bug fix is only in version 2.0.0a2 https://pypi.org/project/w1thermsensor/#history

You'd need to do a `pip install --upgrade --pre w1thermsensor` to get that, but it might be worth commenting on https://github.com/timofurrer/w1thermsensor/issues/96 to ask if they'll include that bugfix in the 1.3.x stable branch.

I'm going to close this issue for the moment, as I believe it's an upstream issue. Feel free to continue to comment though if you think there's something we can do.

closed 2021-01-24 by flyte

Maerchensarg, 2021-01-24:
Getting the same error now.

pip install --upgrade --pre w1thermsensor
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
  Requirement already up-to-date: w1thermsensor in /usr/local/lib/python2.7/dist-packages (1.3.0)
  Requirement already satisfied, skipping upgrade: click in /usr/local/lib/python2.7/dist-packages (from w1thermsensor) (7.1.2)

flyte, 2021-01-24:
It looks like `w1thermsensor` version `2.0.0a2` only supports Python 3.

Python 2:
```
$ pip install --pre w1thermsensor
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Collecting w1thermsensor
  Downloading w1thermsensor-1.3.0-py2.py3-none-any.whl (14 kB)
Collecting click
  Using cached click-7.1.2-py2.py3-none-any.whl (82 kB)
Installing collected packages: click, w1thermsensor
Successfully installed click-7.1.2 w1thermsensor-1.3.0
```

Python 3:
```
$ pip install --pre w1thermsensor
Collecting w1thermsensor
  Downloading https://files.pythonhosted.org/packages/00/d6/4b06ece24dd0d0e11d3a8cb0710c1b06adfbd2dee7c2981555debb5a8dc8/w1thermsensor-2.0.0a2-py2.py3-none-any.whl
Collecting click (from w1thermsensor)
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/7d/5c/104f7ef490e1a585c748a788b331bec1f4948e1a474bb6be7c73a5e8bc10/click-8.0.0a1-py3-none-any.whl (89kB)
    100% |████████████████████████████████| 92kB 2.9MB/s 
Installing collected packages: click, w1thermsensor
Successfully installed click-8.0.0a1 w1thermsensor-2.0.0a2
```

Maerchensarg, 2021-01-24:
Alright. Thanks again. What worked for me (maybe with sudo):

apt-get install python3-pip

pip uninstall w1thermsensor
pip uninstall pi-mqtt-gpio

pip3 install --pre w1thermsensor
pip3 install pi-mqtt-gpio

python3 -m pi_mqtt_gpio.server config.yml

-

I am running this with output right now. Lets see how long. ^^

flyte, 2021-01-24:
Can you do a `pip3 freeze` and paste the output?

bsimmo, 2021-01-24:
if data[0].strip()[-3:] != "YES": 

shows it's not using the fix.
What version installed ?

if it was the fix (but failed) then it would read
not data or data[0].strip()[-3:] != "YES"

Maerchensarg, 2021-01-25:
Sure:

pi@Project-X:~ $ pip3 freeze
Adafruit-DHT==1.4.0
asn1crypto==0.24.0
Cerberus==1.3.2
certifi==2018.8.24
chardet==3.0.4
click==8.0.0a1
cryptography==2.6.1
entrypoints==0.3
enum34==1.1.10
idna==2.6
keyring==17.1.1
keyrings.alt==3.1.1
paho-mqtt==1.5.1
pi-mqtt-gpio==0.5.6
pycrypto==2.6.1
PyGObject==3.30.4
python-apt==1.8.4.3
pyxdg==0.25
PyYAML==5.4.1
requests==2.21.0
RPi.GPIO==0.7.0
SecretStorage==2.3.1
six==1.12.0
ssh-import-id==5.7
urllib3==1.24.1
w1thermsensor==2.0.0a2

flyte, 2021-01-25:
OK, you've got the latest code, so at this point I suggest cutting out this software entirely and testing directly with the w1thermsensor examples from their readme. If you still have problems, raise an issue on their repository.

Unfortunately they've slightly unhelpfully locked the issue we were discussing this on, but they don't seem to have an actually released version of their software that doesn't contain this bug (correct me if I'm wrong). We don't currently have an option to install prerelease packages for our modules, so hopefully they can make a stable release soon.

JuergenKnill, 2021-01-25:
Ok 

i have done this but now i cant start

root@serverraum:/etc/systemd/system# python3 -m pi_mqtt_gpio.server
/etc/pi-mqtt-gpio.yml
2021-01-25 12:52:50,216 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.7/dist-packages/pi_mqtt_gpio/server.py",
line 1269, in <module>
    main(args)
  File "/usr/local/lib/python3.7/dist-packages/pi_mqtt_gpio/server.py",
line 1085, in main
    SENSOR_MODULES[sensor_config["name"]] =
configure_sensor_module(sensor_config)
  File "/usr/local/lib/python3.7/dist-packages/pi_mqtt_gpio/server.py",
line 590, in configure_sensor_module
    return sensor_module.Sensor(sensor_config)
  File
"/usr/local/lib/python3.7/dist-packages/pi_mqtt_gpio/modules/ds18b.py",
line 28, in __init__
    for w1_sensor_type, w1_sensor_name in
W1ThermSensor.TYPE_NAMES.items():
AttributeError: type object 'W1ThermSensor' has no attribute
'TYPE_NAMES'
root@serverraum:/etc/systemd/system#

The attribute TYPE_NAMES

sensor_modules:
  - name: ds18b22_sensor_srv
    module: ds18b
    type: DS18B20
    address: 01131e94465d

Kind Regards

Jürgen Knill

  
  

Knill Software & Kommunikation
Bildstock 9/2
88085 Langenargen
 
Tel.:+49 (0) 7543/953180
Fax :+49 (0) 7543/953182
J.Knill@knill.de
www.knill.de

Knill Software & Kommunikation * Bildstock 9/2 * 88085 Langenargen  
Tel.: +49 7543/953180 * Fax: +49 7543/953182 * email: info@knill.de *
Internet: www.knill.de  
UST-IdNr.:DE175009989 * Steuer-Nr.:61157/38804 
  
  >>> Maerchensarg <notifications@github.com> Sonntag, 24. Januar 2021
19:19 >>>
Alright. Thanks again. What worked for me (maybe with sudo):
apt-get install python3-pip
pip uninstall w1thermsensor
pip uninstall pi-mqtt-gpio
pip3 install --pre w1thermsensor
pip3 install pi-mqtt-gpio
python3 -m pi_mqtt_gpio.server config.yml
	●  
I am running this with output right now. Lets see how long. ^^
—
(
https://github.com/flyte/pi-mqtt-gpio/issues/155#issuecomment-766407014)
(

BenjiU, 2021-01-25:
May you create a new Issue with this topic, please?
It looks like, they changed the w1thermsensor api a lot. So we have to adapt to their new version.

Thanks

flyte, 2021-01-25:
OK, it looks like our module needs to change in order to support v2.x of the w1thermsensor package. I'm reluctant to do this before they release a stable version, since pre-release requirements won't work with our dependency installation system.

Unfortunately this means we're stuck for the moment, until the w1thermsensor guys backport the fix to 1.3.x or release their 2.x version properly. The latter will still mean that we have to drop support for Python 2 when using this module unfortunately.

flyte, 2021-01-25:
> May you create a new Issue with this topic, please?

I've made a new issue for this at #157