flyte/mqtt-io #156: Could not find sensor of type DS18S20 with id XYZ

Maerchensarg, 2021-01-24:
Hi. Thanks for this great project. Somehow I don't understand something or run into a mistake somewhere else. I am trying to connect a 1-wire temperature sensor.

pi@raspberrypi:~ $ cat /sys/bus/w1/devices/28-01131e858aef/w1_slave
  48 01 4b 46 7f ff 0c 10 96 : crc=96 YES
  48 01 4b 46 7f ff 0c 10 96 t=20500

sensor_modules:
  - name: ds18b22_sensor
    module: ds18b
    type: DS18S20
    address: 01131e858aef

sensor_inputs:
  - name: ds18b22_temperature
    module: ds18b22_sensor
    interval: 10
    digits: 2

Error:
  Could not find sensor of type DS18S20 with id 01131e858aef
  Please check cabling and check your /boot/config.txt for
  dtoverlay=w1-gpio

My config includes the above adjustment. Any idea what is missing?

flyte, 2021-01-24:
I'm afraid I haven't used this sensor before, @BenjiU wrote this module so maybe he has some insight that I don't.

I'd suggest ruling out any issues with pi-mqtt-gpio by trying to read from the device using the https://github.com/timofurrer/w1thermsensor library directly (the one we use). Read through the README on that project and try out the examples it gives.

Maerchensarg, 2021-01-24:
Thanks for your hint, I got the wrong sensor. Its working with "DS18B20".

closed 2021-01-24 by Maerchensarg