flyte/mqtt-io #157: Update ds18b module to use w1thermsensor v2.x

flyte, 2021-01-25:
w1thermsensor has [just had a v2.0.0 release](https://github.com/timofurrer/w1thermsensor/issues/102), so we can update our module to support that now, in order to avoid the bug mentioned in https://github.com/flyte/pi-mqtt-gpio/issues/155.

Unfortunately this means that using the ds18b module will no longer be supported on Python 2. I may introduce a Python version check system so that our users are informed when they try to configure a Python 3 only module (or just release the asyncio Python 3 only update finally).

bsimmo, 2021-01-25:
Thanks for getting v2.0 released, been waiting for that for some time.

JuergenKnill, 2021-02-23:
Hello 
can you tell me a timeline when i can use the W1termsensor with the v2.0?

Kind regards 
Jürgen

flyte, 2021-02-23:
@JuergenKnill I've updated it in the new asyncio version of this project. It's quite close to being released, but it'd be helpful if you could test it out on the `feature/asyncio` branch. It should be compatible with most existing configs. You can run it with `python -m mqtt_io config.yml`.

I no longer intend to update this version of the project, so I'm going to close this issue since it's been done on the new version.

closed 2021-02-23 by flyte

flyte, 2021-02-23:
Instructions for testing out the asyncio version are in https://github.com/flyte/pi-mqtt-gpio/issues/141

JuergenKnill, 2021-03-08:
Hello 

now i have time to test but we get an error 

root@serverraum:/usr/local/lib/python3.7/dist-packages# python3 -m
asyncio_mqtt /etc/pi-mqtt-gpio.yml
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.7/runpy.py", line 142, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.7/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File
"/usr/local/lib/python3.7/dist-packages/asyncio_mqtt/__init__.py", line
3, in <module>
    from .client import Client, Will
  File "/usr/local/lib/python3.7/dist-packages/asyncio_mqtt/client.py",
line 2, in <module>
    import asyncio
  File "/usr/lib/python3.7/asyncio/__init__.py", line 8, in <module>
    from .base_events import *
  File "/usr/lib/python3.7/asyncio/base_events.py", line 18, in
<module>
    import concurrent.futures
  File "/usr/lib/python3.7/concurrent/futures/__init__.py", line 8, in
<module>
    from concurrent.futures._base import (FIRST_COMPLETED,
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 7, in
<module>
    import logging
  File "/usr/lib/python3.7/logging/__init__.py", line 26, in <module>
    import sys, os, time, io, traceback, warnings, weakref,
collections.abc
  File "/usr/lib/python3.7/traceback.py", line 5, in <module>
    import linecache
  File "/usr/lib/python3.7/linecache.py", line 11, in <module>
    import tokenize
  File "/usr/lib/python3.7/tokenize.py", line 33, in <module>
    import re
  File "/usr/lib/python3.7/re.py", line 143, in <module>
    class RegexFlag(enum.IntFlag):
AttributeError: module 'enum' has no attribute 'IntFlag'

kind regards

Jürgen Knill

  
  

Knill Software & Kommunikation
Bildstock 9/2
88085 Langenargen
 
Tel.:+49 (0) 7543/953180
Fax :+49 (0) 7543/953182
J.Knill@knill.de
www.knill.de

Knill Software & Kommunikation * Bildstock 9/2 * 88085 Langenargen  
Tel.: +49 7543/953180 * Fax: +49 7543/953182 * email: info@knill.de *
Internet: www.knill.de  
UST-IdNr.:DE175009989 * Steuer-Nr.:61157/38804 
  
  >>> Ellis Percival <notifications@github.com> Dienstag, 23. Februar
2021 15:43 >>>
Instructions for testing out the asyncio version are in #141
( https://github.com/flyte/pi-mqtt-gpio/issues/141) 
—
(
https://github.com/flyte/pi-mqtt-gpio/issues/157#issuecomment-784250859)
(

JuergenKnill, 2021-03-19:
Hello

i have do the install for the asyncio but we get the error

***@***.***:/home/pi# pip3 install --pre mqtt-io
Looking in indexes: https://pypi.org/simple,
https://www.piwheels.org/simple
Collecting mqtt-io
  Using cached
https://files.pythonhosted.org/packages/22/d1/99a456bd6ecaa8eb0cb8114fb069eadefd3c96305763fe358658d3e81a67/mqtt_io-2.1.1-py3-none-any.whl
Collecting Cerberus<2.0.0,>=1.3.2 (from mqtt-io)
  Using cached
https://www.piwheels.org/simple/cerberus/Cerberus-1.3.2-py3-none-any.whl
Collecting typing-extensions<4.0.0,>=3.7.4 (from mqtt-io)
  Using cached
https://files.pythonhosted.org/packages/60/7a/e881b5abb54db0e6e671ab088d079c57ce54e8a01a3ca443f561ccadb37e/typing_extensions-3.7.4.3-py3-none-any.whl
Collecting asyncio-mqtt<0.9.0,>=0.8.1 (from mqtt-io)
  Using cached
https://files.pythonhosted.org/packages/7a/c7/02372db3f34f15a1aed25a0f71505d9cd54aaed54c0fef3211c2cc7ba1af/asyncio_mqtt-0.8.1-py3-none-any.whl
Collecting PyYAML>=5.3.1 (from mqtt-io)
  Using cached
https://files.pythonhosted.org/packages/a0/a4/d63f2d7597e1a4b55aa3b4d6c5b029991d3b824b5bd331af8d4ab1ed687d/PyYAML-5.4.1.tar.gz
  Installing build dependencies ... done
    Complete output from command python setup.py egg_info:
    running egg_info
    creating pip-egg-info/PyYAML.egg-info
    writing pip-egg-info/PyYAML.egg-info/PKG-INFO
    writing dependency_links to
pip-egg-info/PyYAML.egg-info/dependency_links.txt
    writing top-level names to
pip-egg-info/PyYAML.egg-info/top_level.txt
    writing manifest file 'pip-egg-info/PyYAML.egg-info/SOURCES.txt'
    cythoning yaml/_yaml.pyx to yaml/_yaml.c
    Traceback (most recent call last):
	  File "<string>", line 1, in <module>
	  File "/tmp/pip-install-il6mcq6l/PyYAML/setup.py", line 295, in
<module>
	    python_requires='>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*,
!=3.4.*, !=3.5.*',
	  File
"/tmp/pip-build-env-lmkr16tl/lib/python3.7/site-packages/setuptools/__init__.py",
line 153, in setup
	    return distutils.core.setup(**attrs)
	  File "/usr/lib/python3.7/distutils/core.py", line 148, in
setup
	    dist.run_commands()
	  File "/usr/lib/python3.7/distutils/dist.py", line 966, in
run_commands
	    self.run_command(cmd)
	  File "/usr/lib/python3.7/distutils/dist.py", line 985, in
run_command
	    cmd_obj.run()
	  File
"/tmp/pip-build-env-lmkr16tl/lib/python3.7/site-packages/setuptools/command/egg_info.py",
line 299, in run
	    self.find_sources()
	  File
"/tmp/pip-build-env-lmkr16tl/lib/python3.7/site-packages/setuptools/command/egg_info.py",
line 306, in find_sources
	    mm.run()
	  File
"/tmp/pip-build-env-lmkr16tl/lib/python3.7/site-packages/setuptools/command/egg_info.py",
line 541, in run
	    self.add_defaults()
	  File
"/tmp/pip-build-env-lmkr16tl/lib/python3.7/site-packages/setuptools/command/egg_info.py",
line 577, in add_defaults
	    sdist.add_defaults(self)
	  File "/usr/lib/python3.7/distutils/command/sdist.py", line
228, in add_defaults
[... 102 lines cut ...]
  
  

Knill Software & Kommunikation
Bildstock 9/2
88085 Langenargen
 
Tel.:+49 (0) 7543/953180
Fax :+49 (0) 7543/953182
***@***.***
www.knill.de

Knill Software & Kommunikation * Bildstock 9/2 * 88085 Langenargen  
Tel.: +49 7543/953180 * Fax: +49 7543/953182 * email: ***@***.*** *
Internet: www.knill.de  
UST-IdNr.:DE175009989 * Steuer-Nr.:61157/38804 
  
  >>> Ellis Percival ***@***.***> Dienstag, 23. Februar
2021 15:43 >>>
Instructions for testing out the asyncio version are in #141
( https://github.com/flyte/pi-mqtt-gpio/issues/141) 
—
(
https://github.com/flyte/pi-mqtt-gpio/issues/157#issuecomment-784250859)
(