flyte/mqtt-io #158: Sensors stop updating/publishing

Arsenal10108, 2021-01-27:
**Describe the bug**
I have a LM75AD temp sensor and a bh1750 and several reed switches hooked up. The sensors are connected to the i2c bus obviously and the reed switches are just tied to gpio pins. After a random amount of time the sensors will just stop reporting. I don't see any errors in the log file and everything seems to be working fine otherwise. Only way to get the sensors to come back is to reboot the pi. I am not sure where else to check for errors as to what's causing this unless there's some other log file I am missing. Could this be an i2c issue since the reed switches are working normally?

**Expected behavior**
A clear and concise description of what you expected to happen.
I expect the sensors to continue to report
**Error messages and traceback**
Can't find any, simply just stops

**Hardware**
 - Platform: pi2b with the lastest raspbian

**System:**
 - OS: raspbian buster
 - Python version: 2.7.16
 - User you're running as: root under virrualenv
 - Using a virtualenv?: Yes

flyte, 2021-01-31:
This will need some extra investigation to figure out what's going on. It might be worth enabling debug logging. Have a look here https://github.com/flyte/pi-mqtt-gpio/wiki/Logging-Configuration and change the `INFO` level to `DEBUG`.

Otherwise, it might be worth trying the new asyncio version of the code to see if that has the same problem. It's on `feature/asyncio`. If you want some help with that, then ask on Discord (link at the top of the readme) and I'll help out if I'm around.

Arsenal10108, 2021-02-07:
I am beginning to think this is not an issue with the code at all. I was unaware but apparently i2c has very limited cable length as it runs at 100khz. I relocated my LM75AD temp sensor about 3M from the Pi and I am thinking that is the issue. Weird thing is in testing I could still use adafruits demo sensors python scripts to still pull the sensor readings so it made me think that it wasn't the sensors that were the issue. I it possible that I am right on the fringe of reliability and the sensor dips out which causes something in the code to drop the sensors and not pick it back up?? dropt the i2c sensors only I should say. 

For testing I dropped my i2c Bus rate down to 10khz to see if that increased the reliability.

flyte, 2021-02-25:
Ah yes, I've had issues with line-length on I2C in the past. Shielding the wire helped quite a lot with that.

The new version is easy to install now, so I'd be interested to know how it handles the issues you're seeing. Instructions are on #178

Arsenal10108, 2021-03-01:
This has been resolved by adding in a Adafruit LTC4311 i2c extender/active terminator. Cable run was about 3m, cable is shielded 4 conductor alarm cable which helped a bit but the run was just too long. Dropping clock rate down to 10khz seemed to help a bit but did not resolve it fully. Adding in the LTC4311 and bumping rate back up to 100khz it has been solid for a good time now.

closed 2021-03-01 by Arsenal10108

flyte, 2021-03-01:
Excellent. Glad to hear you've got it working! :tada: