flyte/mqtt-io #163: [Opt-in] Skip sensor errors. Don’t make them unavailable if value temporarily null

a-x-, 2021-02-10:
**Is your feature request related to a problem? Please describe.**
For instance dht22 is well-known buggy sensor. It frequently sends a unreadable trash.
It will be good to add an option to no say about read fails to MQTT

**Describe the solution you'd like**
When I parse it manually, I omit fails.
Also, I take 3 last values and avg it.

**Describe alternatives you've considered**
Do not use pi-mqtt-gpio

a-x-, 2021-02-10:
[image]

a-x-, 2021-02-10:
[image]

Right after sending this message I again got `2021-02-11 00:45:16,226 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'wc_humidity' value 40.5` for a second, after a few minutes of nulls

a-x-, 2021-02-10:
I don’t understand why it becomes Unavailable.
I cannot sniff anything weird over MQTT.
Also, code looks ok:
https://github.com/flyte/pi-mqtt-gpio/blob/35b8f2269d1217d239f6a1a7898a97efb50d5486/pi_mqtt_gpio/server.py#L752-L757

a-x-, 2021-02-11:
I guess I find the reason.
pi-mqtt-gpio sends config:
```yaml
name: wc_humidity
unique_id: pi-mqtt-gpio-146b5dae_wc_climate_dht22_output_wc_humidity
state_topic: wc/sensor/wc_humidity
availability_topic: wc/status
payload_available: running
payload_not_available: dead
expire_after: 25
device:
  manufacturer: MQTT GPIO
  identifiers:
    - mqtt-gpio
    - pi-mqtt-gpio-146b5dae
  name: MQTT GPIO
platform: mqtt
```

where expire_after: 25 # sec
is too short

a-x-, 2021-02-11:
Next, I found the code:
https://github.com/flyte/pi-mqtt-gpio/blob/35b8f2269d1217d239f6a1a7898a97efb50d5486/pi_mqtt_gpio/server.py#L1016

Looks I can hack it by increasing interval in my config.

a-x-, 2021-02-11:
It might be helpful to add more straight config as I ask.
But it’s up to you

flyte, 2021-02-11:
Ah, I see what you mean. The expire_after should be configurable in the config, yeah. I'll add it to the list, thanks.

a-x-, 2021-02-12:
Thanks! I'll be waiting this config option

flyte, 2021-02-25:
This is now implemented in the new version #178 so please give that a try.

closed 2021-02-25 by flyte