flyte/mqtt-io #164: Correction option for sensor values is usefull

alive-corpse, 2021-02-10:
I think it will be convinient to append some correction option for temperature, humidity and other float or integer metrics. Most of cheap chineese tempeature modules gives incorrect values and difference between two modules can be more than 4 degrees in one point of space.

Examples:
  correction: 2.3 # append 2.3 to value before sending to mqtt

  correction: -1.5 # decrease value by 1.5

flyte, 2021-02-11:
Thanks for the suggestion. That would probably be a good feature to have.

andyfsimon, 2023-08-09:
I second this request. Would be handy to have something along the lines of linear calibration as done in ESPHOME. 
DHT22 sensors are always a bit off and just add/subtract just doesn't do justice to the readings.

ilvalle, 2024-11-03:
@flyte do you have any plan to implement the basic feature to add or subtract scalar?

flyte, 2024-11-05:
Personally no, but I still think it sounds like a good idea!