flyte/mqtt-io #165: Add support for BME680

kapratt, 2021-02-13:
**Describe the solution you'd like**
BME680 is similar to the BME280 sensor, but adds air quality sensors.

**Describe alternatives you've considered**
I attempted to use the BME280 module, but the readings are wildly inaccurate.

> 2021-02-13 14:34:15,397 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'temperature' value 139.7925
> 2021-02-13 14:34:15,416 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'humidity' value 100.0
> 2021-02-13 14:34:15,435 mqtt_gpio (INFO): sensor_timer_thread: reading sensor 'pressure' value -1465.4358

**Additional context**
Adafruit Circuitpython package for the BME680 is here: https://github.com/adafruit/Adafruit_CircuitPython_BME680

kapratt, 2021-02-15:
A little more research has found an easier package for the BME680 I think: https://github.com/pimoroni/bme680-python.

I'll try to write something up based on the current BME280 code, but I'm not the best with Python yet!

kapratt, 2021-02-16:
I put the code together for only temperature, humidity, and pressure, and it seems to work well! Hopefully someone can look at it and make sure that it looks ok.

#166

flyte, 2021-02-26:
Implemented in #166 :tada:

closed 2021-02-26 by flyte