flyte/mqtt-io #174: Load actual config.schema.yml into pi_mqtt_gpio/__init__.py

a-x-, 2021-02-22:
**Is your feature request related to a problem? Please describe.**
Currently contributors used to change the same config 2 times

**Describe the solution you'd like**
Load actual file instead of `CONFIG_SCHEMA = yaml.safe_load("""`.
Perform any transformations if they required.

**Describe alternatives you've considered**
continue change config 2 times, suffer, forget to sync changes; stop to using the pi-mqtt-gpio

flyte, 2021-02-23:
Yeah, this is annoying, sorry about that! Although I've changed this in the asyncio version in `feature/asyncio`. Feel free to give that a try. It's pretty much compatible with existing configs.

I don't plan on fixing it on this version of the code, since I'm releasing the async version soon, so I'll close this issue.

closed 2021-02-23 by flyte

flyte, 2021-02-23:
Instructions for testing out the asyncio version are in https://github.com/flyte/pi-mqtt-gpio/issues/141