flyte/mqtt-io #185: KeyError: 'digits'

imkebe, 2021-03-09:
```
2021-03-09 22:18:53 mqtt_io.server [INFO] Read sensor 'kichen_temperature' value of 22.56
2021-03-09 22:18:53 mqtt_io.server [ERROR] Exception in task: <Task finished coro=<MqttIo._init_sensor_inputs.<locals>.publish_sensor_callback() done, defined at /home/pi/ve/lib/python3.7/site-packages/mqtt_io/server.py:491> exception=KeyError('digits')>:
Traceback (most recent call last):
  File "/home/pi/ve/lib/python3.7/site-packages/mqtt_io/server.py", line 1038, in _remove_finished_transient_tasks
    raise exception
  File "/home/pi/ve/lib/python3.7/site-packages/mqtt_io/server.py", line 493, in publish_sensor_callback
    digits: int = sens_conf["digits"]
KeyError: 'digits'
```

config
```
# extra stuff was excluded 

sensor_modules:
  - name: kichen_temperature
    module: ds18b
    type: DS18B20
    address: "800000036406"

sensor_inputs:
  - name: kichen_temperature
    module: kichen_temperature 
    type: temperature
    interval: 60
    ha_discovery:
      component: binary_sensor
      name: Temperature
      device_class: temperature
```

It doesn't matter if i declare /digits/ in config or not

imkebe, 2021-03-11:
Thx for quick action. Looks like that #185 patch introduced another problem:

```
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/ve/lib/python3.7/site-packages/mqtt_io/__main__.py", line 38, in <module>
    main()
  File "/home/pi/ve/lib/python3.7/site-packages/mqtt_io/__main__.py", line 33, in main
    mqtt_gpio.run()
  File "/home/pi/ve/lib/python3.7/site-packages/mqtt_io/server.py", line 1137, in run
    self._init_sensor_inputs()
  File "/home/pi/ve/lib/python3.7/site-packages/mqtt_io/server.py", line 518, in _init_sensor_inputs
    sens_conf, sensor_module
  File "/home/pi/ve/lib/python3.7/site-packages/mqtt_io/config/__init__.py", line 198, in validate_and_normalise_sensor_input_config
    return validate_and_normalise_config(config, sensor_input_schema, allow_unknown=False)
  File "/home/pi/ve/lib/python3.7/site-packages/mqtt_io/config/__init__.py", line 106, in validate_and_normalise_config
    "Config did not validate:\n%s" % yaml.dump(validator.errors)
mqtt_io.exceptions.ConfigValidationFailed: Config did not validate:
type:
- unknown field
```

flyte, 2021-03-11:
Added a new issue for this one at #186

closed 2021-03-11 by flyte