flyte/mqtt-io #186: Unknown 'type' field in sensor validation

flyte, 2021-03-11:
Thx for quick action. Looks like that [patch](https://github.com/flyte/mqtt-io/commit/b50699c76594cad09bd35afccafcf208029925da) introduced another problem:

```
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/ve/lib/python3.7/site-packages/mqtt_io/__main__.py", line 38, in <module>
    main()
  File "/home/pi/ve/lib/python3.7/site-packages/mqtt_io/__main__.py", line 33, in main
    mqtt_gpio.run()
  File "/home/pi/ve/lib/python3.7/site-packages/mqtt_io/server.py", line 1137, in run
    self._init_sensor_inputs()
  File "/home/pi/ve/lib/python3.7/site-packages/mqtt_io/server.py", line 518, in _init_sensor_inputs
    sens_conf, sensor_module
  File "/home/pi/ve/lib/python3.7/site-packages/mqtt_io/config/__init__.py", line 198, in validate_and_normalise_sensor_input_config
    return validate_and_normalise_config(config, sensor_input_schema, allow_unknown=False)
  File "/home/pi/ve/lib/python3.7/site-packages/mqtt_io/config/__init__.py", line 106, in validate_and_normalise_config
    "Config did not validate:\n%s" % yaml.dump(validator.errors)
mqtt_io.exceptions.ConfigValidationFailed: Config did not validate:
type:
- unknown field
```

_Originally posted by @imkebe in https://github.com/flyte/mqtt-io/issues/185#issuecomment-796883370_

flyte, 2021-03-11:
@imkebe Could you please install version 2.1.0 and add the following to your config file:

```yaml
reporting:
  enabled: yes
  issue_id: 186
```

and run the software again? This will send an error report that will include more information to help me diagnose this.

See the [reporting documentation](https://flyte.github.io/mqtt-io/#/config/reference/reporting/) for information about this reporting feature.

imkebe, 2021-03-15:
@flyte I need few days more since my sensor started to randomly change address. Then I can validate.

flyte, 2021-03-29:
Don't worry about the issue being closed. Just reply when you've got time
to test it and I'll reopen it.

closed 2021-04-15 by no-response[bot]