flyte/mqtt-io #187: Not reconnecting when MQTT broker not available

pilmrc, 2021-03-15:
I have MQTT-IO on a different system from where MQTT is installed.
If I have to reboot the raspberry pi where MQTT is running, MQTT-IO is loosing connection but not trying to connect anymore.

When MQTT broker come up, any event MQTT-IO tries to publish is reporting the following error:

`2021-03-15 08:07:40 mqtt_io.server [INFO] Digital input 'alarm_out1' value changed to False
2021-03-15 08:07:40 mqtt_io.server [ERROR] Exception while handling MQTT task:
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 996, in _mqtt_task_loop await entry.coro
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 635, in _mqtt_publish await self.mqtt.publish(msg)
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/mqtt/asyncio_mqtt.py", line 76, in publish topic=msg.topic, payload=msg.payload, qos=msg.qos, retain=msg.retain
  File "/usr/local/lib/python3.7/dist-packages/asyncio_mqtt/client.py", line 128, in publish raise MqttCodeError(info.rc, 'Could not publish message')
asyncio_mqtt.error.MqttCodeError: [code:4] Could not publish message
2021-03-15 08:14:04 mqtt_io.server [INFO] Waiting for our tasks to complete...
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code  exec(code, run_globals)
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/__main__.py", line 38, in <module> main()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/__main__.py", line 33, in main mqtt_gpio.run()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 1174, in run self.loop.run_until_complete(self._main_task)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete return future.result()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 1168, in run await self.shutdown()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 1229, in shutdown wait=False,
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 635, in _mqtt_publish await self.mqtt.publish(msg)
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/mqtt/asyncio_mqtt.py", line 76, in publish topic=msg.topic, payload=msg.payload, qos=msg.qos, retain=msg.retain
  File "/usr/local/lib/python3.7/dist-packages/asyncio_mqtt/client.py", line 128, in publish raise MqttCodeError(info.rc, 'Could not publish message')
asyncio_mqtt.error.MqttCodeError: [code:4] Could not publish message`

flyte, 2021-03-16:
Thanks for the report. I was supposed to link into this at the weekend but I had to work on something else. It's a high priority and I'll try to get it fixed soon.

MTJoker, 2021-03-26:
Very cool that this is already a known issue. 
Just wanted to comment that yesterday I noticed the same issue in my setup. 
I got this behavior as I updated my mosquitto (via apt) to a new version. The mosquitto is restarted due to the update and therefore not available for a very short time. mqtt-io results in the above behavior.

closed 2021-04-21 by flyte

pilmrc, 2021-04-22:
> Closed #187 via #207.

I’ve upgraded Mqtt-io to 2.1.8 but is neither reconnecting after disconnession nor accepting the new configuration parameter for delay in reconnecting and reconnecting count. 

Is the version correct? I’m not too much familiar with python, is there any cache to clean? From pip3 I forced Mqtt-io==2.1.8 and it looks installed (and removed the 2.1.7). 

Thanks for your work. 

Marco

flyte, 2021-04-22:
Ah, so this is because the documentation follows the `develop` branch and I haven't released a new version with the reconnect code yet. I'll need to put some versioning on the docs I reckon. I have some more work to do before I can release that, but you're welcome to try the `develop` branch in the meantime.

flyte, 2021-04-29:
This reconnection code is released in version 2.2.6 but it's maybe still a little buggy. Please do try it out and report back if you have any issues.

RAYs3T, 2022-05-04:
@flyte This still seems to not work properly.
If I restart my broker (Mosquitto 6.1.1) mqtt_io does not notice this. therefore Events in topics (inserted by other applications) are ignored by mqtt_io until restarted.

My config:
```yaml
mqtt:
  host: hass.homenet
  user: rpi-electric-box
  password: "*secret*"
  topic_prefix: mqtt-io/rpi-electric-box
  ha_discovery:
    enabled: yes
  keepalive: 3
  reconnect_delay: 1
  reconnect_count: 0

# GPIO
gpio_modules:
  # Use the Raspberry Pi built-in GPIO
  - name: rpi
    module: raspberrypi

digital_inputs:
  # Pin 0 is an input connected to a doorbell button
  - name: doorbell
    pulldown: true
    bouncetime: 2000
    module: rpi
    pin: 4
    ha_discovery:
      component: binary_sensor
      name: Doorbell
      device_class: motion

digital_outputs:
  # Pin 1 is an output connected to a light
  - name: door_buzzer
    module: rpi
    initial: 'high'
    publish_initial: true
    inverted: true
    timed_set_ms: 2500
    pin: 17
    ha_discovery:
      component: switch
      name: Stairwell Buzzer

```

mqtt_io log (with broker restarts inserted)
```
2022-05-04 11:38:18 mqtt_io.server [INFO] Digital input 'doorbell' value changed to False
2022-05-04 11:38:18 mqtt_io.server [INFO] Connecting to MQTT...
2022-05-04 11:38:18 mqtt_io.server [INFO] Connected to MQTT
2022-05-04 11:38:18 mqtt_io.server [INFO] Subscribed to topic: 'mqtt-io/rpi-electric-box/output/door_buzzer/set'
2022-05-04 11:38:18 mqtt_io.server [INFO] Subscribed to topic: 'mqtt-io/rpi-electric-box/output/door_buzzer/set_on_ms'
2022-05-04 11:38:18 mqtt_io.server [INFO] Subscribed to topic: 'mqtt-io/rpi-electric-box/output/door_buzzer/set_off_ms'

MQTT BROKER STOPPED: Wed  4 May 11:39:45 CEST 2022
...no new log lines in mqtt_io log...
MQTT BROKER STARTED: Wed  4 May 11:40:10 CEST 2022
...no new log lines in mqtt_io log...
at Wed  4 May 11:42:03 CEST 2022 still no reconnect attempt (it somehow must think it still has a connection? So what is the keepalive doing?)
```

restarting mqtt_io resolves the issue and it connects again.

any ides? I tried clean_session aswell as different values for `keepalive`, `reconnect_delay` and `reconnect_count`

PR #207 (2021-04-21): Auto-Reconnect on failure
Fixes #187 

This PR adds two new configuration-options:
reconnect-delay: `Time in seconds to wait between reconnect attempts.`

reconnect-count: 
`Max number of retries of connections before giving up (and exiting).
-1 means infinite reconnects.
The counter is reset if a connection was established successfully.
`