flyte/mqtt-io #188: I can't get it to work

thelegendclub, 2021-03-15:
hi, i'm trying a thousand ways (i'm quite a newbie) and i can't.

First install it normal, then with supervisor either.

I get these errors:

2021-03-15 18: 29: 29,475 mqtt_gpio (INFO): Startup
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec (code, run_globals)
  File "/home/pi/pi_mqtt_gpio/lib/python3.7/site-packages/pi_mqtt_gpio/server.py", line 1269, in <module>
    main (args)
  File "/home/pi/pi_mqtt_gpio/lib/python3.7/site-packages/pi_mqtt_gpio/server.py", line 1064, in main
    client = init_mqtt (config ["mqtt"], config ["digital_outputs"], config ["stream_writes"])
  File "/home/pi/pi_mqtt_gpio/lib/python3.7/site-packages/pi_mqtt_gpio/server.py", line 441, in init_mqtt
    client.tls_set (** tls_kwargs)
  File "/home/pi/pi_mqtt_gpio/lib/python3.7/site-packages/paho/mqtt/client.py", line 823, in tls_set
    context.load_cert_chain (certfile, keyfile)
FileNotFoundError: [Errno 2] No such file or directory

bluestar90, 2021-03-16:
Hi,
give us a look on your conf.yml and your start statement.

Regards Matthias

thelegendclub, 2021-03-16:
el yml es este:

```
mqtt:
  host: 192.168.1.16
  port: 1883
  user: "******"
  password: "********"
  topic_prefix: home/recibidor/interfono
  ha_discovery:
    enabled: yes

    
gpio_modules:
  - name: raspberrypi
    module: raspberrypi
    cleanup: yes

digital_inputs:
  - name: button_1
    module: raspberrypi
    pin: 17
    on_payload: "Off"
    off_payload: "On"
    pullup: yes
    pulldown: no
  - name: button_2
    module: raspberrypi
    pin: 27
    on_payload: "Off"
    off_payload: "On"
    pullup: yes
    pulldown: no
```

flyte, 2021-03-16:
Ah, I think I see what's happened. You seem to be using the old version of the code (pi-mqtt-gpio as opposed to mqtt-io).

@thelegendclub  Could you try again with the instructions here? https://mqtt-io.app/#/deployment/supervisor

thelegendclub, 2021-03-20:
Lo pruebo,

closed 2021-03-30 by no-response[bot]