flyte/mqtt-io #190: Error messages at starting

AingTii, 2021-03-20:
I installed fresh buster and fresh mqttio after starting I got this messages:

`python3 -m mqtt_io config.yml
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting RPi.GPIO
  Downloading https://www.piwheels.org/simple/rpi-gpio/RPi.GPIO-0.7.0-cp37-cp37m-linux_armv7l.whl (69kB)
    100% |████████████████████████████████| 71kB 813kB/s
Installing collected packages: RPi.GPIO
Successfully installed RPi.GPIO-0.7.0
2021-03-20 11:23:36 mqtt_io.server [INFO] Digital input 'DoorBell_1' value changed to True
2021-03-20 11:23:36 mqtt_io.server [INFO] Digital input 'DoorBell_2' value changed to False
2021-03-20 11:23:36 mqtt_io.server [INFO] Connecting to MQTT...
2021-03-20 11:23:36 mqtt_io.server [INFO] Waiting for our tasks to complete...
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/asyncio_mqtt/client.py", line 79, in connect
    await loop.run_in_executor(None, self._client.connect, self._hostname, self._port, 60)
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/paho/mqtt/client.py", line 941, in connect
    return self.reconnect()
  File "/home/pi/.local/lib/python3.7/site-packages/paho/mqtt/client.py", line 1075, in reconnect
    sock = self._create_socket_connection()
  File "/home/pi/.local/lib/python3.7/site-packages/paho/mqtt/client.py", line 3546, in _create_socket_connection
    return socket.create_connection(addr, source_address=source, timeout=self._keepalive)
  File "/usr/lib/python3.7/socket.py", line 727, in create_connection
    raise err
  File "/usr/lib/python3.7/socket.py", line 716, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 1142, in run
    await self._connect_mqtt()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 549, in _connect_mqtt
    await self.mqtt.connect()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/mqtt/asyncio_mqtt.py", line 63, in connect
    await self._client.connect(timeout=timeout)
  File "/home/pi/.local/lib/python3.7/site-packages/asyncio_mqtt/client.py", line 88, in connect
    raise MqttError(str(error))
asyncio_mqtt.error.MqttError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/__main__.py", line 85, in <module>
    main()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/__main__.py", line 80, in main
    mqtt_gpio.run()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 1174, in run
    self.loop.run_until_complete(self._main_task)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 1168, in run
    await self.shutdown()
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 1229, in shutdown
    wait=False,
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 635, in _mqtt_publish
    await self.mqtt.publish(msg)
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/mqtt/asyncio_mqtt.py", line 76, in publish
    topic=msg.topic, payload=msg.payload, qos=msg.qos, retain=msg.retain
  File "/home/pi/.local/lib/python3.7/site-packages/asyncio_mqtt/client.py", line 128, in publish
    raise MqttCodeError(info.rc, 'Could not publish message')
`

my config:

`mqtt:
  host: 192.168.5.4
  port: 1889
  user: "myUser"
  password: "MyPassword"
  topic_prefix: doorpi
  client_id: doorpi

gpio_modules:
  - name: rpi
    module: raspberrypi

digital_inputs:
  - name: DoorBell_1
    module: rpi
    pin: 3 
    on_payload: "true"
    off_payload: "false"

  - name: DoorBell_2
    module: rpi
    pin: 4
    on_payload: "true"
    off_payload: "false"

digital_outputs:
  - name: DoorOpen
    module: rpi
    pin: 18
    on_payload: "true"
    off_payload: "false"

  - name: DoorOpen1
    module: rpi
    pin: 23
    on_payload: "true"
    off_payload: "false"

  - name: DoorOpen2
    module: rpi
    pin: 24
    on_payload: "true"
    off_payload: "false"
`

flyte, 2021-03-20:
Looks like it's not able to connect to your MQTT server. Are you sure it's
running and accessible with those details?

AingTii, 2021-03-20:
i tried the whole day, yes you are rigth, the iobroker mqtt is running on the slave,
so I need to use the slave IP. sorry
now its running, thank you

flyte, 2021-03-20:
Glad you got it sorted! We should have some better log output in this case.
My fault for not making it clearer.

AingTii, 2021-03-23:
Maybe, but in this Issue the Problem was in front of the screan ;-)
Tank you for the very quick Response.

closed 2021-03-23 by flyte