flyte/mqtt-io #191: Change of pin numbering?

MTJoker, 2021-03-22:
Hi,
I have been using pi-mqtt-gpio since a longer time now and wanted to change to the new mqtt-io. I am using the piface2 module. 
Was there a change regarding the pin numbering? In my old configuration, I had for example:

```
digital_inputs:
  - name: bellbutton_garden
    module: piface
    pin: 0
    on_payload: "on"
    off_payload: "off"
    pullup: yes
    pulldown: no

digital_outputs:
  - name: bellrelay
    module: piface
    pin: 0
    timed_set_ms: 500
    on_payload: "on"
    off_payload: "off"
```
Now this previously working configuration gives the following error:
```
Config did not validate due to errors in the following sections:
gpio_modules:
  piface:
  - Pin '0' has more than one configuration in digital_inputs or digital_outputs
```

Is this the intended behavior? If yes, then how are the pins of the piface2 module numbered now? In the documentation of the pifacedigitalio python module, it seems that the same pin numbers are available as input and output, e.g. output_pins[0..7] and input_pins[0..7].

flyte, 2021-03-22:
That's interesting! I've added that validation on the assumption that it
would be impossible to do that! How exactly is the same pin used as an
input and an output? Or does the piface have a separate input 0 pin and
output 0 pin?

MTJoker, 2021-03-22:
The latter, so these are separate pins. My understanding is, that pin 0 to 7 exist for input as well as for output. This is also how they are labeled on the board, see for example the illustration on the middle of this page:
http://www.piface.org.uk/products/piface_digital_2/

flyte, 2021-03-24:
I've released version 2.1.2 which no longer uses that config validation. Hope that works for you. Just comment here again if it doesn't.

closed 2021-03-24 by flyte

MTJoker, 2021-03-26:
Works for me now, thanks for the fast fix.
One additional comment: Maybe it would be reasonable to still fail the validation, if one pin number is configured twice as input or twice as output. This is also not possible for the piface2. There is only the possibility to configure the same pin number (number, not the same pin) as input and output.