flyte/mqtt-io #193: switch timed_set_ms gets stuck in opposite state from initial

se7enmilgram, 2021-03-25:
Greetings, I have some digital outputs that I'm using to fire a relay to close a momentary dry contact, and I think I found an area where some enhancement could be introduced.

If multiple messages arrive in the timed_set_ms interval, the switch can be toggled mid toggle, which flips its state, ie:

1st trigger -> set to on state for timed_set_ms and then turn back off
2nd trigger -> set to off state for timed_set_ms, and then turn back on

The easiest thing might be to make timed_set_ms ignore further commands until the interval has expired. Optionally, you could introduce an attribute to restore this functionality, ie "timed_ignore_messages", although I can't think of two many instances where you'd want a momentary low digital output to switch to a momentary high digital output. Pretty sure I could submit a PR for this after I get some feedback.

flyte, 2021-03-25:
Hey @se7enmilgram 

Yeah, this had occurred to me in the past. I kept it as simple as possible for the moment, until someone found a valid reason for it to be more complicated!

By default, I wouldn't want the software to just ignore commands while the timer is running. It should make every effort to accommodate the commands being sent, leaving it up to the software sending the commands to ensure that it doesn't make the devices do weird things. However, if this means that mqtt-io gets stuck in a weird state then that's probably something that should be fixed.

I'm not quite sure I understand the details of your report fully though. Are you saying that the first off -> wait -> on timer ends up flipping to off -> wait -> off if you send another command in the middle?

se7enmilgram, 2021-03-25:
Yeah. To be more specific: 

`  - name: shed_relay1
    module: rpi
    pin: 23
    inverted: true
    timed_set_ms: 25
    initial: high
    publish_initial: true
    ha_discovery:
        name: Shed Relay1
        component: switch`

Intent is that the output is brought low for 25ms, and then brought back high. If there's a second command via MQTT during that interval while it's low, it's state is toggled high again, then the coroutine sets it high, and the second coroutine sets it back to low. 

How this happens in Home Assistant is spamming the switch control so that you send two on events before mqtt-io's state update message gets back to HA, at least as near as I can tell. I'm going to set a longer timed_set_ms and see if I can sort out the behavior by generating some near-concurrent MQTT messages outside of HA.

I think it could be as simple as having a set of output_conf / payloads to silently swallow during the timed_set_ms interval, and having the coroutine that sets the state back after the interval remove it from the list when it's done. Alternatively, maybe it would make more sense to put it back in the queue instead of silently swallowing it? Trying to keep with your aim of simplicity, as I could just send a set_ms message from HA and probably avoid the state being flipped from off-on-off to on-off-on. On the other hand, for momentary dry contact applications, the timed_set_ms option is excellent and works well with HA / other applications, so maybe a little complexity is worth it?

se7enmilgram, 2021-03-25:
Just validated the behavior, this happens because timed_set_ms affects both ON and OFF setting payloads, and I'm a bit of an idiot.

flyte, 2021-03-25:
Ah, I SEE! I thought we were talking about the `set_on_ms` and `set_off_ms` topic suffixes. My mistake.

So where `-` and `_` represent 5ms time blocks of `high` and `low` logic levels:

### Single request

1. Set low for 25ms

```
----------1_________________________1----------
```

This is fine.

### Two opposing requests

1. Set low for 25ms
2. Set high for 25ms

```
----------1_____2--------------------1-----2__________
```

2 overrides 1 here, since it will be set to high for 25ms.

### Two of the same requests

1. Set low for 25ms
2. Set low for 25ms

```
----------1_____2_______________1-----2----------
```

1 overrides 2 here, since it will be low for 25ms from 1 but not from 2.

So the change in behaviour here might be to cancel 1 and keep 2, so that the latter commands always override the earlier ones?

se7enmilgram, 2021-03-25:
I was wondering if it would make sense to add additional attributes of timed_ms_on and timed_ms_off, where they only affect ON or OFF messages. Might introduce some interesting capabilities where you want different intervals for on and off messages, but could also introduce even weirder behaviors. 

A trickier thing is user intention: if my output is already off and I send another off message, we get:

```--------1-----------1____________```

So by sending an off, I turn it on after the interval.

flyte, 2021-03-26:
> if my output is already off and I send another off message

I find it's clearer to talk about it in terms of 'high' and 'low' here, since 'on' and 'off' could mean either.

At the moment, if the output is high already and a user sends a `set_on_ms` (where on == high), then yeah, it'll go low after the period.

1. Set high for 25ms

```
----------1-------------------------1__________
```

I find that the "least astonishing" behaviour. What do you expect it to do?

I'd accept a PR for adding `timed_set_on_ms` and `timed_set_off_ms`. I'd expect those config values to override anything set in `timed_set_ms`. If `timed_set_on_ms`, `timed_set_off_ms` and `timed_set_ms` were all set, perhaps it should fail config validation too, although I think I'd actually prefer a warning log message on start up that says something about `timed_set_ms` being overridden instead.

se7enmilgram, 2021-03-26:
I was using the the payload values that trigger the change. The log message
and documentation say "setting to the previous state", but it's really just
setting to the opposite of the requested state without looking at the
existing state. Generally  I wouldn't expect an off payload to behave as if
I'd sent an on payload when the output was already off. Honestly I can't
think of an application where I'd want both momentary on and off states On
the same digital output.

I'll look at introducing the on/off attributes, config validations, and
warnings about overriding. My preferred behavior would be that you could
set them independently, and in the case where there's a timed_set_ms
and timed_set_on_ms,
the on attribute would be used for ON payloads and timed_set_ms would still
apply to OFF payloads.

se7enmilgram, 2021-03-26:
Could use some feedback on this because I'm am a total noob when it comes to async:

https://github.com/se7enmilgram/mqtt-io/blob/develop/mqtt_io/server.py (line 1072)

These changes are producing my desired results. SET messages only trigger reset_timer() for ON payloads when timed_set_on_ms is specified. However, I still had problems when triggering multiple messages. That's when I changed the definition of reset_timer() to pass in value. Things started behaving. Got me thinking that if this is the fix, on the original code I should be able to modify what reset_timer() does on another digital_output on the same module. Turns out, this is the case. If I turn set a digital_output with a timed_set_ms configured, and during the interval set another one to the opposite state, the original task changes what it's doing. 

If my understanding is right, my fix is only working because we're redefining reset_timer() each time the loop skips the continue statement in your error handling. Would it make more sense to define reset_time() outside of the loop and pass in the desired values to the coroutine from within the loop?

flyte, 2021-03-27:
If you make a draft PR then it'll be easier to review and comment on the changes you're making.

closed 2021-04-15 by no-response[bot]