flyte/mqtt-io #195: MCP3008 Sensor Module Code Typo

skilletjohn, 2021-03-26:
I believe this should be a quick fix as well:

`pi@control-panel:~ $ python3 -m mqtt_io config.yml
/home/pi/.local/lib/python3.7/site-packages/mqtt_io/modules/gpio/raspberrypi.py:56: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self.io.setup(pin, direction, pull_up_down=pullup, initial=initial_int)
2021-03-26 10:51:20 mqtt_io.server [INFO] Connecting to MQTT...
2021-03-26 10:51:20 mqtt_io.server [INFO] Connected to MQTT
2021-03-26 10:51:20 mqtt_io.server [INFO] Subscribed to topic: 'cpanel/output/lt_garage/set'
2021-03-26 10:51:20 mqtt_io.server [INFO] Subscribed to topic: 'cpanel/output/lt_garage/set_on_ms'
2021-03-26 10:51:20 mqtt_io.server [INFO] Subscribed to topic: 'cpanel/output/lt_garage/set_off_ms'
2021-03-26 10:51:20 mqtt_io.server [INFO] Digital output 'lt_garage' set to False (off)
2021-03-26 10:51:21 mqtt_io.server [ERROR] Exception in task: <Task finished coro=<MqttIo._init_sensor_inputs.<locals>.poll_sensor() done, defined at /home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py:525> exception=TypeError("'<=' not supported between instances of 'int' and 'str'")>:
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 1038, in _remove_finished_transient_tasks
    raise exception
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/server.py", line 530, in poll_sensor
    value = await sensor_module.async_get_value(sens_conf)
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/modules/sensor/__init__.py", line 57, in async_get_value
    return await loop.run_in_executor(ThreadPoolExecutor(), self.get_value, sens_conf)
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/mqtt_io/modules/sensor/mcp3008.py", line 56, in get_value
    return cast(int, self.mcp.read_adc(sens_conf["channel"]))
  File "/home/pi/.local/lib/python3.7/site-packages/Adafruit_MCP3008/MCP3008.py", line 53, in read_adc
    assert 0 <= adc_number <= 7, 'ADC number must be a value of 0-7!'
TypeError: '<=' not supported between instances of 'int' and 'str'
`
Thank you!

flyte, 2021-03-27:
@skilletjohn I've pushed a fix for that to `develop`. Looks like I messed up the module when I moved it over to the new version. You should be able to install a version directly from the develop branch using pip:

```bash
pip install git+https://github.com/flyte/mqtt-io.git@develop#egg=mqtt-io
```

Could you give it a try and let me know please?

@BenjiU does this module still work for you too?

skilletjohn, 2021-03-27:
@flyte - Sorry, I'm a noob when it comes to Python dev.  I can't get your dev package to install.

```
pi@control-panel:~ $ pip install git+https://github.com/flyte/mqtt-io.git@develop#egg=mqtt-io
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting mqtt-io from git+https://github.com/flyte/mqtt-io.git@develop#egg=mqtt-io
  Cloning https://github.com/flyte/mqtt-io.git (to revision develop) to /tmp/pip-install-DLRu4p/mqtt-io
  Installing build dependencies ... done
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    IOError: [Errno 2] No such file or directory: '/tmp/pip-install-DLRu4p/mqtt-io/setup.py'
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-DLRu4p/mqtt-io/
```

I can wait until the next release once @BenjiU gets back with you.  Thanks!

flyte, 2021-03-27:
Ahh yeah, my bad. I guess you can't install like that when using poetry.

skilletjohn, 2021-04-01:
@flyte - MCP3008 sensor seems to be working fine (VM as well).  Thank you so much, this is a great Python module to use with my Home Assistant automation.  Once word gets out, this should become very popular.  Appreciate all your hard work on this project!

closed 2021-04-01 by skilletjohn

flyte, 2021-04-01:
Thanks for letting me know, and you're welcome!