flyte/mqtt-io #197: Stream Output is Folder not datapoint

AingTii, 2021-03-28:
Hi flyte,

I just test the serial stream, and my prblem to work with that in ioBroker is, that the
incomming Message its shown in the Folder because the send dataoint is below that.

Now its like:

stream
    RFID (name:)   <= here it shows the incomming stream
          send

[image]

Better its like:

stream
      RFID
           received
           send          
          
  thank you

flyte, 2021-04-01:
As far as I'm aware, there's nothing in the MQTT spec that dictates that messages should not be published on topics which have messages published on sub-topics.

I haven't used ioBroker before, so I don't really know what issue this causes. Could you explain a bit more about the problems it causes? I would expect any software that interfaces with MQTT to be able to handle this configuration.

Cheers

AingTii, 2021-04-01:
Ok, I understand.
It seems ioBroker manage the topic path.

for example if the topic is like doorio/stream/RFID and another topic is like doorio/stream/RFID/send
in ioBroker is  a folder path

stream => folder
RFID => folder (because there is one another after RFID in the complete topic
send => shown a datapoint.

So my idea to make that looks better also in iobroker

is it possible, you add on outgoing topic behind the name what we gibt in the config file another
suffix like /received 

so in the config file we set

```
mqtt:
  host: xxx.xx.x.x
  port: 1883
  user: "xxxx"
  password: "xxxxr"
  topic_prefix: doorpi
  client_id: doorpi

# Streams
stream_modules:
  # A serial port to communicate with the house alarm system
  - name: RFID
    module: serial
    device: /dev/ttyUSB0
    baud: 9600
```

and the outgoing topic will be:
doorio/stream/RFID/received

in subscribed topic for the input will be:
doorio/stream/RFID/send

I hope I explained it understandable, my english is not very well while explaining

Thank you