flyte/mqtt-io #218: Pulse counter feature

RobHeijligersHolland, 2021-04-28:
Hi there,

Can I suggest an pulse counter feature.
So add +1 to an integer value if a digital input interrupts.
The integer value is then send trough MQTT.

The integer value should be retentive (stored in case of a system reboot)
I considered a small SQLITE database (good write protection and data corruption guard)

Furthermore
 - Possible to reset or even change the value trough MQTT
 - Input should support debounce 
 - Only do MQTT update after a configurable amount of counts (to prevent many messages)
 - Could have: a configurable amount to add so +0,5 or 3

I am new to Python and Github contributions.
If you want, I could try to implement this myself.

Ps: The reason for this request is that I'm helping out a friend that is restoring a actual Dutch windmill I was asked if I could track the mills total revelations. How cool is that :)

flyte, 2021-04-29:
Hey, that _is_ pretty cool!

I like the idea, it's a feature I'd gladly include.

I'd want to keep the "authoritative value" somewhere other than in this software, which means that we would need some kind of ability to set the value via MQTT or have the state pulled from HTTP or file system etc. on initialisation. Permanent state is not really within scope of this project - it's better done by some kind of real database.

As far as implementing it yourself, being new to Python it's probably not going to be the easiest. It'll require some non-trivial changes to the core (`server.py`) which even I find daunting and I wrote the thing :grin:.

I'll keep the issue open and have a think about how it might be done, then implement it when I've got some time (can't really say when that'll be unfortunately).

RobHeijligersHolland, 2021-04-29:
Great,

You first think about it. And let me know if I can help in any way.  

I wrote similar code (as needed here) in C#, and also applied an SQLite database. I was really impressed with the ease of implementation and the write stability to a SD card on a Raspberry Pi in a battery powered situation.
Anyway have a thought about it, if you want me to provide a Python interface for SQL storage or anything I'm happy to help.

DominicWindisch, 2021-12-14:
Hey, I've been using mqtt-io for a few days now to count pulses of my gas meter and publish those to home assistant. So far, I'm using a custom little automation in HA to count the pulses. It works, but it's of course not a really good solution regarding missing pulses due to reboots of HA after updates etc. 

So, I thought to myself, how hard can it be to add this. I'm quite experienced in a few programming languages - however, not including python. And so it turned out to be harder than I anticipated. 😬 

Having a look at [server.py](https://github.com/flyte/mqtt-io/blob/6297a22f1200a35f890c2487428ee84ab2c671cd/mqtt_io/server.py#L357), I think one could rather easily add the incrementing in the definition of the callback. An easy way to retain the state would be a local file. Extending the config (specifically `self.digital_input_configs`) with  
- a filepath,
- an initial value (in case the file does not exist yet), 
- the incrementation per pulse and
- pulses to accumulate before publishing to MQTT

should do the trick. 
That's of course only a very basic implementation (without the cool MQTT config stuff) but miles ahead of losing steps on the receiving end due to a multitude of reasons...

Does this make any sense?

DominicWindisch, 2021-12-14:
There's a working demo here: https://github.com/DominicWindisch/mqtt-io 
The config entry looks like this:

```
digital_inputs:
  - name: impulse_gaszaehler_counter
    module: rpi
    pin: 4
    is_counter: true
    old_value: 18500
    increment_per_impulse: 0.01
    min_pulses_between_transmission: 1
    file_path: /home/pi/counter_value.txt
```

`is_counter` enables counting the pulses
`old_value` is used as an initial value when the file does not exist
`increment_per_impulse` is the increment of the value for each time the digital input is "true" (may still be inverted)
`min_pulses_between_transmission` is the number of necessary pulses before publishing an MQTT message
`file_path` is the path to the file which holds the current value (ascii floating point number)

Value is rounded to 6 decimal places to avoid floating point imprecision (values like 15.1299999999999). 

There's pretty much zero error checking in this implementation but it works as a proof of concept. :)

BenjiU, 2022-11-04:
@DominicWindisch: would it be possible to bring your POC back to this repo? And may be "expand" the error checking. ;-)

DominicWindisch, 2022-11-04:
@BenjiU Appears like I'm not allowed to create a pull request for this from my fork into dev branch. Should be able to merge without problems, though. I'd go ahead and add some basic error checking when the general approached I've used (zero experience with mqtt-io let alone python) is approved. 

Changes can be seen [here](https://github.com/flyte/mqtt-io/compare/develop...DominicWindisch:mqtt-io:develop).

sandip1894, 2023-01-03:
An alternative to a storing the accumulated value is to just increment the counter (with whatever value) and then output the value at some time interval and at the same time reset the counter to zero to start over. This way it would be a kind of speedometer or, as I want to use the same feature, a measure of the water flow.