flyte/mqtt-io #220: Sensordata as string instead of number

birdie1, 2021-04-28:
I just submitted a pull request for a new module for the ina219 dc current sensor.

It returns up to 4 values. My current homeassistant setup uses mostly json data over mqtt. As i have many of the ina219 sensors, it would be for me easier to return all values from one sensor in one mqtt message. 

```
testing/sensor/ina_main_out_power 48.35
testing/sensor/ina_main_out_current 4.03
testing/sensor/ina_main_out_shunt_voltage 2.98
testing/sensor/ina_main_out_bus_voltage 12.78
```
vs 
```
testing/sensor/ina_main_out {"power": 48.35, "current": 4.03, "shunt_voltage": 2.98, "bus_voltage": 12.78}
```

The poll_sensor method tries to round all data. Would it be possible to send string or numbers?

```
2021-04-29 01:32:14 mqtt_io.server [ERROR] Exception in task: <Task finished coro=<MqttIo._init_sensor_inputs.<locals>.poll_sensor() done, defined at /root/mqtt-io-mount/mqtt_io/server.py:542> exception=TypeError("type str doesn't define __round__ method")>:
Traceback (most recent call last):
  File "/root/mqtt-io-mount/mqtt_io/server.py", line 1076, in _remove_finished_transient_tasks
    raise exception
  File "/root/mqtt-io-mount/mqtt_io/server.py", line 568, in poll_sensor
    value = round(value, sens_conf["digits"])
TypeError: type str doesn't define __round__ method
2021-04-29 01:32:14,030 - ERROR - INA219 Exception in task: <Task finished coro=<MqttIo._init_sensor_inputs.<locals>.poll_sensor() done, defined at /root/mqtt-io-mount/mqtt_io/server.py:542> exception=TypeError("type str doesn't define __round__ method")>:
Traceback (most recent call last):
  File "/root/mqtt-io-mount/mqtt_io/server.py", line 1076, in _remove_finished_transient_tasks
    raise exception
  File "/root/mqtt-io-mount/mqtt_io/server.py", line 568, in poll_sensor
    value = round(value, sens_conf["digits"])
TypeError: type str doesn't define __round__ method```

BenjiU, 2021-07-22:
The Problem here is, that mqtt_io processes every "sensor value" (e.g power, current, shunt_voltage, bus_voltage) sequential or even parallel in different threads. For your issue, mqtt_io must be changed basically in order to group items and transmit them as string. I guess, that is very difficult, right @flyte?
@birdie1: How could a valid yaml configuration look like for your issue? How do you want to group the items?

Or a new ina219_xml driver could be created, that reads all values "intern" and just returns that string. Then mqtt_io could implement an exception, that the value is not rounded and just transmitted.

BenjiU, 2022-11-04:
I close this, as the module is merged. But for the json convertion there was no response. Please open this bug again, if there is still a wish for json strings over mqtt.

closed 2022-11-04 by BenjiU